micro-ROS/micro_ros_espidf_component #1: build issue

MS71, 2020-09-13:
Hi,
suddenly, i get a build issue with this component.

My build commands:
rm -rf build components/micro-ROS/micro_ros_* components/micro-ROS/esp32_toolchain.cmake components/micro-ROS/include/ components/micro-ROS/*.a
export IDF_PATH=/home/maik/workspace/esp32/stable
source /opt/ros/foxy/setup.sh
. $IDF_PATH/export.sh
echo $IDF_PATH
#idf.py menuconfig
idf.py -v build

Any hint howto solve this issue?

[build.txt](https://github.com/micro-ROS/micro_ros_espidf_component/files/5213818/build.txt)

pablogs9, 2020-09-14:
Hello @MS71, I guess that this is an error about mixing two different builds. Anyways remember that this is experimental and we are polishing the integration.

Which IDF version are you using? Does it appear when you follow the steps in a clean environment? Have you tried to delete also the IDF build folder?

It seems that CMake is not finding the correct version of the serialization library (Micro-CDR), could you provide some replication steps to reproduce this from scratch?

Thanks

MS71, 2020-09-14:
Hi,

Which IDF version are you using? 
I checked with ESP IDF v4.1 and v4.0.1.

Does it appear when you follow the steps in a clean environment? 
Yes

Have you tried to delete also the IDF build folder?
=> rm -rf build components/micro-ROS/micro_ros_* components/micro-ROS/esp32_toolchain.cmake components/micro-ROS/include/ 
Also, with clean checkout from your gitrep.

So, Yes

>could you provide some replication steps to reproduce this from scratch?
rm -rf build components/micro-ROS/micro_ros_* components/micro-ROS/esp32_toolchain.cmake components/micro-ROS/include/ components/micro-ROS/*.a
... or a clean clone of your gitrep.

export IDF_PATH=/home/maik/workspace/esp32/esp-idf-v4.1
source /opt/ros/foxy/setup.sh
. $IDF_PATH/export.sh
echo $IDF_PATH
idf.py menuconfig
idf.py -v build

pablogs9, 2020-09-14:
Well, you are having troubles with `fastcdr` this is not a package downloaded by this module [check this](https://github.com/micro-ROS/micro_ros_espidf_component/blob/fb22d6fa083d8aafdb2bbb7ed8384ce5838bb00c/components/micro-ROS/libmicroros.mk#L46).

Why are you sourcing the ROS 2 foxy installation?

For me it works as expected in a clean IDF environment, with no ROS 2 Foxy sourcing, you can check my test here:

https://asciinema.org/a/GGaO235tw7zc0MDiOhgUyT0ZT

MS71, 2020-09-14:
Hi,
yes you are right. Without "source /opt/ros/foxy/setup.sh", all fine.
Sorry about this issue.

closed 2020-09-14 by MS71

pablogs9, 2020-09-15:
Don't you worry, don't hesitate to ask, and please provide feedback! Thanks!