micro-ROS/micro_ros_espidf_component #5: Hardware supported list

farmange, 2020-10-07:
Hi there,

Is there a list of supported hardware somewhere ?
I plan to get an ESP32 DevKitC but I want to be sure it could be compatible with your work.

Thank you

ralph-lange, 2020-10-07:
Hi @farmange, we maintain a list of supported boards at https://micro-ros.github.io/docs/overview/hardware/
Note that the list lacks of details on the RTOS. For example, in case of the ESP32, micro-ROS provides support for FreeRTOS but not for Zephyr.

pablogs9, 2020-10-07:
Hello @farmange, this is ESP-IDF component and it is supposed to work on all boards supported by ESP-IDF build system. I only have tested it in the ESP32 DevkitC v4 and it works, please report if works on other ESP32 hardware.

closed 2020-10-07 by pablogs9