micro-ROS/micro_ros_espidf_component #8: Problems when adding subscriber

AndyBlightLeeds, 2020-10-19:
I have added a subscriber to the demo code in this repo but it continuously reboots.  I started with a working version of the code in this repo and then merged in changes from the file: 
<https://github.com/micro-ROS/micro-ROS-demos/blob/foxy/rclc/int32_publisher_subscriber/main.c>
My code builds and flashes without any problem.

I start the pre-built docker agent: microros/micro-ros-agent:foxy, and run the installed code on my ESP32.  When start the monitor (`idf.py monitor`) I get see these error messages:
```text
...
I (1536) wifi station: connected to ap SSID:TP-LINK_AP_8670
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d5671  PS      : 0x00060730  A0      : 0x800d7030  A1      : 0x3ffca300  
0x400d5671: rcl_node_init at ??:?

A2      : 0x3ffcb044  A3      : 0x3f402ea0  A4      : 0x3f402774  A5      : 0x3ffcb000  
A6      : 0x3ffcafa0  A7      : 0x3ffcafa4  A8      : 0x800d5652  A9      : 0xa5a5a5a5  
A10     : 0x00000000  A11     : 0x3ffca320  A12     : 0x3ffca320  A13     : 0x400d7228  
0x400d7228: __default_zero_allocate at allocator.c:?

A14     : 0x00000000  A15     : 0x3ffcab74  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xa5a5a5b9  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

ELF file SHA256: daeebb271d1632ea

Backtrace: 0x400d566e:0x3ffca300 0x400d702d:0x3ffcaba0 0x400d516a:0x3ffcafe0 0x40087205:0x3ffcb0f0
0x400d566e: rcl_node_init at ??:?

0x400d702d: rclc_node_init_default at ??:?

0x400d516a: appMain at /home/build/micro_ros_idf/build/../main/rclc_int32_publisher_subscriber.c:52 (discriminator 3)

0x40087205: vPortTaskWrapper at /opt/esp/idf/components/freertos/port.c:143

Rebooting...
```

My code is here: https://github.com/AndyBlightLeeds/micro_ros_espidf_component/tree/publisher-1.

Any idea what I'm doing wrong?

pablogs9, 2020-10-19:
Hello @AndyBlightLeeds, have you configured the RMW in the `colcon.meta` file?

Check here and rebuild: https://github.com/AndyBlightLeeds/micro_ros_espidf_component/blob/d08716642e7ecd6c14391c1423f519624ab78c14/components/micro-ROS/colcon.meta#L36

If you want to know more about the static memory pools in the micro-ROS middleware you can read this: https://micro-ros.github.io/docs/tutorials/core/microxrcedds_rmw_configuration/

Let me know if this solves your problem!

AndyBlightLeeds, 2020-10-19:
Hi @pablogs9,
Thanks for the quick reply. 

I have modified the `colcon.meta` file by changing `"-DRMW_UXRCE_MAX_SUBSCRIPTIONS=0",` to 
`"-DRMW_UXRCE_MAX_SUBSCRIPTIONS=1",` rebuilt and flashed but I had exactly the same result.  I then put some rubbish characters in the `colcon.meta` file to generate an error message and rebuilt but the changes were not picked up.  How do I force the build system to rebuild the `colcon.meta` file?

pablogs9, 2020-10-19:
I guess that we have work to do here because files that are generated inside the component/micro-ROS folder are not cleaned properly. 

Could you test this in a brand new fork?

AndyBlightLeeds, 2020-10-19:
I'll test on a new clone of the repo and let you know how I get on.

AndyBlightLeeds, 2020-10-19:
I have created a new clone using the foxy branch with no changes, did menuconfig, build and flash.  All OK. 

I then added rubbish characters to the the `colcon.meta` file and rebuilt.  The build succeeded so the file isn't being picked up.

Just to find out what how the build fails when `colcon.meta` won't compile, I started again with a new clone, did menuconfig, added the characters `AJB` on a new line and rebuilt. The build failed as expected with the message:
```text 
...
Starting >>> rmw_microxrcedds
--- stderr: rmw_microxrcedds
CMake Error: The source directory "/home/build/micro_ros_idf/components/micro-ROS/micro_ros_src/build/rmw_microxrcedds/AJB"-DRMW_UXRCE_MAX_SUBSCRIPTIONS=0"" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
...
```
So the `colcon.meta` file does not get checked on the second rebuild.

Let me know if there is anything else you want me to test.

pablogs9, 2020-10-19:
Ok, so I will keep this issue open in order to remind us that this should be taken into account in further builds of the project.

Regarding your test, are you able to use the micro-ROS subscription by modifying the `colcon.meta` before building the project? Is the `Guru Meditation Error` solved?

AndyBlightLeeds, 2020-10-19:
I took another look at the error messages, I took a guess that the `(discriminator 3)` text means the 3rd parameter to the function 
`rclc_node_init_default`.  I reviewed my code changes again and realised that the `support` structure was not initialised when being used by `rclc_node_init_default`.  Having fixed the code and used the `colcon.meta` with your suggested changes, I now have a working publisher and subscriber. Yay!

The working code is on my branch `publisher-1`. 

Thanks for your help and support.

AndyBlightLeeds, 2020-10-19:
BTW, when cleaning the repo to force a rebuild of the `colcon.meta` file, I used the following commands:

```bash
cd <repo_root>
git reset --hard
git clean -dfx
rm -rf components/micro-ROS/micro_ros_dev/ components/micro-ROS/micro_ros_src
```
These commands reset the `sdkconfig.defaults` file so I keep a modified copy outside the repo with my WiFi SSID and  password and the IP address of my server and copy it over the file in the repo root.  After that, it is the usual sequence of commands to rebuild and run.

zhivko, 2021-01-01:
Can you add subscriber example? Repository link?

AndyBlightLeeds, 2021-01-04:
@zhivko My personal repo has an example of the subscriber, see:
<https://github.com/AndyBlightLeeds/micro_ros_espidf_component/blob/publisher-1/main/rclc_int32_publisher_subscriber.c>
Hope this helps!

pablogs9, 2021-01-11:
Hello @AndyBlightLeeds it would be super nice if you create a PR with your code to our repo. This way everyone will have access to it.

Hope that you have some time for this. Thanks!

AndyBlightLeeds, 2021-01-11:
@pablogs9 Good idea!  I'm busy with other work at the moment but I should be able to create a PR in a few days time.

AndyBlightLeeds, 2021-04-07:
@pablogs9 Finally found time to add my example code to PR #56. Please let me know if anything else needs to be done.

closed 2021-04-07 by pablogs9