micro-ROS/micro_ros_espidf_component #9: Build with arduino-esp32 component

dsryzhov, 2020-10-20:
It would be great to use micro_ros_espidf_component with arduino-esp32 component in applications.
Sometimes it’s more convinient to use arduino libraries than using low level hardware API from esp-idf sdk.

There is esp-idf sdk version mismatch currently for building these components.
Now it's possible to build both using esp-idf 4.0 but some modifications are needed in example app. 
(see needed modifications in branch arduino-component https://github.com/dsryzhov/micro_ros_espidf_component/tree/feature/arduino-component)
see also modification description on thread
https://discourse.ros.org/t/micro-ros-porting-to-esp32/16101/11?u=dmitry_ryzhov

May be two example app should be created (with using arduino-esp32 component and without).

By the way I think about that it would be great to change structure for micro_ros_espidf_component repository. 
Standard way to add components to esp32 applications is to clone them from git to components directory.

Currently repo micro_ros_espidf_component contains not component but example app (that integrate microros component).
Component should be on the first level. And example apps in an example subdirectory. 

This app could be added to example directory

dsryzhov, 2020-10-20:
As I understand the more preferable way to use microros with arduino is to build it with esp32-arduino-lib-builder
https://github.com/espressif/esp32-arduino-lib-builder
It will help to build arduino and micro-ros components once and use them in all applications without rebuilding.
If arduino \ microros components are included in components folders of applications they should be build for every applications that is time consuming. 
As I understand this way is a right direction for microros component integration to Arduino IDE.

pablogs9, 2020-10-20:
Hello @dsryzhov thanks for the information.

Could you explain which are the features that `arduino-esp32 component` enables? Can it coexists with a usual IDF project?

We would like to integrate it as a part of this sample repo, but I was wondering if this should be integrated with this or if we should add some instructions in the README for those ones who are interested in this arduino component features.

As I told you, an integration of micro-ROS in the Arduino IDE is going to be public during these weeks as a beta release.

dsryzhov, 2020-10-20:
It enables to use high level esp32 API in regular esp-idf projects. 
It's the same high level API that is used in Arduino IDE projects. 

For example I create now hadabot firmware (https://www.hadabot.com/) based on microros. Microros enable me to communicate effectively with ROS. I also need to control esp32 hardware.
First version was based on low level esp32 hardware API. It works. But I want it to be refactored and make it more high level
to make public. I don't want to create new high level API for motor, GPIO, timer and so on but use standard high level API from arduino-esp32 component.

Glad to hear about beta release of Arduino IDE integration this week!

dsryzhov, 2020-10-20:
You can checkout arduino-esp32 component to components\arduino project (the same level as Micro-ROS) and build example app with two components using idf.py build command using ESP IDF SDK v.4.0. (4.1 is no supported by arduino-esp32 component yet). Before build you need to configure arduino component (idp.py menuconfig) and make some modifications in example app (to make it compatible with ESP IDF SDK 4.0).
You can build example app with your own docker esp-idf-microros container based on ESP-IDF release 4.0 or use my dsryzhov/esp-idf-microros:idf-v4.0

dsryzhov, 2020-10-20:
Here an instructions how to use arduino-esp32 as esp idf component.
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

pablogs9, 2020-10-20:
Is possible for you to write an instruction list here? I think we can add a section in the README (o a complete new markdown file ) with these instructions.

dsryzhov, 2020-10-20:
yes, I will add instructions.

But to compile code modification of application is needed (to compile with ESP IDF v4.0)
you can see needed modifications one the link
https://github.com/dsryzhov/micro_ros_espidf_component/tree/feature/arduino-component/main
commit
https://github.com/dsryzhov/micro_ros_espidf_component/commit/0febd55b08132abf17f78643b6a5996f1e246f1f

dsryzhov, 2020-10-23:
Hi,

I've made a new docker image for building Micro-ROS applications with arduino-esp32 component.

In this image two components are preinstalled and prebuil inside the image
- Micro-ROS 
- arduino-esp32

You don't need to have this components locally to build you application.
To use them you only need to set EXTRA_COMPONENT_DIRS in your project CMakeLists.txt

set (EXTRA_COMPONENT_DIRS "~/ext_components")

~/ext_components - is the path to the directory with preinstalled components in docker container that is used to build an application. 

Dockerfile for new image is based on the previous one
https://github.com/dsryzhov/hadabot_rds/tree/master/hadabot_microros_firmware/docker/esp-idf-microros-v4.0-arduino

Prebuild docker image is published on hub.docker.com
dsryzhov/esp-idf-microros:idf-v4.0-arduino

You can see an example Micro-ROS\arduino-esp32 application where this docker image  is used 
https://github.com/dsryzhov/hadabot_rds/tree/master/hadabot_microros_firmware

As I'have wrote before curent version of this repository can't be build by ESP-IDF 4.0.
At this time arduino-esp32 is only compatible with this version and can't be built by version 4.1.
Because of this it's not possible to add instructions without code modifications. 

Regards,
Dmitry

dsryzhov, 2020-10-23:
After directory structure will change I can add the second example based on Micro-ROS and arduino-esp32 component. 
The int32_published example would be clear Micro-ROS app (build by ESP IDF v 4.1)
The second example would be Micro-ROS \ arduion-esp32 app (build by ESP IDF v 4.1 using new docker container)

pablogs9, 2020-10-23:
Please PR your proposals

pablogs9, 2021-12-17:
Closing due to inactivity. Reopen if needed.

closed 2021-12-17 by pablogs9