micro-ROS/micro_ros_espidf_component #18: Unable to compile for ESP32-S2 and ESP IDF v4.2

MS71, 2020-11-28:
Hi,
i am trying to build the int32_publisher example for the ESP32-S2.

My steps:

1.)
Install IDF as defined in:
https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/get-started/index.html
git clone -b release/v4.2 --recursive https://github.com/espressif/esp-idf.git

2.)
checkout micro_ros_espidf_component

3.)
idf.py set-target esp32s2
idf.py build

Problem A:
This build is not working because the cmake IDF_TARGET variable is not used.
IDF_TARGET contains "esp32" or "esp32s2".

If I modify the include path in esp32_toolchain.cmake.in, then i am able to compile 
the example. I think esp32_toolchain.cmake.in should use the IDF_TARGET variable
for building the include path.

[esp32s2.patch.txt](https://github.com/micro-ROS/micro_ros_espidf_component/files/5610409/esp32s2.patch.txt)

Unfortunately, it is still not linking. I get this error:
[ 99%] Linking CXX executable int32_publisher.elf
.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/xtensa/libxtensa.a(stdatomic.c.obj): in function `__atomic_fetch_add_8':
esp-idf-v4.2/components/xtensa/stdatomic.c:116: multiple definition of `__atomic_fetch_add_8'; micro_ros_espidf_component/libmicroros.a(librcutils-atomic_64bits.c.obj):atomic_64bits.c:(.text.__atomic_fetch_add_8+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/int32_publisher.elf.dir/build.make:396: int32_publisher.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:2245: CMakeFiles/int32_publisher.elf.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
make failed with exit code 2

Do you have got any hint?

I would be great, if this component could support esp32-s2 too.

pablogs9, 2020-11-30:
Hello @MS71, thanks for the ideas. I'm creating a PR for integrating this ESP32-S2 support: https://github.com/micro-ROS/micro_ros_espidf_component/pull/19

Regarding the atomics: we have a problem here, micro-ROS provides a minimal implementation of [64-bits atomics](https://github.com/micro-ROS/rcutils/blob/foxy/src/atomic_64bits.c) that of course can be enabled and disabled with the flag [RCUTILS_NO_64_ATOMIC](https://github.com/micro-ROS/rcutils/blob/d9e4c2e1cb309ca9f0eea8b2809598636e3bb7e3/CMakeLists.txt#L8) in the colcon.meta.

This optional extension provides implementations if the platform does not have support for: 
- __atomic_load_8
- __atomic_store_8
- __atomic_exchange_8
- __atomic_fetch_add_8

The point is that ESP-IDF for ESP32-S2 is providing just [some of them](https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/components/xtensa/stdatomic.c#L106):
- __atomic_fetch_add_8

This way, if you disable the micro-ROS 64-bit atomic extension it complains about missing load and store, but if you enable it it complains about having fetch_add twice.

Do you have any suggestion?

pablogs9, 2020-11-30:
Cool, we are going to merge the PR: https://github.com/micro-ROS/micro_ros_espidf_component/pull/19/checks?check_run_id=1472900469.

Reopen this if you have any problem or question.

BTW: I don't have here an ESP32-S2 for testing this. If you could give us some feedback, it would be super nice.

closed 2020-11-30 by pablogs9