micro-ROS/micro_ros_espidf_component #20: ESP32 Power Managment added in example application

MS71, 2020-11-30:
Hi, i made some tests with the newly added ESP32-S2 support.
I added a path with the required changes. Maybe, this is 
a good starting point for somebody.

Required Change in configuration:
	CONFIG_PM_ENABLE=y
	CONFIG_PM_DFS_INIT_AUTO=y
	CONFIG_ESP_MAIN_TASK_STACK_SIZE=3000
	CONFIG_FREERTOS_USE_TICKLESS_IDLE=y

ESP32 (WRover Module, 16MByte Flash, 8MByte PSRAM):
	The VCC current is pulsating between min and max value.

	wifi listen_interval = 5
	ros2 topic send interval = 1000 ms

	The mean current (estimated) is: 26 mA

ESP32-S2 (WRover Module, 4MByte Flash, 8MByte PSRAM):
	The VCC current is pulsating between min and max value.

	wifi listen_interval = 5
	ros2 topic send interval = 1000 ms

	The mean current (estimated) is: 14 mA
			
	By increasing the wifi listen_interval (e.g. 30) and 
	the topic send interval (e.g. 60000), the mean 
	current might be reduced to 5 mA.

[esp32_pm_support.patch.txt](https://github.com/micro-ROS/micro_ros_espidf_component/files/5618189/esp32_pm_support.patch.txt)

pablogs9, 2020-12-01:
Could you please open a PR adding a new example in example folder with this new demo?

pablogs9, 2020-12-02:
Closing since https://github.com/micro-ROS/micro_ros_espidf_component/pull/22 merged

closed 2020-12-02 by pablogs9