micro-ROS/micro_ros_espidf_component #23: Custom Message Types?

billieblaze, 2020-12-12:
I ran through most of the tutorials and docs at this point and have a working esp-idf app.    I'm wondering how to add custom message types now, the directions in the docs seem more focused on *firmware.sh files.   

Might be nice to have an example or some light docs around this

billieblaze, 2020-12-13:
after looking at my problem a little more, I think I actually am going to go a different route.  I'm ultimately trying to respond to JointTrajectory _things_ and I don't really need custom messages to do that.

pablogs9, 2020-12-14:
Hello @billieblaze, some points:

1. We definitely should add a way for adding new packages to the IDF Component. I propose something in the [downloading step](https://github.com/micro-ROS/micro_ros_espidf_component/blob/5e2920364dbf8f0979a63574d298bc05eade3071/libmicroros.mk#L47) that copy a folder from the `components` folder. The Arduino port makes something like this to handle extrapackages check [here](https://github.com/micro-ROS/micro_ros_arduino/blob/dea1073c91abc10cf4e7266d04e14abd1b64068a/extras/library_generation/library_generation.sh#L47) and [here](https://github.com/micro-ROS/micro_ros_arduino/issues/14#issuecomment-722242175)

- Regarding implementing services in micro-ROS, you have an example [here](https://github.com/micro-ROS/micro-ROS-demos/blob/foxy/rclc/fibonacci_action_server/main.c). Right now they are using the RCL API, but we are trying to migrate them to RCLC. This is work in progress right now.

I'm going to keep this issue open in order to implement the "add extra packages" functionality.

pablogs9, 2021-01-18:
Solved here: https://github.com/micro-ROS/micro_ros_espidf_component/pull/30

closed 2021-01-18 by pablogs9