micro-ROS/micro_ros_espidf_component #25: "modules" folder

grafoteka, 2020-12-15:
Hello @pablogs9, thank you for sharing the develop for ESP32 modules. 

I have followed the micro-ROS tutorials and now I am trying to implement this component, but I have a key question, please, could you tell me where or what is the "modules" folder that you call in the Usage section. 

Thanks in advance.

Kinds regards, 
Jorge

pablogs9, 2020-12-15:
There is a typo, I meant `components`. Just changed it.

Yes, usually in an ESP32 IDF project you have the following structure: 

```
- myProject/
             - CMakeLists.txt
             - sdkconfig
             - components/ - micro_ros_espidf_component/ 
                           - component2/
             - main/       - CMakeLists.txt
                           - src1.c
                           - src2.c

             - build/
```

So, this repo can be cloned inside the components folder and build your IDF app as usual.

If you have more questions please notice that we are also in the micro-ROS Slack channel: https://join.slack.com/t/micro-ros/shared_invite/zt-k2bfsevp-BgsBVSiv~KaibR17NAU_JA

pablogs9, 2020-12-15:
Please close if resolved.

grafoteka, 2020-12-15:
Sorry, I think I am a little bit confused about how to use this repo.
I tough it was possible to use the code from _firmware/freertos_apps/apps/int32_publisher_ and upload to the board. 
So, basically, I have to create a conventional ESP project?

pablogs9, 2020-12-15:
`freertos_apps` is an independent repo for standalone FreeRTOS applications in the micro-ROS build system. You can read more about the micro-ROS build system [here](https://micro-ros.github.io/docs/concepts/build_system/) and [here](https://github.com/micro-ROS/micro_ros_setup).

As you can see on our web page, this repo is a standalone ESP32 IDF component, so you don't any of the other machinery for building micro-ROS in the ESP-IDF.

This component can be added to your existing ESP-IDF projects and in the folder `examples` you can find a couple of demos projects.

pablogs9, 2021-01-11:
Hello @grafoteka is there any update on this issue?

closed 2021-03-10 by pablogs9