micro-ROS/micro_ros_espidf_component #27: ./main/main.c:11:10: fatal error: rcl/rcl.h: No such file or directory

simonjcarr, 2021-01-10:
I have followed your instructions, but I am obviously doing something wrong. Can you help?

1. I cloned this repo into /esp/esp-idf/components
2. I copied /esp/esp-idf/components/micro_ros_espidf_component/examples/int32_publisher to /esp/int32_publisher
3. cd /esp
4. . $IDF_PATH/export.sh
5. cd /esp/int32_publisher
6. idf.py menuconfig (setup wifi network)
7. idf.py build

I get the following error

../main/main.c:11:10: fatal error: rcl/rcl.h: No such file or directory
 #include <rcl/rcl.h>
                ^~~~~~~~~~~

simonjcarr, 2021-01-10:
I think I understand why i was getting this error. I was using this repo in the wrong place in my code.

closed 2021-01-10 by simonjcarr

wegunterjrFIrefly, 2021-03-30:
would you mind providing the answer for this?  I followed the example here and am still getting the same error you are getting.
I am in the project folder, and have included the micro_ros_espidf_component folder
```
Project:
- components
- -micro_ros_espidf_component
- CMakelists.txt
- main.c
```