micro-ROS/micro_ros_espidf_component #28: I have the library compiled in the components folder --- Now What?????

simonjcarr, 2021-01-10:
Can you provide some instructions on how to use this library. 

1. I created a copy of the IDF hello_world project
2. I created a folder "components" in this project
3. I cloned your repo into the components folder
4. I ran idf.py menuconfig (and configured my wifi settings)
5. I ran idf.py build
6. I ran idf.py flash monitor (I see the IDF code printing Hello World to the terminal)
7. I ran the docker agent ***** I don't see any communication ***** 

I don't understand how your repo is supposed to be used. 
I don't know how your micro-ros code knows which agent to talk to over UDP?? 

Can you help.

uLipe, 2021-01-10:
Hi @simonjcarr , have you checked the examples? They provide how to get started on a micro-ros project.

This repository is intended to be used as a component in ESP-IDF based projects, that is it:

- Install esp-idf in your machine;
- Create your esp-idf project folder, besides main, create a folder called components;
- then you can clone this repo into that components folder.
- test if this repository was detected, by typing idf.py menuconfig you should see the Micro ROS settings menu.
- Refer the examples to check how to create the pubs/subs and how to start the esp32 side ros-node and place it in your code.

Please let us know if there any specific step you may having difficulties 👍

simonjcarr, 2021-01-10:
@uLipe Thanks for the quick response. The repository is detected and I can set the IP and Port in menuconfig.

The part I am having problems with is how to reference the repository from my project hello_world_main.c

I am new to ROS and IDF so this is all a massive learning curve. I have tried adding the #include's from the example projects.

```
#include <uros_network_interfaces.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <std_msgs/msg/int32.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rmw_uros/options.h>
#include "uxr/client/config.h"
```

But when I try to build the project, I get the following error

```
fatal error: rcl/rcl.h: No such file or directory
 #include "rcl/rcl.h"

```

h-valdes, 2021-02-10:
Hi, 
I am experiencing the same error when I try to call the script `idf.py build`.
In the step `idf.py menuconfig` works everything fine and micro ros configuration is shown in the program.
Do I need to include rcl directly in my CMakelists.txt or should it be automatically include by the component?

Thank you in advance

camrbuss, 2021-04-12:
It seems like a couple people are experiencing a couple different problems, but hopefully this comment will add some clarity to using this repo as a submodule (component) of a esp-idf project.

1. Create a folder for you esp-idf project `mkdir my-uros-edf-proj`
2. Add all needed esp-idf config files (CMakeLists.txt, Makefile, sdkconfig, Kconfig.projbuild, etc.) Examples can be found in [this repos examples](https://github.com/micro-ROS/micro_ros_espidf_component/tree/foxy/examples/int32_publisher)
3. Add this repository as a submodule to a `components` directory in your project folder. 
3. Make sure your "main" .c file has all the juicy rclc code that will publish or subscribe.
4. Make sure to do `idf.py menuconfig` to set the micro-ros agents ip address
5. Build and flash the firmware and resolve any CMake issues `idf.py build` `idf.py flash`
6. Start up your micro-ros agent `ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6`
7. Check `idf.py monitor` for any issues the esp32 may be running into.

VS728, 2021-06-02:
Hello,
I'm new to ros2, micro ros and espidf as well.
I created a project folder as described but have the same problem with my includes like @simonjcarr.
I tried to fix it by modifying the CMakeLists.txt in my main folder.  

idf_component_register(SRCS publisher.c
                        INCLUDE_DIRS "../components/micro_ros_espidf_component/micro_ros_src/src/rcl/rcl/include")

The next error i got was an include in the rcl.h file, so I repeated this step by including all folders from /micro_ros_src/src by hand but got an error from #include <std_msgs/msg/int32.h> in my main file (publisher.c) ...

Do someone have an idea what went wrong? 
In my understanding the components are like plug and play (put them into a components folder next to the main folder and use them) without modifying anything or do I have to edit something in my esp-idf folder or the project folder itself because it looks like the idf can not resolve them while building..
Thanks for your help.

closed 2022-01-28 by pablogs9