micro-ROS/micro_ros_espidf_component #29: wrong freertos include path in esp32_toolchain.cmake.in

BrettRD, 2021-01-15:
Build fails with the latest IDF master
```
$ idf.py --version
ESP-IDF v4.3-dev-2398-g2bfdd036b
```
This component can't find `freertos/FreeRTOSConfig.h` because it's looking in `${idf_path}/components/freertos/xtensa/include` but the latest idf puts it under `${idf_path}/components/freertos/port/xtensa/include`

The following change gets the compilation further:
```
$ git diff
diff --git a/esp32_toolchain.cmake.in b/esp32_toolchain.cmake.in
index 9059159..465e026 100644
--- a/esp32_toolchain.cmake.in
+++ b/esp32_toolchain.cmake.in
@@ -38,7 +38,7 @@ include_directories(
         ${idf_path}/components/vfs/include
         ${idf_path}/components/log/include
         ${idf_path}/components/freertos/include
-        ${idf_path}/components/freertos/xtensa/include
+        ${idf_path}/components/freertos/port/xtensa/include
         ${idf_path}/components/soc/soc/${idf_target}/include
         ${idf_path}/components/wifi_provisioning/include
         ${idf_path}/components/pthread/include
```

After that it demands the empty string as its fastcdr version and fails again, but that's a different issue

pablogs9, 2021-01-15:
Could you please add a PR fixing this for the last IDF version?

Thanks!

pablogs9, 2021-01-15:
Edit: as far as I know v4.2 is latest ESP-IDF stable version right?

closed 2021-01-15 by pablogs9

PaddyCube, 2021-01-26:
I had exactly the same issue and after adding this line to esp32_toolchain.cmake.in it worked
${idf_path}/components/freertos/port/xtensa/include

So it might be a good idea to add this. Also iyou should add a note not to source ROS2 during build time as this causes the issue related to fastcdr.

pablogs9, 2021-01-26:
Which version of the ESP-IDF are you targeting? 

We have a nightly build that build everyday v4.2 and v4.1: https://github.com/micro-ROS/micro_ros_espidf_component/actions?query=workflow%3ANightly

Could you please show us how are you building this repo to get the error?

PaddyCube, 2021-01-26:
I'm running on fthis version: ESP-IDF v4.3-dev-2586-g526f68239 
and I just use 

idf.py build

I cleaned everything up (make -f libmicroros.mk clean) and tried again to show you the result
`ompilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:388: CMakeFiles/microxrcedds_client.dir/src/c/profile/transport/ip/ip_posix.c.obj] Error 1
In file included from /home/ros/esp32/esp-idf/components/esp_common/include/esp_task.h:30,
                 from /home/ros/esp32/esp-idf/components/lwip/port/esp32/include/lwipopts.h:43,
                 from /home/ros/esp32/esp-idf/components/lwip/lwip/src/include/lwip/opt.h:51,
                 from /home/ros/esp32/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:42,
                 from /home/ros/esp32/esp-idf/components/lwip/lwip/src/include/compat/posix/sys/socket.h:33,
                 from /home/ros/esp32/ArdumowerMircoRos/components/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/profile/discovery/transport/udp_transport_datagram_posix_nopoll.c:4:
/home/ros/esp32/esp-idf/components/freertos/include/freertos/FreeRTOS.h:58:10: fatal error: freertos/FreeRTOSConfig.h: No such file or directory
 #include "freertos/FreeRTOSConfig.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:414: CMakeFiles/microxrcedds_client.dir/src/c/profile/discovery/transport/udp_transport_datagram_posix_nopoll.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:76: CMakeFiles/microxrcedds_client.dir/all] Error 2
make[1]: *** [Makefile:130: all] Error 2
---
Failed   <<< microxrcedds_client [6.23s, exited with code 2]
Aborted  <<< rmw [1.71s]
`

After adding the line to esp32_toolchain.make.in, it works

pablogs9, 2021-01-26:
Related: https://github.com/micro-ROS/micro_ros_espidf_component/pull/33