micro-ROS/micro_ros_espidf_component #31: Feature request: allow extra packages by VCS import or git submodules

BrettRD, 2021-01-18:
This follows https://github.com/micro-ROS/micro_ros_espidf_component/pull/30

When setting up an application and tracking it with git, I import micro_ros_espidf_component as a git submodule.
I'd like to track additional application-specific ROS2 packages as either more git submodules, or with an `extra_packages.repos` file.
At the moment, extra packages cannot be tracked with git submodules without a race condition on import (my bad)

It seems like either the `extra_packages` folder or `extra_packages.repos` (or both) should sit outside this component, in the application root.
However, looking for a file in `$(EXTENSIONS_DIR)/../../extra_packages.repos` would prevent the user from being able to install this component IDF-wide.

Is there a makefile variable that can target the application root? or can the application CMakeLists.txt pass it on?

pablogs9, 2021-01-18:
I really don't know, I will check it when I have a moment. Let me know if you find a solution.

BrettRD, 2021-01-20:
I've found a workaround, but it's kinda ugly.

In the project root, create a folder called `extra_ros_packages`, and use the CMakeLists.txt file from the project root to copy it into the micro_ros component at build time:
```
file(COPY ${CMAKE_CURRENT_SOURCE_DIR}/extra_ros_packages DESTINATION ${CMAKE_CURRENT_SOURCE_DIR}/components/micro_ros_espidf_component/extra_packages)
```

This might be viable for `extra_packages.repos` and does work for submodules, but I'd prefer not to have such excessive duplication.

I'll keep looking for a solution based on parameter passing.

donRaphaco, 2021-08-20:
I agree that it would be nice if additional ROS2 packages could be managed outside of this component.
Therefore I created this PR #83 , which makes it possible to have additional ROS2 packages in the `extra_ros_packages` which should be located in the IDF project's root directory.
It also also imports packages from a `extra_packages.repos` file, which should be located inside the `extra_ros_packages` folder.

pablogs9, 2021-08-23:
Closed by https://github.com/micro-ROS/micro_ros_espidf_component/pull/83

closed 2021-08-23 by pablogs9