micro-ROS/micro_ros_espidf_component #34: Cmake version error when trying to run idf.py menuconfig

slahrberg, 2021-01-28:
I installed esp-idf following this guide: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/
I successfully built and flashed the helloWorld example from the guide.

Now I want to use this repo micro_ros_espidf_component.

I tried to use examples/int32_publisher. So I created a folder inside examples/int32_publisher called components and i cloned this repo into the compononents folder.

I run `. $IDF_PATH/export.sh` and `idf.py menuconfig`. This produces an error:

`steffen@steffen-ubuntu:~/esp/int32_publisher$ idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /home/steffen/esp/int32_publisher/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/steffen/esp/int32_publisher"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/steffen/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/steffen/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/steffen/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
CMake Error at /home/steffen/esp/esp-idf/tools/cmake/component.cmake:221 (message):
  CMake Deprecation Warning at /home/steffen/airwave/CMakeLists.txt:1
  (cmake_minimum_required):

    Compatibility with CMake < 2.8.12 will be removed from a future version of
    CMake.

  

    Update the VERSION argument <min> value or use a ...<max> suffix to tell
    CMake that the project does not need compatibility with older versions.

  Call Stack (most recent call first):

    /home/steffen/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:60 (include)
    /home/steffen/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:76 (__component_get_requirements)

  

  

  CMake Error at /home/steffen/airwave/CMakeLists.txt:4 (project):

    project command is not scriptable

  Call Stack (most recent call first):

    /home/steffen/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:60 (include)
    /home/steffen/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:76 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  /home/steffen/esp/esp-idf/tools/cmake/build.cmake:405 (__component_get_requirements)
  /home/steffen/esp/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/steffen/esp/int32_publisher/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
`

It seems like my cmake is outdated?
When i run `cmake --version` i get:

`steffen@steffen-ubuntu:~/esp/int32_publisher$ cmake --version
cmake version 3.19.3
`

How can I solve this? Running `idf.py menuconfig` when I follow the guide from espressif works fine.

BrettRD, 2021-01-29:
I'm running `cmake version 3.16.3`, so your issue is more likely to be project layout or component dependencies.

The errors you have suggest that the IDF is forwarding an error from
`CMake Error at /home/steffen/airwave/CMakeLists.txt:4`
and is suggesting you look at the version requirements in
`CMake Deprecation Warning at /home/steffen/airwave/CMakeLists.txt:1`
I have no idea what `airwave` is or why it's getting linked to your micro_ros build

pablogs9, 2021-01-29:
Could you provide steps for replicating this issue?

I just have checked this steps and everything works:

```bash
git clone https://github.com/micro-ROS/micro_ros_espidf_component
cd micro_ros_espidf_component
. $IDF_PATH/export.sh
cd examples/int32_publisher
idf.py menuconfig
# Set your micro-ROS configuration and WiFi credentials under micro-ROS Settings
idf.py build 
idf.py flash 
idf.py monitor 
```

slahrberg, 2021-02-06:
Hello, thank you for the responses. I have not tried to solve this problem any longer. I have followed this guide (https://micro-ros.github.io/docs/tutorials/core/first_application_rtos/freertos/) to create micro-ROS applications on my esp32 which works fine so far.

closed 2021-02-06 by slahrberg