micro-ROS/micro_ros_espidf_component #36: Error: rcl not set

giuseppelamarca, 2021-01-31:
Hi I'm facing a problem, I'm running ubuntu18.04 with dashing so to set up everything I decided to use the docker.
Inside the components folder I copied the repository and later run the command 

`docker run -i --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8889 -v6`
to start the agent and later the command to build and flash the programm 

`docker run -it --rm --user espidf --volume="/etc/timezone:/etc/timezone:ro" -v  $(pwd):/micro_ros_espidf_component -v  /dev:/dev --privileged --workdir /micro_ros_espidf_component microros/esp-idf-microros:latest /bin/bash  -c "cd examples/int32_publisher; idf.py menuconfig build flash monitor"`

Now during the monitor I can see that the board is connected to my wifi but I get this error. 
[image]
The ip of the agent I gave the ip of the machine where is running the docker of the agent
Do you have any idea what can be? Thanks for your time

pablogs9, 2021-02-01:
The default configuration of the micro-ROS components for IDF sets the micro-ROS port to 8888, but you are launching the agent on 8889.

I have just replicated all your commands and if you set the port correctly it works.

Please close if this resolves your issue.

giuseppelamarca, 2021-02-01:
Thank's a lot for the help. My bad I changed the port but the problem persisted. Apparently, it was a problem with the firewall ( I had this problem also with ros2 running nodes on different computers). Thanks again for the help

closed 2021-02-01 by pablogs9