micro-ROS/micro_ros_espidf_component #44: USB interface

shecker, 2021-03-08:
Hi
Thanks for this great library. 
Is it possible to use USB to publish ROS messages instead of Wifi or UART pins?
We are using an ESP32 connected to a Linux machine via USB and would like to send messages across that interface.
I was looking at the custom_publisher example but I'm a bit unsure how to proceed in order to use USB as far as I understood it's used for UART communication via arduino pins.
Any info would be greatly appreciated, thanks!
Thanks

pablogs9, 2021-03-08:
For sure, usually in ESP32 boards USB interface is tunneling a UART port. You should check which of your ESP32 UART ports ends in the USB interface. Once you have that, just follow [this](https://github.com/micro-ROS/micro_ros_espidf_component#using-serial-transport).

You also have an example [here](https://github.com/micro-ROS/micro_ros_espidf_component/tree/foxy/examples/int32_publisher_custom_transport). 

Let me know if you have any other problems.

shecker, 2021-03-08:
Thanks for the quick response!
Yes it should be UART port 0 as per the schematic in the top right corner: https://cdn.sparkfun.com/assets/learn_tutorials/5/0/7/esp32-thing-schematic.pdf

I have thus changed the interface to use UART port 0 (set manually because for some reason it doesn't propagate the values I set in the menuconfig to the actual C code)
```
size_t uart_port = 0;
rmw_uros_set_custom_transport(
        true,
        (void *) &uart_port,
        esp32_serial_open,
        esp32_serial_close,
        esp32_serial_write,
        esp32_serial_read
);
```
However when I try to launch with the docker container with your command line:
`docker run -it --rm -d /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/ttyUSB0 -v6
`
I get the error 
`docker: invalid reference format.
`
I'm able to launch the docker with just a simple interactive session but then I'm unsure how to start the communication. 
I also have ros2 foxy installed native but I'm unsure how that communication is supposed to be established via UART/USB. I'm guessing here is where the micro-ROS Agent comes into play, but as said we are unable to launch this in the docker. Any pointers would be greatly appreciated!
Thanks

pablogs9, 2021-03-09:
Have you rebuilt the library after changing `"-DRMW_UXRCE_TRANSPORT=custom",`in the colcon.meta?

In order to clean all and rebuild from scratch you can do a `make -f libmicroros.mk clean` in the component directory.

shecker, 2021-03-09:
@pablogs9  yes I changed the ` colcon.meta ` and also cleaned the library including removing `libmicroros.a`. If I use the serial monitor I get the following error:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7000
load:0x40078000,len:14336
load:0x40080400,len:3672
0x40080400: _init at ??:?

entry 0x40080678
I (27) boot: ESP-IDF v4.3-beta1-24-g0bbc721a6 2nd stage bootloader
I (27) boot: compile time 11:28:05
I (27) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 16MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0c524h ( 50468) map
I (129) esp_image: segment 1: paddr=0001c54c vaddr=3ffb0000 size=02e98h ( 11928) load
I (133) esp_image: segment 2: paddr=0001f3ec vaddr=40080000 size=00c2ch (  3116) load
I (136) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=26a80h (158336) map
I (199) esp_image: segment 4: paddr=00046aa8 vaddr=40080c2c size=0b908h ( 47368) load
I (225) boot: Loaded app from partition at offset 0x10000
I (225) boot: Disabling RNG early entropy source...
I (236) cpu_start: Pro cpu up.
I (237) cpu_start: Starting app cpu, entry point is 0x40081128
0x40081128: call_start_cpu1 at /home/nico/Software/esp/esp-idf/components/esp_system/port/cpu_start.c:143

I (0) cpu_start: App cpu up.
I (251) cpu_start: Pro cpu start user code
I (251) cpu_start: cpu freq: 160000000
I (251) cpu_start: Application information:
I (255) cpu_start: Project name:     hello-world
I (261) cpu_start: App version:      1
I (265) cpu_start: Compile time:     Mar  9 2021 11:28:03
I (271) cpu_start: ELF file SHA256:  c2e434f9915eb9a6...
I (277) cpu_start: ESP-IDF:          v4.3-beta1-24-g0bbc721a6
I (284) heap_init: Initializing. RAM available for dynamic allocation:
I (291) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (297) heap_init: At 3FFB8948 len 000276B8 (157 KiB): DRAM
I (303) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (309) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (316) heap_init: At 4008C534 len 00013ACC (78 KiB): IRAM
I (323) spi_flash: detected chip: gd
I (326) spi_flash: flash io: dio
I (332) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Failed status on line 78: 11. Aborting.
```

Using the docker command I get the same output as posted in my previous comment. 
Do you have any idea what is going on?
Thanks so much for your help.

pablogs9, 2021-03-09:
Ok, I have been testing the app and I found a couple of problems. I have fixed them in https://github.com/micro-ROS/micro_ros_espidf_component/pull/46.

Please use a clean environment to ensure that you have the fix, the procedure should be:

``` bash
git clone https://github.com/micro-ROS/micro_ros_espidf_component
cd micro_ros_espidf_component

# Modify the meta
sed -i 's/DRMW_UXRCE_TRANSPORT=udp/DRMW_UXRCE_TRANSPORT=custom/' colcon.meta

cd examples/int32_publisher_custom_transport

idf.py build flash 

# Do not monitor because it will occupy the serial port that the micro-ROS Agent needs

docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/serial/by-id/[YOUR DEV] -v6
```
Let me know if this works

shecker, 2021-03-09:
Hi @pablogs9 thanks for the quick reply. 
I have tried running everything in a clean environment but unfortunately no luck. Here are my steps to reproduce, please do let me know if I have an error somewhere:

```
mkdir esp_clean && cd esp_clean
git clone https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout release/v4.3 # this is due to compatibility with arduino esp32 component
./install.sh
. ./export.sh
cd .. && mkdir micro-ros && cd micro-ros 
mkdir components && cd components
git clone https://github.com/micro-ROS/micro_ros_espidf_component.git
cd micro_ros_espidf_component
sed -i 's/DRMW_UXRCE_TRANSPORT=udp/DRMW_UXRCE_TRANSPORT=custom/' colcon.meta
cd examples/int32_publisher_custom_transport
idf.py build flash 
```

Flashing of the arduino is successful.
I follow with your command 

`docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:foxy serial --dev /dev/serial/by-id/usb-Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_020A8AA5-if00-port0 -v6`

but am stuck with the same logger_info as the image I shared above, but nothing else happens. I have tried logging into the docker and running `ros2 node list` or `ros2 topic list` but no micro-ros shows up. 
I'm using a ESP32 Thing WROOM from Sparkfun connected via USB to an x86 Ubuntu 20.04 running ROS2 foxy.

pablogs9, 2021-03-10:
In order to discard a problem in your serial port, could you open a serial terminal and check if your board is writing something in the serial port?

Something like:

```bash
miniterm.py [your serial port] 115200
```

or

```bash
minicom -D [your serial port] -b 115200
```

shecker, 2021-03-10:
@pablogs9 here is the output of miniterm. In the beginning it seems to print something fine (5-7 seconds) and then after a while prints the uart driver error
[image]

pablogs9, 2021-03-11:
Ok if you open the agent and reset the board it should connect and start publishing.

Then do a `ros2 topic echo [your topic name]` and how long is it publishing?

shecker, 2021-03-11:
It only publishes around 2 seconds before automatically resetting. I've tried both physical reset button and `ard-reset-arduino`

pablogs9, 2021-03-11:
Two ideas:
- How much stack does have your application configured?
- If the client is resetting you are having a hard fault somewhere. Maybe it is a good idea to configure micro-ROS to use another serial port and use a USB-to-serial cable, and let the onboard USB-Serial print the log of the MCU. Can you try this and paste here what is the MCU logging system printing while using micro-ROS in a UART on some pins?

shecker, 2021-03-11:
> How much stack does have your application configured?

We set the micro ROS stack to 20000

> Maybe it is a good idea to configure micro-ROS to use another serial port and use a USB-to-serial cable, and let the onboard USB-Serial print the log of the MCU. 

I tried using your WiFi example instead, again it can compile and flash. But I cannot get an output on the micro-ros docker. The esp32 board is pingable and has a valid IP address (192.168.0.115). I set that in the menuconfig but unfortunately no ros2 output.

pablogs9, 2021-03-12:
In the micro-ROS menuconfig you should set the **micro-ROS Agent IP**, this is your PC IP, not the IP of the micro-ROS client running in the ESP32. Can you try that?

shecker, 2021-03-12:
Okay, so UDP configuration works very well after changing the IP. No resets of the board, automatic connection, very nice. 
Still no luck with the serial tho, so I'll have to debug further. It simply resets after a short while, sometimes making it to 5 before resetting back to 0.

pablogs9, 2021-03-12:
I think that a good solution is to separate the debug and micro-ROS serial ports in order to check what is happening.

shecker, 2021-05-16:
I solved this by ensuring a proper install of the micro ros agent and correct esp-idf versions that are compatible.

closed 2021-05-16 by shecker