micro-ROS/micro_ros_espidf_component #45: Problem building with make

Joelmaca9, 2021-03-09:
## Issue template

- Hardware description: ESP32 WROVER-E
- RTOS: FREERTOS
- Installation type: micro_ros_espidf_component
- Version or commit hash: foxy

#### Steps to reproduce the issue
Create a makefile under int32_publisher
Build int32_publisher example with make

#### Expected behavior
Build the project as with idf.py

#### Actual behavior
Make gives the following error:
/SDK/esp-idf/components/micro_ros/examples/int32_publisher/main/main.c:10:10: fatal error: uros_network_interfaces.h: No such file or directory
 #include <uros_network_interfaces.h>

#### Aditional info
The Makefile that I have created is like the one of the SDK examples:

`
PROJECT_NAME` := int32_publisher

include `$(IDF_PATH)/make/project.mk`

Also wen I open menuconfig with `make menuconfig` I see the example-app settings but not the micro_ros configuration

pablogs9, 2021-03-09:
I guess that you are trying to build an IDF project using another tool... I'm not sure if this is a micro-ROS-related issue.

Why don't you use the IDF tools for building the proyect?

Joelmaca9, 2021-03-09:
Because in my project I'm using Arduino as a component (as defined here https://github.com/espressif/arduino-esp32) I need to build using make, until now I have had no problems.

Thanks for your fast response!

Joelmaca9, 2021-03-09:
All right, I have make a little try with idf.py and now I am able to build and flash the example and my arduino libraries but I get the next error:

[ERROR] [0000000015.071651000] [rclc]: [rclc_init] Error in rcl_init: error not set

PD: I have seen your video, to include the arduino repository you must clone the branch that matches the idf version (4.2)

pablogs9, 2021-03-09:
That's the branch I checked out. Plese provide some instructions to replicate the error.

closed 2021-03-31 by pablogs9