micro-ROS/micro_ros_espidf_component #47: running int32_subscriber example and get an error 'usleep'/'fseek'

wegunterjrFIrefly, 2021-03-22:
## Issue template

- Hardware description: esp32 
- RTOS: freertos
- Installation type: micro_ros_setup build_firmware.sh
- Version or commit hash: foxy

#### Steps to reproduce the issue

running the tutorial for FreeRTOS on esp32
https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/

#### Expected behavior
Expecting code to compile, and then be able to run the flash step, but it fails.

#### Actual behavior
Build fails, and I get the following error:
error: implicit declaration of function 'usleep'; did you mean 'fseek'

#### Additional information

pablogs9, 2021-03-23:
This tutorial is intended for Olimex STM32-E407, so it will not work with ESP32. If you want to test micro-ROS in your ESP32 the easiest way is to use this IDF component. Assuming that you have IDF-ESP installed, just do:

Dependencies:

```bash
sudo sh -c 'echo "deb [arch=amd64,arm64] http://repo.ros2.org/ubuntu/main `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'
sudo curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -
sudo apt update
sudo apt install python3-colcon-common-extensions

. $IDF_PATH/export.sh
pip3 install catkin_pkg lark-parser empy
```

In order to test a int32_publisher example:

```bash
. $IDF_PATH/export.sh
git clone https://github.com/micro-ROS/micro_ros_espidf_component
cd examples/int32_publisher
idf.py menuconfig
# Set your micro-ROS configuration and WiFi credentials under micro-ROS Settings
idf.py build 
idf.py flash 
idf.py monitor 
```

closed 2021-03-23 by pablogs9

ralph-lange, 2021-03-23:
Last fall, I've successfully gone through the tutorial with an ESP32. The main change is to replace the line
```
  ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407
```
with
```
  ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32
```
Note that some ESP32 require to press the boot button for flashing.

ralph-lange, 2021-03-23:
I just tried the tutorial (ping_pong app) in a fresh workspace with an ESP32 NodeMCU without any issues.

wegunterjrFIrefly, 2021-03-23:
@ralph-lange   nice!   Can you try the subscribe example?