micro-ROS/micro_ros_espidf_component #50: Micro XRCE-DDS session error using esp32 wifi transport.

LegoYoda112, 2021-03-28:
Hi, just started working with micro ros today. This issue could likely be me missing something silly, but I have run out of things to check. 

- Hardware description: MakerHawk ESP32 dev board
- RTOS: FreeRTOS
- Installation type: micro_ros_setup, ubuntu 20.04.2, vmware virtual machine
- Version: foxy

#### Steps to reproduce the issue
Following code from https://micro.ros.org//blog/2020/08/27/esp32/ and the setup tutorials: https://micro.ros.org//docs/tutorials/core/first_application_linux/.

#### Expected behavior
Following steps in example commands: Client should connect with agent and messages should be published to the ROS2 topic.

#### Actual behavior
- Flash esp32
- Start agent
```
$ ros2 run micro_ros_agent micro_ros_agent udp4 -p 8888
[1616886503.942609] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1616886503.942856] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1616886542.788064] info     | Root.cpp           | create_client            | create                 | client_key: 0x43F8851C, session_id: 0x81
```

- Reboot esp32
- Agent receives connection and session is established

_esp32 serial (through putty)_
```
micro-ROS transport: connected using UDP mode, ip: '192.168.1.71', port: '8888'
```

_Agent terminal:_
```
[1616886542.788128] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1140360476, address: 192.168.1.73:51198
```

- **Error thrown on esp32 and it shuts down.**
```
[ERROR] [0000000011.566660000] [rclc]: [rclc_publisher_init_default] Error in rcl_publisher_init: Issues running micro XRCE-DDS session, at /home/[username]/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/utils.c:42, at /home/[username]/microros_ws/firmware/mcu_ws/uros/rcl/rcl/src/rcl/publisher.c:188

Failed status on line 48: 1. Aborting.
```

For reference, line 43-48 on `app.c` are:
``` 
// create publisher
	RCCHECK(rclc_publisher_init_default(
		&publisher,
		&node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
		"freertos_int32_publisher"));
```

The error on the esp32 appears to be coming from the RCCHECK, and implies that the message type is not supported, or the publisher is not able to be created for some other reason.

Lines 36 - 44 of `utils.c`
```
// This only handles one request at time
        uint8_t status;
        if (!uxr_run_session_until_all_status(&context->session,
                RMW_UXRCE_ENTITY_CREATION_DESTROY_TIMEOUT,
                &requests, &status, 1))
        {
            RMW_SET_ERROR_MSG("Issues running micro XRCE-DDS session");
            return false;
        }
```

Lines 187 - 188 of `publisher.c`
```
RCL_CHECK_FOR_NULL_WITH_MSG(
    publisher->impl->rmw_handle, rmw_get_error_string().str, goto fail);
```

#### Additional information
I have tested serial transport which works fine; the topic is published as expected.
I have also tested as the ping_pong demo app, which works as expected using the serial transport, but not over wifi.

This leads me to believe the error is with micro XRCE-DDS over wifi, however investigating the stack traces didn't give me any more clues.

pablogs9, 2021-03-29:
Hello @LegoYoda112,

- Can you run the agent with the flag `-v6` and paste here the output?
- Also, how much stack are you assigning to the micro-ROS task in freeRTOS?

@Acuadros95 you test this scenario last week right? Could you replicate this (ESP32+ micro_ros_setup) and provide here instructions?

Acuadros95, 2021-03-29:
Hi @LegoYoda112,

I have just tested this and it is working fine over wifi, could the problem be related with your network?

LegoYoda112, 2021-03-29:
Hello, thank you so much for you help!

Here's what I tested:
- Disconnected my ethernet connection
- Ran using the docker command and -v6

This appeared to work initially, but after some more testing, looks like it fails about half of the time. (once it gets past the initial setup, it appears to be stable). 

I also occasionally get this error in addition to the other:
``` [WARN] [0000000047.759640000] [rclc]: [rclc_node_init_default] Warning in rclc_node_init_with_options: error not set ```

Here is the full output from the agent:
https://gist.github.com/LegoYoda112/18bc4a2fd302359c83e441afc2647a5f

My end goal is to run it over ethernet, so if this is likely an issue with my wifi connection then I am ok dealing with intermittent faults while testing.

Acuadros95, 2021-03-30:
I would bet on the wifi connection, anyway I am closing this for now as it is enough for your testing.

Let us know if the problem persist using Ethernet!

closed 2021-03-30 by Acuadros95