micro-ROS/micro_ros_espidf_component #53: would you mind providing the answer for this?  I followed the example here and am still getting the same error you are getting.

wegunterjrFIrefly, 2021-03-30:
would you mind providing the answer for this?  I followed the example here and am still getting the same error you are getting.
I am in the project folder, and have included the micro_ros_espidf_component folder
```
Project:
- components
- -micro_ros_espidf_component
- CMakelists.txt
- main.c
```

_Originally posted by @wegunterjrFIrefly in https://github.com/micro-ROS/micro_ros_espidf_component/issues/27#issuecomment-810404986_

pablogs9, 2021-03-31:
I have replicated this with a simple project and it works:

https://asciinema.org/a/G39IxQZkzipXSxrnbqepXh6iF

wegunterjrFIrefly, 2021-03-31:
What version of esp-idf are you downloading? release4.2, release v4.3?
Do you delete the .espressif folder and run install.sh again when you switch versions of esp-idf?

pablogs9, 2021-03-31:
In this video I'm on v4.3 but as you can see in our CI we support v4.1, v4.2 and v4.3: https://github.com/micro-ROS/micro_ros_espidf_component/actions/runs/703714929

wegunterjrFIrefly, 2021-03-31:
this was helpful.  Not sure what I was doing wrong, but I am sure it was a combo of things.
If I follow the example on github and in the video you shared
- adding the components folder  `mkdir components`
- and cloning the micro_ros_esp_idf  
-- `cd components`
-- clone the micro_ros_esp_idf
- cd to main folder `cd ..`
- source esp-idf (make sure you have added the `pip3 install catkin_pkg lark-parser empy`
- idf.py

I find that if you have been changing versions of esp-idf, it is necessary to delete the `~/.espressif` folder.

wegunterjrFIrefly, 2021-11-07:
hate to reopen an issue, but planning to use microros and having similar issues:

- idf.py version ESP-IDF v4.2.2-383-g125cf671e7
- microros - foxy c0c003dc1d3c46410603b7879e819f94b25bf529 (HEAD -> foxy, origin/foxy)
- pwd components/micro_ros_espidf_component/examples/int32_publisher
I did ensure all required software and the installation was followed...including the colcon and empy

It is failing at `/main/main.c:11:10: fatal error: rcl/rcl.h: No such file or directory`

wegunterjrFIrefly, 2021-11-07:
ah..hmmm... The solution is that it wasn't completely downloading the micro_ros_src and other libraries which contain things like rcl.h   I deleted the entire micro_ros_espidf_component and tried again with the docker container and success.

closed 2021-11-07 by wegunterjrFIrefly

MiguelMartinez-HAW, 2022-08-24:
> I have replicated this with a simple project and it works:
> 
> https://asciinema.org/a/G39IxQZkzipXSxrnbqepXh6iF

In this video, you already start with a folder named int32_publisher. Could you show the steps you followed before this video started, please? I assumed the previous steps but I must have done something wrong because I am getting errors.

pablogs9, 2022-08-24:
The folder is this one: https://github.com/micro-ROS/micro_ros_espidf_component/tree/humble/examples/int32_publisher

MiguelMartinez-HAW, 2022-08-24:
> The folder is this one: https://github.com/micro-ROS/micro_ros_espidf_component/tree/humble/examples/int32_publisher

Thank Pablo, for your quick reply. So I downloaded the folder `int32_publisher` into my `esp-idf` folder. Then I followed the instructions exactly as in the video but when I run `idf.py` build it throws `command not found`. Then I go back to the `esp-idf` folder and run `./install.sh` and `. ./export.sh` and `pip3 install catkin_pkg lark-parser empy colcon-common-extensions` . Then I go to the `int32_publisher` folder again and run` idf.py set-target esp32` or `idf.py menuconfig` or  `idf.py build` and they all throw errors. I must be doing something wrong but I don't know what.

MiguelMartinez-HAW, 2022-08-30:
> . ./export.sh
I tried repeating the steps a few times, also with other examples but this is the error I get. 
[image]

What I don't know if I'm doing right is that I copy the example folder, in this case `int32_publisher_custom_transport`, from the repository. Then I follow the video, which clones the repository in the components folder which also copies again the examples.

Am I doing something wrong?

Pankaj-s-dev, 2024-01-11:
I am able to build these examples on Ubuntu 22.04, i am getting error while building in Ubuntu 20..