micro-ROS/micro_ros_espidf_component #54: int32_publisher example, failed status on line

wegunterjrFIrefly, 2021-04-01:
## Issue template

- Hardware description: esp32
- RTOS: FREE RTOS
- Installation type: micro_ros_espidf_component (using default examples/int32_publisher) - checked out latest
- Version or commit hash: foxy, espidf v4.2, micro_ros_espidf_component 6e172d5aa2e23872c30d07c4238f3e36773468b4

#### Steps to reproduce the issue
running the example int32_publisher
- flash successful
- idf.py monitor
- shows successful connection to micro_ros_agent running on linux
- immediately gives me:
- 

#### Expected behavior
- successful connection to the agent
- send message every second
- 
#### Actual behavior
- shows successful connection to micro_ros_agent running on linux
- immediately gives me:
`
  [INFO] [0000000010.052555000] []: Created a timer with period 1000 ms.
  
  Failed status on line 29: 1. Continuing.
`
#### Additional information

vinnnyr, 2021-04-01:
I can corroborate that when following the tutorials for `examples/int32_publisher` I am getting a bootloop triggered seemingly by the same issue:
```
micro-ROS transport: connected using UDP mode, ip: '192.168.0.219', port: '8888'
[INFO] [0000000013.118672000] []: Created a timer with period 1000 ms.
***ERROR*** A stack overflow in task uros_task has been detected.
abort() was called at PC 0x40087440 on core 0
ELF file SHA256: 736b4e279db03d83
Backtrace: 0x40087085:0x3ffc9600 0x40087429:0x3ffc9620 0x40087440:0x3ffc9640 0x40088b2f:0x3ffc9660 0x40087d88:0x3ffc9680 0x40087d3e:0xffffffff |<-CORRUPTED
Rebooting...
```

pablogs9, 2021-04-05:
IMO your issues are different:
- @vinnnyr I guess that you should increase the stack assigned to the micro-ROS task. Try to increase [this](https://github.com/micro-ROS/micro_ros_espidf_component/blob/6e172d5aa2e23872c30d07c4238f3e36773468b4/examples/int32_publisher/main/main.c#L99) and if everything works please consider opening a PR update the value [here](https://github.com/micro-ROS/micro_ros_espidf_component/blob/6e172d5aa2e23872c30d07c4238f3e36773468b4/examples/int32_publisher/main/Kconfig.projbuild#L5)

- @wegunterjrFIrefly could you please share the output of the micro-ROS agent using the flag `-v6`?

wegunterjrFIrefly, 2021-04-13:
k...so, here is the int32_publisher example, and the output from the esp32 and the agent.
Interesting, is that idf.py monitor outputs:
```
[INFO] [0000000002.654506000] []: Created a timer with period 1000 ms.

Failed status on line 42: 1. Continuing.
Failed status on line 42: 1. Continuing.
Failed status on line 42: 1. Continuing.
Failed status on line 42: 1. Continuing.
Failed status on line 42: 1. Continuing.
Failed status on line 42: 1. Continuing.
```

```
[1618283476.003305] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x3375675B, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 49 00 49 00 80
[1618283476.003318] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x3375675B, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 4A 00 00 00 80
[1618283476.003410] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x3375675B, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 4A 00 00 00 80
[1618283476.003452] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x3375675B, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 4A 00 00 00 80
[1618283476.003476] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x3375675B, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 4A 00 00 00 80
````

pablogs9, 2021-04-13:
I have just tested the example and it works: https://asciinema.org/a/gK7AlLeHnJSON3sI3Ab7wZYZ3

Can you check your network configuration?
How are you opening the micro-ROS agent?

wegunterjrFIrefly, 2021-04-13:
i have tried on a few networks, but right now i am using my phone hotspot for wifi.   
I see the device connect to the network, but it is failing again.
I am opening the micro-ROS agent like this:
```
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6

```

wegunterjrFIrefly, 2021-04-15:
it's not dying or aborting, just continuing. So, i guess that works for now. Unless you have some additional thoughts?
I do not appear to be dropping any messages.

pablogs9, 2021-04-15:
But did you solve the issue?

wegunterjr, 2021-04-15:
I am not sure what to check.  I am not even sure what the issue is.   I connect to the network and then to the agent.   I get messages sent that I can then echo with ros2 topic echo....The esp32 monitor shows `dropping`...Is that okay?

wegunterjrFIrefly, 2021-11-09:
I find that if I use the galactic docker image for the agent that it works (it has the continuing, but it doesn't fail)
`RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));`

closed 2022-01-28 by pablogs9