micro-ROS/micro_ros_espidf_component #58: any thoughts on static ip?

wegunterjrFIrefly, 2021-04-15:
## Issue template

    Hardware description: esp32
    RTOS: FREE RTOS
    Installation type: micro_ros_espidf_component (using default examples/int32_publisher) - checked out latest
    Version or commit hash: foxy, espidf v4.2, micro_ros_espidf_component latest

#### Steps to reproduce the issue
I am adding a wifi_init_sta to set the static ip address.

#### Expected behavior
Expect the wif station to start and connect using a static ip.

#### Actual behavior
I am getting a panic boot loop on the line in my main that calls the wifi_init_sta in my code.

#### Additional information

wegunterjr, 2021-04-15:
I could add another uros header file and send in a flag for DHCP or Static IP, and send the CONFIG's for IP addresses.  In there I could disable the DHCP prior to starting the WIFI STA.  
Thoughts?

pablogs9, 2021-04-15:
Closing since not related to micro-ROS. Maybe https://github.com/espressif/esp-idf or https://docs.espressif.com/projects/esp-idf/en/latest/esp32/ is a better place.

closed 2021-04-15 by pablogs9

wegunterjr, 2021-04-15:
This is related to miCro ros because of how you setup the library to enable the wifi library.  It requires changed to microros.

pablogs9, 2021-04-15:
Any tip on that @uLipe?

uLipe, 2021-04-15:
Since the IDF supports both DHCP or static IP through its API, I think we could add to the Kconfig options to let the user select between static or the dynamic IP segment of the selected netif and derive the microROS IP settings derived from that.

Since the netif code is now isolated from microROS code, it should be straightforward to implement. 

What do you think @pablogs9

wegunterjr, 2021-04-16:
I don't mind adding this... any suggested approach for adding the menu items? or passing the CONFIGs down to the network interace.c?