micro-ROS/micro_ros_espidf_component #60: Build fails if ROS 2 setup script has been sourced

RussellJoyce, 2021-05-06:
- Hardware description: ESP32
- RTOS: FreeRTOS / ESP-IDF v4.2.1
- Installation type: micro_ros_espidf_component
- Version or commit hash: foxy

#### Steps to reproduce the issue

Building the micro-ROS component after sourcing the ROS 2 setup scripts into the environment causes the build to fail while compiling `builtin_interfaces`.
Running `idf.py build` for a second time then fails with a `fatal error: rcl/rcl.h: No such file or directory` error.

E.g. the following fails during build, but will work fine if the first line is excluded:

```
. /opt/ros/foxy/setup.zsh
export IDF_PATH=/home/russell/esp/esp-idf-v4.2.1
. $IDF_PATH/export.sh
git clone https://github.com/micro-ROS/micro_ros_espidf_component.git
cd micro_ros_espidf_component/examples/int32_publisher
idf.py set-target esp32
idf.py build
```

#### Additional information

This has briefly been covered previously in #1, and I suspect it could have also been the cause of the problems described in #27 and #53 (although those builds might have just failed for some other reason).

While somehow changing the build scripts to work while the ROS 2 environment is present would obviously be best, for now just mentioning explicitly in the README that the environment should be clean and the ROS setup script should *not* be sourced before running the build would be good.

I'm sure I'm not the only person who sources the ROS scripts in their `.zshrc` file (or equivalent), or might assume that they should be present when working with anything ROS related, which sometimes makes it more of an explicit task to not source the scripts, rather than the other way around.

pablogs9, 2021-05-07:
Cool, I understand. IMO two actions can be taken:
1. Add a line in the readme warning about that
2. Is there any way of checking if the ROS 2 environment is enabled? Maybe we can make a check [here](https://github.com/micro-ROS/micro_ros_espidf_component/blob/e174a69371ae8f7bcb6145fc6847f6f456450074/libmicroros.mk#L78)

Would you mind making a contribution PR for this?

Thanks a lot!

RussellJoyce, 2021-05-10:
Thanks for your reply. 

I agree - both of those seem like reasonable actions, but I guess the second would be better.
I'll take a look at trying to implement it when I get a chance.