micro-ROS/micro_ros_espidf_component #70: Failed on compile 'FAILED: libmicroros-prefix...

wegunterjr, 2021-06-09:
## Issue template

- Hardware description: esp32s2
- RTOS:FreeRTOS
- Installation type: idf.py build 
- Version or commit hash:  foxy

#### Steps to reproduce the issue
Clone micro_ros_espidf_component into components folder
idf.py menuconfig - set MicroRos menu parameters
idf.py build

#### Expected behavior
Successful build so I can flash

#### Actual behavior

FAILED: libmicroros-prefix/src/libmicroros_project-stamp/libmicroros_project-build ../components/micro_ros_espidf_component/libmicroros.a

#### Additional information

pablogs9, 2021-06-09:
Which version of ESP-IDF are you using?

wegunterjr, 2021-06-16:
4.2, but I think the problem was that I was doing this on windows os.  I didn't even think about it til well into the day that I wasn't on linux OS, and didn't have make tools installed... shmaybe?

closed 2021-06-16 by wegunterjr