micro-ROS/micro_ros_espidf_component #74: issues getting to node init

wegunterjr, 2021-06-18:
## Issue template

- Hardware description: esp32
- RTOS: freertos
- Installation type: components module, esp-idf
- Version or commit hash: 
- -- micro_ros_agent, foxy, rmw=fast_rtps
- -- micro_ros_esp_idf_component: 76b97c
- -- esp-idf v4.2: c9cf7b, ESP-IDF v4.2-408-gc9cf7bcb0

#### Steps to reproduce the issue

- source ros2 and microros_ws
- idf.py monitor

#### Expected behavior
Expected to connect to agent and publish message
#### Actual behavior
Failing on line node init with return value of 1:
- 	// Create node.
	rcl_node_t node ;
	RCCHECK(rclc_node_init_default(&node, "esp32_int32_publisher", "", &support));

- micro_ros_agent shows initial connection in log (-v6),
#### Additional information
Connected to wifi network, no issues pinging or connecting to each device

pablogs9, 2021-06-21:
Can you share the output of the agent with the `-v6` flag?
Also, can you share the output of the ESP32 given with `idf.py monitor`?

frett27, 2021-12-25:
i got the same configuration (esp32, with the micro_ros_espidf_component, and a fresh source compiled microros agent on foxy.
the agent running
the first microros connection, stopped the agent ?

here is the logs in -v6 

```
[nix-shell:~/ros-nix/dev-ws-ros2]$ ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
[1640456550.953239] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1640456550.953363] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1640456556.823682] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 7A D8 DD 56 81 00 FC 01
[1640456556.823763] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 7A D8 DD 56 81 00 FC 01
[1640456556.823888] info     | Root.cpp           | create_client            | create                 | client_key: 0x7AD8DD56, session_id: 0x81
[1640456556.823956] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x7AD8DD56, address: 192.168.1.23:37062
[1640456556.823994] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x7AD8DD56, address: 192.168.1.23:37062
[1640456556.824104] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7AD8DD56, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1640456556.824174] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x7AD8DD56, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1640456556.833159] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7AD8DD56, len: 40, data: 
0000: 81 80 00 00 01 07 20 00 00 0A 00 01 01 03 00 00 12 00 00 00 00 01 00 00 0A 00 00 00 61 72 6D 5F
0020: 72 6F 62 6F 74 00 00 00
[1640456556.835062] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7AD8DD56, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1640456556.835094] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7AD8DD56, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1640456556.837678] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7AD8DD56, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1640456556.837713] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x7AD8DD56, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80

```

pablogs9, 2022-01-05:
If you use micro-ROS agent docker for galactic it works but with foxy no? have I understood correctly?

pablogs9, 2022-01-28:
Any update on this?

frett27, 2022-01-28:
i don't know for the original author, i switched to docker, for the microros agent, works fine  on ubuntu. Will give it a try on galactic

closed 2022-01-31 by pablogs9