micro-ROS/micro_ros_espidf_component #81: UCLIENT_PROFILE_MULTITHREAD breaks build

BrettRD, 2021-08-10:
## Issue template

- Hardware description: ESP-WROOM-32 
- RTOS: freertos
- Installation type: idf build
- Version or commit hash: foxy latest

#### Steps to reproduce the issue
turn on middleware thread-safety
```
colcon.meta:
         "microxrcedds_client": {
             "cmake-args": [
+                "-DUCLIENT_PROFILE_MULTITHREAD=ON",
```

#### Expected behavior
build without errors

#### Actual behavior
errors indicating microxrcedds_client cannot properly identify the platform

`typedef struct uxrMutex` is declared empty, and the string `PLATFORM_NAME_FREERTOS` is not found anywhere in the build files except microxrcedds_client source.

pablogs9, 2021-08-10:
Check [this](https://github.com/micro-ROS/micro_ros_espidf_component/pull/82).

Let me know if this solves the issue.

BrettRD, 2021-08-16:
This solved my issue, thanks for the example code!