micro-ROS/micro_ros_espidf_component #90: Slow publishing rate since update

timongentzsch, 2021-09-16:
## Issue template

- Hardware description: ESP32 Wroom
- RTOS: FreeRTOS
- Installation type: micro_ros_espidf_component
- Version or commit hash: foxy (latest)
esp-idf: release/v4.3

Since upgrading to the latest micro_ros_espidf_component we are seeing a large performance drop on the ros publishing side. 
We previously were able to publish sensor data at around 150hz (~32byte message at 1'000'000 baud). However now since upgrading, we are only seeing around 10hz. 

Does anyone know what the possible cause and potential solution could be?
Thanks!

pablogs9, 2021-09-16:
Which was your previous commit and the current one? I need this to check the changes.

Also, can you share your code so we can check how you are creating the entities?

timongentzsch, 2021-09-16:
We also just tried increasing the frequency of the int32_publisher_custom_transport example and it seems to be capped at 10hz. 
Have you been able to increase that?

timongentzsch, 2021-09-16:
We are also a bit confused on the micro_ros source/build process, it seems that even if we check out the earlier commit, we are still obtaining a new version when building. 
For example, the old  <rmw_uros/options.h> is not available even after checkout. It seems as tho micro_ros_component will always pull the latest repos it needs to compile?

pablogs9, 2021-09-16:
I have been able to replicate the problem, working on that.

timongentzsch, 2021-09-16:
okay great, thanks
btw we also noticed that when using a similar paradigm to your multithread example we then get a split of 5hz per publisher. Might help you narrow down the issue.

pablogs9, 2021-09-16:
This should solve your issue: https://github.com/micro-ROS/micro_ros_espidf_component/pull/91

It is related with the heartbeat process in reliable communications. In fact with your current setup, if you make a best effort publisher (which is recommended for those rates), it will be faster than 10 Hz. With the patch that I'm applying the reliable publisher should reach at least 60 Hz.

timongentzsch, 2021-09-16:
Okay thanks
So if I'm understanding you correctly, using the best effort publisher we will be able to reach the 150hz we were able to achieve previously?

timongentzsch, 2021-09-16:
Awesome, this fixed our problem.
We have a quick follow-up question tho, is there any documentation on these flags? 
We were not aware of `"-DUCLIENT_MIN_HEARTBEAT_TIME_INTERVAL=1"` and are also wondering what `                "-DUCLIENT_PROFILE_MULTITHREAD=ON"` this does exactly?
We do have a mutlithreaded publisher running and would like to make it as stable as possible. 
Thanks

pablogs9, 2021-09-16:
`UCLIENT_MIN_HEARTBEAT_TIME_INTERVAL` is documented here: https://micro-xrce-dds.docs.eprosima.com/en/latest/client.html?highlight=UCLIENT_MIN_HEARTBEAT_TIME_INTERVAL#configurations

`DUCLIENT_PROFILE_MULTITHREAD` allows to use an XRCE-DDS session (the micro-ROS middleware) from different threads or tasks using internal mutexes. By default the middleware is not locked and access from multiple threads will cause probrems.

Regarding best-effort vs reliable in the XRCE-DDS protocol, you can check here: https://micro-xrce-dds.docs.eprosima.com/en/latest/client.html?highlight=reliable#streams

pablogs9, 2021-09-16:
I'm closing, do not hesitate to reopen if you have more problems.

closed 2021-09-16 by pablogs9

timongentzsch, 2021-09-16:
Thanks for the info and your help! All good for now :+1: