micro-ROS/micro_ros_espidf_component #94: ESP32 via UDP: `Failed status on line 48: 1. Aborting.`

leonardogargani, 2021-09-23:
## Issue template

- Hardware description: ESP32
- RTOS: FreeRTOS
- Installation type: micro-ROS component for ESP-IDF
- Version: foxy

#### Steps to reproduce the issue

Install ESP-IDF and create a first project, as written [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/):

```bash
sudo pacman -S --needed gcc git make flex bison gperf python-pip cmake ninja ccache dfu-util libusb
mkdir -p ~/esp
cd ~/esp
git clone --recursive https://github.com/espressif/esp-idf.git
cd ~/esp/esp-idf
./install.sh esp32
. $HOME/esp/esp-idf/export.sh
cd ~/esp
cp -r $IDF_PATH/examples/get-started/hello_world .
cd ~/esp/hello_world
idf.py set-target esp32
idf.py build
idf.py -p /dev/ttyUSB0 flash
```

At this point with `idf.py -p /dev/ttyUSB0 monitor` I can correctly see the "Hello world!" output being generated by the ESP32.

Moving on, install the micro-ROS component for ESP-IDF and run the _int32_publisher_ example, as written [here](https://github.com/micro-ROS/micro_ros_espidf_component/tree/foxy#readme):

```bash
. $IDF_PATH/export.sh
pip3 install catkin_pkg lark-parser empy colcon-common-extensions
cd ~/esp/hello_world
mkdir components
cd ~/esp/hello_world/components
git clone -b foxy https://github.com/micro-ROS/micro_ros_espidf_component.git
cd ~/esp/hello_world/components/micro_ros_espidf_component/examples/int32_publisher
idf.py set-target esp32
idf.py menuconfig
# set WiFi credentials, and agent's port (8888) and IP address (same as local machine: 192.168.1.7)
idf.py build 
idf.py -p /dev/ttyUSB0 flash  
```

Run a micro-ROS Agent on a new terminal:
```bash
docker run -it --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6
```

Monitor the ESP32:
```bash
 idf.py -p /dev/ttyUSB0 monitor 
```

#### Expected behavior

I should be able to see the ESP32 publishing messages on a new topic.

#### Actual behavior

The ESP32 does not publish any message, and no new topic is created.

From the terminal monitoring the ESP32 I see that it successfully connects to the WiFi network, but then the error `Failed status on line 48: 1. Aborting.` appears, referring to [this line](https://github.com/micro-ROS/micro_ros_espidf_component/blob/foxy/examples/int32_publisher/main/main.c#L48).

The output of the micro-ROS Agent is the following:
```bash
➜  ~ docker run -it --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6
[1632387100.513237] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1632387100.513442] info     | Root.cpp            | set_verbose_level        | logger setup           | verbose_level: 6
```

pablogs9, 2021-09-23:
Which are the IP address of the ESP32 and the IP address of your machine?

leonardogargani, 2021-09-23:
- IP address of the ESP32: `192.168.1.6`
- IP address of my machine: `192.168.1.7`

pablogs9, 2021-09-23:
It seems that your ESP32 and the agent have no network communication. Because ESP32 packets are not reaching your machine.

Can you check that with another network example?

Also, can you hardcode the agent address [here](https://github.com/micro-ROS/micro_ros_espidf_component/blob/54a4d3f2067a20c3b7151f824a1e8c2534cae295/examples/int32_publisher/main/main.c#L44)?

leonardogargani, 2021-09-23:
I tried other examples and every time it fails (aborting on the same line).

I tried hardcoding the agent address with `RCCHECK(rmw_uros_options_set_udp_address("192.168.1.7", CONFIG_MICRO_ROS_AGENT_PORT, rmw_options));` and it still fails (aborting on the same line).

Moreover, I tested branch _release/v4.3_ of the ESP-IDF (instead of _main_), and once again the result is the same.

pablogs9, 2021-09-24:
I meant another networking example like opening a UDP port in the computer and sending data from the ESP... I guess that there is a network connectivity problem in your setup.

leonardogargani, 2021-09-24:
Oh I got it, finally.
I started digging into the low-level stuff but it was simply due to my **firewall rules**, don't know why I didn't think about that before...
Hope this will save some time to anyone facing the same problem in future: just update your firewall rules.
And, of course, thank you for the support and for your work on micro-ROS!

closed 2021-09-24 by pablogs9

AntumArk, 2022-03-05:
@leonardogargani Were you running it on windows? Can you provide more details on what kind of rules have you updated?

leonardogargani, 2022-03-05:
> @leonardogargani Were you running it on windows? Can you provide more details on what kind of rules have you updated?

Assuming you're asking about the micro-ROS Agent, I was running it as a docker container from a Linux machine.

Look at the way I was launching it:
```bash
docker run -it --rm --net=host microros/micro-ros-agent:foxy udp4 --port 8888 -v6
```

The solution was simply opening port 8888 for UDP traffic... pretty obvious, but I wasn't thinking about it at that moment.

AntumArk, 2022-03-05:
Yes I was asking about the agent. Unfortunately this solution does not work for me in Windows, but works on Linux straight away. Have you adjusted any settings in Linux or gateway? Or just used that line you just wrote and it started working?

leonardogargani, 2022-03-05:
After having added the firewall rule (port 8888 open for UDP traffic), it started working correctly without any other adjustment.

I have no experience with Windows, but if I were you I would make sure that both Windows Firewall and any installed antivirus allow the communication above.