micro-ROS/micro_ros_espidf_component #96: Error during building

minutiaes, 2021-10-09:
## Issue template

- Hardware description: ESP32
- RTOS: FreeRTOS
- Installation type: micro_ros as idf project component
- Version or commit hash: foxy

#### Steps to reproduce the issue
I create a project and `components` directory in it and then clone this repo in `components` directory
After that I follow the instructions in repo for the example `int32_publisher`
Only difference I change certificate configuration manually as described here 
[https://github.com/espressif/esp-idf/issues/5322#issuecomment-935331910](url)

When I try to build it, it fails at 2 points. Output is below
```bash
FAILED: libmicroros-prefix/src/libmicroros_project-stamp/libmicroros_project-build /home/berk/esp/ros_test/components/micro_ros_espidf_component/libmicroros.a 
```
and
```bash
from catkin_pkg.package import evaluate_condition
ModuleNotFoundError: No module named 'catkin_pkg'
```
If I try to import this statement in a python script, it works as expected.

#### Actual behavior
```bash
Executing action: all (aliases: build)
Running ninja in directory /home/berk/esp/ros_test/components/micro_ros_espidf_component/examples/int32_publisher/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/berk/esp/ros_test/components/micro_ros_espidf_component/examples/int32_publisher/sdkconfig
Loading defaults file /home/berk/esp/ros_test/components/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults...
-- App "int32_publisher" version: 34948a0
-- Adding linker script /home/berk/esp/esp-idf-v4.3.1/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/berk/esp/esp-idf-v4.3.1/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/berk/esp/esp-idf-v4.3.1/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/berk/esp/esp-idf-v4.3.1/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/berk/esp/esp-idf-v4.3.1/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/berk/esp/esp-idf-v4.3.1/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/berk/esp/esp-idf-v4.3.1/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/berk/esp/ros_test/components/micro_ros_espidf_component/examples/int32_publisher/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/berk/esp/esp-idf-v4.3.1/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/berk/esp/esp-idf-v4.3.1/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns micro_ros_espidf_component mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/berk/esp/esp-idf-v4.3.1/components/app_trace /home/berk/esp/esp-idf-v4.3.1/components/app_update /home/berk/esp/esp-idf-v4.3.1/components/asio /home/berk/esp/esp-idf-v4.3.1/components/bootloader /home/berk/esp/esp-idf-v4.3.1/components/bootloader_support /home/berk/esp/esp-idf-v4.3.1/components/bt /home/berk/esp/esp-idf-v4.3.1/components/cbor /home/berk/esp/esp-idf-v4.3.1/components/cmock /home/berk/esp/esp-idf-v4.3.1/components/coap /home/berk/esp/esp-idf-v4.3.1/components/console /home/berk/esp/esp-idf-v4.3.1/components/cxx /home/berk/esp/esp-idf-v4.3.1/components/driver /home/berk/esp/esp-idf-v4.3.1/components/efuse /home/berk/esp/esp-idf-v4.3.1/components/esp-tls /home/berk/esp/esp-idf-v4.3.1/components/esp32 /home/berk/esp/esp-idf-v4.3.1/components/esp_adc_cal /home/berk/esp/esp-idf-v4.3.1/components/esp_common /home/berk/esp/esp-idf-v4.3.1/components/esp_eth /home/berk/esp/esp-idf-v4.3.1/components/esp_event /home/berk/esp/esp-idf-v4.3.1/components/esp_gdbstub /home/berk/esp/esp-idf-v4.3.1/components/esp_hid /home/berk/esp/esp-idf-v4.3.1/components/esp_http_client /home/berk/esp/esp-idf-v4.3.1/components/esp_http_server /home/berk/esp/esp-idf-v4.3.1/components/esp_https_ota /home/berk/esp/esp-idf-v4.3.1/components/esp_https_server /home/berk/esp/esp-idf-v4.3.1/components/esp_hw_support /home/berk/esp/esp-idf-v4.3.1/components/esp_ipc /home/berk/esp/esp-idf-v4.3.1/components/esp_local_ctrl /home/berk/esp/esp-idf-v4.3.1/components/esp_netif /home/berk/esp/esp-idf-v4.3.1/components/esp_pm /home/berk/esp/esp-idf-v4.3.1/components/esp_ringbuf /home/berk/esp/esp-idf-v4.3.1/components/esp_rom /home/berk/esp/esp-idf-v4.3.1/components/esp_serial_slave_link /home/berk/esp/esp-idf-v4.3.1/components/esp_system /home/berk/esp/esp-idf-v4.3.1/components/esp_timer /home/berk/esp/esp-idf-v4.3.1/components/esp_websocket_client /home/berk/esp/esp-idf-v4.3.1/components/esp_wifi /home/berk/esp/esp-idf-v4.3.1/components/espcoredump /home/berk/esp/esp-idf-v4.3.1/components/esptool_py /home/berk/esp/esp-idf-v4.3.1/components/expat /home/berk/esp/esp-idf-v4.3.1/components/fatfs /home/berk/esp/esp-idf-v4.3.1/components/freemodbus /home/berk/esp/esp-idf-v4.3.1/components/freertos /home/berk/esp/esp-idf-v4.3.1/components/hal /home/berk/esp/esp-idf-v4.3.1/components/heap /home/berk/esp/esp-idf-v4.3.1/components/idf_test /home/berk/esp/esp-idf-v4.3.1/components/jsmn /home/berk/esp/esp-idf-v4.3.1/components/json /home/berk/esp/esp-idf-v4.3.1/components/libsodium /home/berk/esp/esp-idf-v4.3.1/components/log /home/berk/esp/esp-idf-v4.3.1/components/lwip /home/berk/esp/ros_test/components/micro_ros_espidf_component/examples/int32_publisher/main /home/berk/esp/esp-idf-v4.3.1/components/mbedtls /home/berk/esp/esp-idf-v4.3.1/components/mdns /home/berk/esp/ros_test/components/micro_ros_espidf_component /home/berk/esp/esp-idf-v4.3.1/components/mqtt /home/berk/esp/esp-idf-v4.3.1/components/newlib /home/berk/esp/esp-idf-v4.3.1/components/nghttp /home/berk/esp/esp-idf-v4.3.1/components/nvs_flash /home/berk/esp/esp-idf-v4.3.1/components/openssl /home/berk/esp/esp-idf-v4.3.1/components/partition_table /home/berk/esp/esp-idf-v4.3.1/components/perfmon /home/berk/esp/esp-idf-v4.3.1/components/protobuf-c /home/berk/esp/esp-idf-v4.3.1/components/protocomm /home/berk/esp/esp-idf-v4.3.1/components/pthread /home/berk/esp/esp-idf-v4.3.1/components/sdmmc /home/berk/esp/esp-idf-v4.3.1/components/soc /home/berk/esp/esp-idf-v4.3.1/components/spi_flash /home/berk/esp/esp-idf-v4.3.1/components/spiffs /home/berk/esp/esp-idf-v4.3.1/components/tcp_transport /home/berk/esp/esp-idf-v4.3.1/components/tcpip_adapter /home/berk/esp/esp-idf-v4.3.1/components/tinyusb /home/berk/esp/esp-idf-v4.3.1/components/ulp /home/berk/esp/esp-idf-v4.3.1/components/unity /home/berk/esp/esp-idf-v4.3.1/components/vfs /home/berk/esp/esp-idf-v4.3.1/components/wear_levelling /home/berk/esp/esp-idf-v4.3.1/components/wifi_provisioning /home/berk/esp/esp-idf-v4.3.1/components/wpa_supplicant /home/berk/esp/esp-idf-v4.3.1/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/berk/esp/ros_test/components/micro_ros_espidf_component/examples/int32_publisher/build
[11/977] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[295/977] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/berk/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
[... cut ...]
make: *** [libmicroros.mk:38: /home/berk/esp/ros_test/components/micro_ros_espidf_component/micro_ros_dev/install] Error 1
make: *** Waiting for unfinished jobs....
Cloning into 'src/unique_identifier_msgs'...
Cloning into 'src/common_interfaces'...
Cloning into 'src/test_interface_files'...
Cloning into 'src/rmw_implementation'...
Cloning into 'src/rcl_logging'...
Cloning into 'src/ros2_tracing'...
warning: redirecting to https://gitlab.com/micro-ROS/ros_tracing/ros2_tracing.git/
Cloning into 'src/micro_ros_utilities'...
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

pablogs9, 2021-10-11:
Have you installed the [dependencies](https://github.com/micro-ROS/micro_ros_espidf_component#dependencies) when your ESP-IDF is sourced?

pablogs9, 2021-10-13:
Yes, the component is the prefered method for IDF developers. In any case, can you try to use it in a clean environment such as a Docker, just to check that this is a misconfiguration in your actual workspace?

wegunterjrFIrefly, 2021-11-07:
i was able to build using the environment on my dev laptop and the docker..but i just started getting this fail on the tracing. 
I was switching between galactic and foxy chasing down some other errors, but after cleaning the workspace and starting getting this.
Of course I can build with the docker...just curious?  My network settings haven't changed since the local env was working earlier today...gotta be something I am missing.

pablogs9, 2022-01-28:
Any update on this?

Acuadros95, 2022-07-26:
Closing due to inactivity, feel free to reopen.

closed 2022-07-26 by Acuadros95

bernacbs, 2024-02-28:
I have a similar problem, any update?

Doing idf.py menuconfig

Ouput:
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.2/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Using Micro XRCE-DDS middleware
CMake Error at micro_ros_espidf_component/CMakeLists.txt:74 (message):
  FAILED: O sistema não pode encontrar o arquivo especificado

-- Configuring incomplete, errors occurred!
See also "C:/IDF-Proj/uros/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

bernacbs, 2024-02-28:
by follow CMakeList.txt:74 i find an if operator. But i am new on Cmake world. Can some one explain why it is working when i comment this 3 lines:

[image]

losingrose, 2024-07-23:
same problem on windows use esp-idf 5.2.2：
`-- Using Micro XRCE-DDS middleware
CMake Error at D:/Git/micro_ros_espidf_component/CMakeLists.txt:77 (message):
  FAILED: 系统找不到指定的文件。

-- Configuring incomplete, errors occurred!
See also "D:/Git/micro_ros_espidf_component/examples/int32_publisher/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the D:\Git\micro_ros_espidf_component\examples\int32_publisher\build\log\idf_py_stderr_output_22884 and D:\Git\micro_ros_espidf_component\examples\int32_publisher\build\log\idf_py_stdout_output_22884`

CMakeLists.txt:76-78 lines
`if(libmicroros_ret AND NOT libmicroros_ret EQUAL 0)
  message(FATAL_ERROR "FAILED: ${libmicroros_ret}")
endif()`