micro-ROS/micro_ros_espidf_component #108: xtensa-esp32-elf esp-2021r2-8.4.0 not working with micro-ROS

pnn16004, 2021-11-05:
## Issue template

- Hardware description: ESP32
- RTOS: FreeRTOS
- Installation type: micro-ROS ESP-IDF component, git cloned ESP-IDF
- Version or commit hash: foxy, ESP-IDF v4.4

#### Steps to reproduce the issue
Use git clone on ESP-IDF v4.4, download submodules, run install.sh, get dependencies for this component.
On the example int32_publisher_custom_transport, put in this component in the components folder (example not within component).
Build and flash component as usual.

#### Expected behavior
Successfully build.

#### Actual behavior
Following errors:

`[100%] Linking CXX executable int32_publisher.elf
/home/skola/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: ../components/micro_ros_espidf_component-foxy/libmicroros.a(librmw-validate_node_name.c.obj): in function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x6a): undefined reference to `__locale_ctype_ptr'
/home/skola/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: ../components/micro_ros_espidf_component-foxy/libmicroros.a(librmw-validate_full_topic_name.c.obj):(.literal.rmw_validate_full_topic_name+0x0): undefined reference to `__locale_ctype_ptr'
/home/skola/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: ../components/micro_ros_espidf_component-foxy/libmicroros.a(librmw-validate_full_topic_name.c.obj): in function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xd3): undefined reference to `__locale_ctype_ptr'
/home/skola/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: ../components/micro_ros_espidf_component-foxy/libmicroros.a(librcl-validate_topic_name.c.obj): in function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x26): undefined reference to `__locale_ctype_ptr'
/home/skola/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0x98): undefined reference to `__locale_ctype_ptr'
/home/skola/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: ../components/micro_ros_espidf_component-foxy/libmicroros.a(librcl-validate_topic_name.c.obj):validate_topic_name.c:(.text.rcl_validate_topic_name+0x193): more undefined references to `__locale_ctype_ptr' follow
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/int32_publisher.elf.dir/build.make:492: int32_publisher.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:2487: CMakeFiles/int32_publisher.elf.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
make failed with exit code 2
`

#### Additional information
Hi, I know v4.4 is not tested yet, but my group member gave me a ESP-IDF project containing the function i2c_master_write_read_device, for me to integrate into micro-ROS. That function is first introduced in ESP-IDF v4.4, which is why I'm experimenting with it.

I'm able to build the code with micro-ROS ESP-IDF component if the project is just as is, and even if I include something like <rclc/rclc.h> it still compiles, but when I try to mix the project with the int32_publisher_custom_transport example, it fails at the last step, complaining about the new xtensa-esp32-elf esp-2021r2-8.4.0 version (v4.1 is esp-2020r2-8.2.0).

Does anyone know if this can be solved? Or should I try to solve the original problem in another way?

pnn16004, 2021-11-08:
My group member rewrote the function for v4.1, but I'm still curious.

pablogs9, 2021-11-10:
Later I will provide support for v4.4 and see if we can replicate this

pablogs9, 2021-11-10:
Check here: https://github.com/micro-ROS/micro_ros_espidf_component/pull/110

pablogs9, 2021-11-10:
v4.4 integrated in CI and working. Closing this. Please reopen if you have more questions.

closed 2021-11-10 by pablogs9