micro-ROS/micro_ros_espidf_component #109: using ethernet on esp32 instead of wifi having some issues with compiling.

wegunterjrFIrefly, 2021-11-07:
## Issue template

- Hardware description: esp32 huzzah32 and ethernet featherweight
- RTOS: freertos
- Installation type:  components
- Version or commit hash: foxy

#### Steps to reproduce the issue
In menuconfig, selected micro-ROS over network uses Ethernet netif.
Ethernet configuration, selected W5500 module.

#### Expected behavior
Expected build to complete successfully and then flash and connect to agent.

#### Actual behavior
` error: unknown type name 'eth_w5500_config_t'; did you mean 'eth_phy_config_t'?
     eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);`
#### Additional information

pablogs9, 2021-11-10:
@uLipe Any idea on that?

uLipe, 2021-11-10:
The Ethernet transport for microROS runs on top of esp-eth component which can select by using internal mac or a external one like the wiznet chips.

The esp-eth is able to resolve internally this when the configurations options are pointing to the SPI-ETH interface.

It seems there is some kind of bad configuration of the ETH netif.

pablogs9, 2022-01-28:
Any update on this?

Acuadros95, 2022-07-26:
Closing due to inactivity, feel free to reopen.

closed 2022-07-26 by Acuadros95

vijhu, 2023-06-14:
Hello @wegunterjrFIrefly ,

I am trying to use the ethernet on ESP32 on linux environment.

The problem is , I am not able to configure the ethernet on microros for ESP32. 

Can you please mention the steps for the configurations?
It will be a great help .

Thank you so much!!