micro-ROS/micro_ros_espidf_component #113: Problems when adding second publisher - Aborts on RCCHECK at publish init

wegunterjr, 2021-11-27:
## Issue template

- Hardware description: esp32
- RTOS: FreeRTOS
- Installation type: micro ros esp idf component
- Version or commit hash: galactic

#### Steps to reproduce the issue
based on int_32 publisher example, but this time adding two publishers

`    // create publisher
    RCCHECK(rclc_publisher_init_best_effort(
        &publisher,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, Temperature),
        "temperature"));

    // //create humidty publisher
    RCCHECK(rclc_publisher_init_best_effort(
        &humidityPublisher,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, RelativeHumidity),
        "humidity"));`
        
 colcon.meta in esp_idf_component is set to 
`        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_XML_BUFFER_LENGTH=400",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=2",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=1",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=1"
            ]
        },`

#### Expected behavior
Expected to publish two topics from same device

#### Actual behavior
Aborts at the line of the publisher init
`Failed status on line 181: 1. Aborting.`
#### Additional information

wegunterjrFIrefly, 2021-11-28:
alright...made some progress.  I had to pull the latest esp component and switch to galactic branch.
Then I could run the micro_ros_agent (galactic).  This may seem obvious but have had issues with foxy in my last round of testing.

pablogs9, 2021-11-29:
That means that the issue is solved?

closed 2021-11-29 by wegunterjr