micro-ROS/micro_ros_espidf_component #115: Can't publish CompressedImage topic

MS71, 2021-12-09:
## Issue template
I like to create a ESP32 Cam node sending a CompressedImage topic.

- Hardware description: ESP32 CAM

#### Steps to reproduce the issue

Code: https://github.com/MS71/ESP32microROSImagePublisher.git

#### Expected behavior

rcl_publish should send the topic without error.

#### Actual behavior

I (1378) spiram: SPI SRAM memory test OK
I (1386) cpu_start: Pro cpu start user code
I (1386) cpu_start: cpu freq: 160000000
I (1386) cpu_start: Application information:
I (1389) cpu_start: Project name:     int32_publisher
I (1395) cpu_start: App version:      ff24a87
I (1400) cpu_start: Compile time:     Dec  9 2021 21:19:55
I (1406) cpu_start: ELF file SHA256:  d8325eb3d28ee7e6...
I (1412) cpu_start: ESP-IDF:          v4.3.1
I (1418) heap_init: Initializing. RAM available for dynamic allocation:
I (1425) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1431) heap_init: At 3FFBD410 len 00022BF0 (138 KiB): DRAM
I (1437) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1444) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1450) heap_init: At 4009A5D4 len 00005A2C (22 KiB): IRAM
I (1457) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1466) spi_flash: detected chip: generic
I (1469) spi_flash: flash io: dio
I (1475) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1484) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1544) wifi_station_netif: ESP_WIFI_MODE_STA
I (1554) wifi:wifi driver task: 3ffcf314, prio:23, stack:6656, core=0
I (1554) system_api: Base MAC address is not set
I (1554) system_api: read default base MAC address from EFUSE
I (1574) wifi:wifi firmware version: 88c8747
I (1574) wifi:wifi certification version: v7.0
I (1574) wifi:config NVS flash: enabled
I (1574) wifi:config nano formating: disabled
I (1584) wifi:Init data frame dynamic rx buffer num: 32
I (1584) wifi:Init management frame dynamic rx buffer num: 32
I (1594) wifi:Init management short buffer num: 32
I (1594) wifi:Init static tx buffer num: 16
I (1604) wifi:Init tx cache buffer num: 32
I (1604) wifi:Init static rx buffer size: 1600
I (1614) wifi:Init static rx buffer num: 10
I (1614) wifi:Init dynamic rx buffer num: 32
I (1614) wifi_init: rx ba win: 6
I (1624) wifi_init: tcpip mbox: 32
I (1624) wifi_init: udp mbox: 6
I (1634) wifi_init: tcp mbox: 6
I (1634) wifi_init: tcp tx win: 5744
I (1634) wifi_init: tcp rx win: 5744
I (1644) wifi_init: tcp mss: 1440
I (1644) wifi_init: WiFi IRAM OP enabled
I (1654) wifi_init: WiFi RX IRAM OP enabled
I (1654) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (1814) phy_init: saving new calibration data because of checksum failure, mode(0)
I (1834) wifi:mode : sta (fc:f5:c4:30:ab:54)
I (1834) wifi:enable tsf
I (1834) wifi_station_netif: wifi_init_sta finished.
I (1844) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (1844) wifi:state: init -> auth (b0)
I (1864) wifi:state: auth -> assoc (0)
I (1874) wifi:state: assoc -> run (10)
I (1894) wifi:connected with ScholzHomeNet, aid = 3, channel 1, 40U, bssid = 7c:ff:4d:4a:f2:dd
I (1894) wifi:security: WPA2-PSK, phy: bgn, rssi: -80
I (1904) wifi:pm start, type: 1

W (1914) wifi:<ba-add>idx:0 (ifx:0, 7c:ff:4d:4a:f2:dd), tid:0, ssn:0, winSize:64
I (1964) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2544) esp_netif_handlers: sta ip: 192.168.0.175, mask: 255.255.255.0, gw: 192.168.0.1
I (2544) wifi_station_netif: got ip:192.168.0.175
I (2544) wifi_station_netif: connected to ap SSID:xxxxxxxxx password:xxxxxxxxx
I (2554) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (2564) cam_hal: cam init ok
I (2574) sccb: pin_sda 26 pin_scl 27
I (2574) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2614) camera: Detected camera at address=0x30
I (2614) camera: Detected OV2640 camera
I (2614) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (2704) cam_hal: buffer_size: 32768, half_buffer_size: 4096, node_buffer_size: 2048, node_cnt: 16, total_cnt: 3
I (2704) cam_hal: Allocating 15360 Byte frame buffer in PSRAM
I (2704) cam_hal: cam config ok
I (2714) ov2640: Set PLL: clk_2x: 0, clk_div: 0, pclk_auto: 0, pclk_div: 8
I (3814) example:take_picture: Taking picture...
I (3814) example:take_picture: Picture taken! Its size was: 2597 bytes 320 240 2597 3000 0x3fff8388
Failed status on line 153: 1. Continuing.
I (4814) example:take_picture: Taking picture...
I (4814) example:take_picture: Picture taken! Its size was: 2617 bytes 320 240 2617 3000 0x3fff8388
Failed status on line 153: 1. Continuing.
I (5814) example:take_picture: Taking picture...
I (5814) example:take_picture: Picture taken! Its size was: 2617 bytes 320 240 2617 3000 0x3fff8388
Failed status on line 153: 1. Continuing.
I (6814) example:take_picture: Taking picture...
I (6814) example:take_picture: Picture taken! Its size was: 2617 bytes 320 240 2617 3000 0x3fff8388
**Failed status on line 153: 1. Continuing.**
I (7814) example:take_picture: Taking picture...
I (7814) example:take_picture: Picture taken! Its size was: 2617 bytes 320 240 2617 3000 0x3fff8388
**Failed status on line 153: 1. Continuing.**
I (8814) example:take_picture: Taking picture...
I (8814) example:take_picture: Picture taken! Its size was: 2617 bytes 320 240 2617 3000 0x3fff8388
**Failed status on line 153: 1. Continuing.**
I (9814) example:take_picture: Taking picture...
I (9814) example:take_picture: Picture taken! Its size was: 2617 bytes 320 240 2617 3000 0x3fff8388
**Failed status on line 153: 1. Continuing.**

#### Additional information

My questions:
A.) What could be the reason for "Failed status on line 153: 1. Continuing."?
B.) How can I debug such microROS error messages. Where is "153: 1."?

pablogs9, 2021-12-10:
I have this working properly: 

```c
int main(int argc, const char * const * argv)
{
  	rcl_allocator_t allocator = rcl_get_default_allocator();
	rclc_support_t support;

	// Create init_options
	RCCHECK(rclc_support_init(&support, argc, argv, &allocator));

	// Create node
	rcl_node_t node;
	RCCHECK(rclc_node_init_default(&node, "node", "", &support));

	// Create publishers

	rcl_publisher_t publisher;
	RCCHECK(rclc_publisher_init_default(
		&publisher,
		&node,
    	ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, CompressedImage),
		"topic"));

	static micro_ros_utilities_memory_conf_t conf = {};

	conf.max_string_capacity = 50;
	conf.max_ros2_type_sequence_capacity = 5;
	conf.max_basic_type_sequence_capacity = 5;

	micro_ros_utilities_memory_rule_t rules[] = {
		{"header.frame_id", 30},
		{"format",4},
		{"data", 3000}
	};
	conf.rules = rules;
	conf.n_rules = sizeof(rules) / sizeof(rules[0]);

	sensor_msgs__msg__CompressedImage msg_static;

	micro_ros_utilities_create_message_memory(
		ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, CompressedImage),
		&msg_static,
		conf
	);

	uint8_t pic[2617];
	memset(pic, 0xAB, sizeof(pic));

	msg_static.data.size = sizeof(pic);

	msg_static.header.frame_id = micro_ros_string_utilities_set(msg_static.header.frame_id, "myframe");
	msg_static.format = micro_ros_string_utilities_set(msg_static.format, "JPEG");
	memcpy(msg_static.data.data,pic,sizeof(pic));

	RCSOFTCHECK(rcl_publish(&publisher, &msg_static, NULL));
}
```
Can you verify this code in your platform?

Some questions:
1. Why are you using ` %zu` to print `pic->len` if [`len` is type `size_t`](https://github.com/espressif/esp32-camera/blob/61400bc73b18e9ffcfd5b3806fee26d135c45fe7/driver/include/esp_camera.h#L134) and width and height are also `size_t` and you are using `%d`?
2. Is correct that a 320x240 image fits in 2617 bytes?

MS71, 2021-12-10:
Hi
i will try your code.

Maybe, timer_callback is called by a different task with limited stack. 
I will check that.

>Is correct that a 320x240 image fits in 2617 bytes?
Yes, it a JPEG image.

Yesterday, i checked with smaller payload. 
Up to ~800byte, rcl_publish is working.

pablogs9, 2021-12-10:
Maybe you can try to increase the XRCE history to 8 in the colcon.meta: [this param](https://github.com/micro-ROS/rmw_microxrcedds/blob/bb581b6e26c908bcff96b93e9b8b7b7ce7f33453/rmw_microxrcedds_c/CMakeLists.txt#L86)

MS71, 2021-12-10:
Ok,
can i have my own colcon.meta, located with the application code? Outsite micro_ros_espidf_component.

MS71, 2021-12-12:
Now, it is working.

        "rmw_embeddedrtps": {
            "cmake-args": [
                "-DRMW_ERTPS_MAX_OUTPUT_BUFFER_SIZE=20100"
            ]
        },

Question: How can i use my own/custom colcon.meta outside micro_ros_espidf_component?

pablogs9, 2021-12-13:
You can use `app-colcon.meta` in the project folder: https://github.com/micro-ROS/micro_ros_espidf_component/blob/4bd55e980e4521d0fff018109829dfad2834d52d/CMakeLists.txt#L11

Take into account that RMW_embeddedrtps is experimental.

closed 2021-12-13 by pablogs9