micro-ROS/micro_ros_espidf_component #116: Build process fails for "int32_publisher_custom_transport" example

githubuser182736, 2021-12-20:
## Issue template

- Hardware description: ESP32
- RTOS: freertos
- Installation type: .
- Version or commit hash: galactic

#### Steps to reproduce the issue
Put following two files (Dockerfile and bash_script.sh) into one folder and run ```docker build -t testcontainer .``` inside this folder.

Dockerfile:
```dockerfile
FROM espressif/idf:release-v4.3

ENV DEBIAN_FRONTEND noninteractive
RUN echo "Set disable_coredump false" >> /etc/sudo.conf
RUN apt update -q && \
    apt install -yq sudo lsb-release gosu nano && \
    rm -rf /var/lib/apt/lists/*

ARG TZ_ARG=UTC
ENV TZ=$TZ_ARG
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

COPY ./bash_script.sh /bash_script.sh

RUN mkdir -p /tmp/bash_script && mv /bash_script.sh /tmp/bash_script/ && \
    /tmp/bash_script/bash_script.sh
``` 
bash_script.sh:
```bash
#!/bin/bash

set -eu

sudo apt update -q
sudo apt install -yq python3-pip
source $IDF_PATH/export.sh
pip3 install catkin_pkg lark-parser empy colcon-common-extensions importlib-resources
git clone https://github.com/micro-ROS/micro_ros_espidf_component.git
cd micro_ros_espidf_component
sed -i 's/DRMW_UXRCE_TRANSPORT=udp/DRMW_UXRCE_TRANSPORT=custom/' colcon.meta
cd examples/int32_publisher_custom_transport
idf.py set-target esp32
idf.py build

set +u
```

#### Expected behavior
Build process finishes without errors.

#### Actual behavior
Build process fails. See attachment "output-int32_publisher_custom_transport.log".
[output-int32_publisher_custom_transport.log](https://github.com/micro-ROS/micro_ros_espidf_component/files/7747074/output-int32_publisher_custom_transport.log)

#### Additional information
As a check the "int32_publisher" example could be built without any error with this bash_script.sh:
```bash
#!/bin/bash

set -eu

sudo apt update -q
sudo apt install -yq python3-pip
source $IDF_PATH/export.sh
pip3 install catkin_pkg lark-parser empy colcon-common-extensions importlib-resources
git clone https://github.com/micro-ROS/micro_ros_espidf_component.git
cd micro_ros_espidf_component/examples/int32_publisher
idf.py set-target esp32
idf.py build

set +u
```
The output of this build process can be found in the attachment "output-int32_publisher.log"
[output-int32_publisher.log](https://github.com/micro-ROS/micro_ros_espidf_component/files/7747077/output-int32_publisher.log)

pablogs9, 2021-12-22:
Hello, the error that appears in your log is related with a bug that we had a couple of days ago, it is solved here:
https://github.com/micro-ROS/rmw_microxrcedds/pull/221

Could you please pull the latest version of all micro-ROS repos, specially rmw_microxrcedds?

githubuser182736, 2021-12-22:
@pablogs9 Thank you! Now it's working :)

closed 2021-12-22 by githubuser182736