micro-ROS/micro_ros_espidf_component #127: Build of "addtwoints_server" example fails using idf.py, despite following all the setup steps

safdark, 2022-02-10:
## Issue template

- Hardware description: Generic desktop running Ubuntu 20.04
- RTOS: FreeRTOS
- Installation type: idf.py
- Version or commit hash: galactic

#### Steps to reproduce the issue

Once esp-idf has been setup under ~/esp:
- cd ~/esp/
- Source export.sh from the esp-idf folder
```
    Detecting the Python interpreter
    Checking "python" ...
    Python 3.8.0
    "python" has been detected
    Adding ESP-IDF tools to PATH...
    Using Python interpreter in /home/safdar/.espressif/python_env/idf4.4_py3.8_env/bin/python
    Checking if Python packages are up to date...
    Python requirements from /home/safdar/esp/esp-idf/requirements.txt are satisfied.
    Added the following directories to PATH:
      /home/safdar/esp/esp-idf/components/esptool_py/esptool
      /home/safdar/esp/esp-idf/components/espcoredump
      /home/safdar/esp/esp-idf/components/partition_table
      /home/safdar/esp/esp-idf/components/app_update
    Done! You can now compile ESP-IDF projects.
    Go to the project directory and run:
      idf.py build
```
- Ensure any ROS environment has not been sourced:
```
    (base) safdar@obelix:~/esp/addtwoints_server$ echo $PATH | grep -i "ros"
    <Nothing>
    
    (base) safdar@obelix:~/esp/addtwoints_server$ env | grep ROS
    <Nothing>
    
    (base) safdar@obelix:~/esp/addtwoints_server$ env | grep -i "ament"
    <Nothing>
```
- cd esp-idf/components
- git clone git@github.com:micro-ROS/micro_ros_espidf_component.git
- cd ~/esp
- Copy over the example project:
```
    cp -r ~/esp/esp-idf/components/micro_ros_espidf_component/examples/addtwoints_server .
```
- Run> idf.py set-target esp32
```
    (base) safdar@obelix:~/esp/addtwoints_server$ idf.py set-target esp32
    Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
    Executing action: fullclean
    Executing action: set-target
    Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
    Running cmake in directory /home/safdar/esp/addtwoints_server/build
    Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/safdar/esp/addtwoints_server"...
    -- Found Git: /usr/bin/git (found version "2.25.1") 
    -- Component directory /home/safdar/Arduino does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/safdar/Desktop does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/safdar/Documents does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/safdar/Downloads does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/safdar/Music does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/safdar/Pictures does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/safdar/Public does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/safdar/Templates does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/safdar/Unity does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/safdar/Videos does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/safdar/anaconda3 does not contain a CMakeLists.txt file. No component will be added
    -- Component directory /home/safdar/code does not contain a CMakeLists.txt file. No component will be added
[... 52 lines cut ...]

Build output:
```
    ...
    [1010/1022] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
    FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj 
    /home/safdar/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I/home/safdar/esp/esp-idf/components/newlib/platform_include -I/home/safdar/esp/esp-idf/components/freertos/include -I/home/safdar/esp/esp-idf/components/freertos/include/esp_additions/freertos -I/home/safdar/esp/esp-idf/components/freertos/port/xtensa/include -I/home/safdar/esp/esp-idf/components/freertos/include/esp_additions -I/home/safdar/esp/esp-idf/components/esp_hw_support/include -I/home/safdar/esp/esp-idf/components/esp_hw_support/include/soc -I/home/safdar/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/safdar/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/safdar/esp/esp-idf/components/heap/include -I/home/safdar/esp/esp-idf/components/log/include -I/home/safdar/esp/esp-idf/components/lwip/include/apps -I/home/safdar/esp/esp-idf/components/lwip/include/apps/sntp -I/home/safdar/esp/esp-idf/components/lwip/lwip/src/include -I/home/safdar/esp/esp-idf/components/lwip/port/esp32/include -I/home/safdar/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/safdar/esp/esp-idf/components/soc/include -I/home/safdar/esp/esp-idf/components/soc/esp32/. -I/home/safdar/esp/esp-idf/components/soc/esp32/include -I/home/safdar/esp/esp-idf/components/hal/esp32/include -I/home/safdar/esp/esp-idf/components/hal/include -I/home/safdar/esp/esp-idf/components/hal/platform_port/include -I/home/safdar/esp/esp-idf/components/esp_rom/include -I/home/safdar/esp/esp-idf/components/esp_rom/include/esp32 -I/home/safdar/esp/esp-idf/components/esp_rom/esp32 -I/home/safdar/esp/esp-idf/components/esp_common/include -I/home/safdar/esp/esp-idf/components/esp_system/include -I/home/safdar/esp/esp-idf/components/esp_system/port/soc -I/home/safdar/esp/esp-idf/components/esp_system/port/public_compat -I/home/safdar/esp/esp-idf/components/esp32/include -I/home/safdar/esp/esp-idf/components/xtensa/include -I/home/safdar/esp/esp-idf/components/xtensa/esp32/include -I/home/safdar/esp/esp-idf/components/driver/include -I/home/safdar/esp/esp-idf/components/driver/esp32/include -I/home/safdar/esp/esp-idf/components/esp_pm/include -I/home/safdar/esp/esp-idf/components/esp_ringbuf/include -I/home/safdar/esp/esp-idf/components/efuse/include -I/home/safdar/esp/esp-idf/components/efuse/esp32/include -I/home/safdar/esp/esp-idf/components/vfs/include -I/home/safdar/esp/esp-idf/components/esp_wifi/include -I/home/safdar/esp/esp-idf/components/esp_event/include -I/home/safdar/esp/esp-idf/components/esp_netif/include -I/home/safdar/esp/esp-idf/components/esp_eth/include -I/home/safdar/esp/esp-idf/components/tcpip_adapter/include -I/home/safdar/esp/esp-idf/components/esp_phy/include -I/home/safdar/esp/esp-idf/components/esp_phy/esp32/include -I/home/safdar/esp/esp-idf/components/esp_ipc/include -I/home/safdar/esp/esp-idf/components/app_trace/include -I/home/safdar/esp/esp-idf/components/esp_timer/include -I/home/safdar/esp/esp-idf/components/mbedtls/port/include -I/home/safdar/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/safdar/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/safdar/esp/esp-idf/components/app_update/include -I/home/safdar/esp/esp-idf/components/spi_flash/include -I/home/safdar/esp/esp-idf/components/bootloader_support/include -I/home/safdar/esp/esp-idf/components/nvs_flash/include -I/home/safdar/esp/esp-idf/components/pthread/include -I/home/safdar/esp/esp-idf/components/esp_gdbstub/include -I/home/safdar/esp/esp-idf/components/esp_gdbstub/xtensa -I/home/safdar/esp/esp-idf/components/esp_gdbstub/esp32 -I/home/safdar/esp/esp-idf/components/espcoredump/include -I/home/safdar/esp/esp-idf/components/espcoredump/include/port/xtensa -I/home/safdar/esp/esp-idf/components/wpa_supplicant/include -I/home/safdar/esp/esp-idf/components/wpa_supplicant/port/include -I/home/safdar/esp/esp-idf/components/wpa_supplicant/esp_supplicant/include -I/home/safdar/esp/esp-idf/components/ieee802154/include -I/home/safdar/esp/esp-idf/components/console -I/home/safdar/esp/esp-idf/components/asio/asio/asio/include -I/home/safdar/esp/esp-idf/components/asio/port/include -I/home/safdar/esp/esp-idf/components/cbor/port/include -I/home/safdar/esp/esp-idf/components/unity/include -I/home/safdar/esp/esp-idf/components/unity/unity/src -I/home/safdar/esp/esp-idf/components/cmock/CMock/src -I/home/safdar/esp/esp-idf/components/coap/port/include -I/home/safdar/esp/esp-idf/components/coap/libcoap/include -I/home/safdar/esp/esp-idf/components/nghttp/port/include -I/home/safdar [... cut ...]
    ../main/main.c:82:40: warning: extra tokens at end of #ifdef directive
     #ifdef CONFIG_MICRO_ROS_ESP_NETIF_WLAN || CONFIG_MICRO_ROS_ESP_NETIF_ENET
                                            ^~
    cc1: some warnings being treated as errors
    [1014/1022] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_core/ed25519/ref10/ed25519_ref10.c.obj
    ninja: build stopped: subcommand failed.
    ninja failed with exit code 1
```

#### Additional information

safdark, 2022-02-10:
However, by following the same steps above for the 'int32_publisher' example, there are no compilation errors.

pablogs9, 2022-02-10:
Solved here: https://github.com/micro-ROS/micro_ros_espidf_component/pull/128. Please retry when merged.

pablogs9, 2022-02-10:
#128 merged, please report if all works ok

pablogs9, 2022-03-18:
Any update on this?

pablogs9, 2022-04-07:
Closing due to inactivity.

closed 2022-04-07 by pablogs9