micro-ROS/micro_ros_espidf_component #131: Cannot build int32_publisher example.

SourabhSPrasad, 2022-02-11:
- Hardware description: Docker with espressif/idf image and ESP32-C3
- RTOS: freeRTOS
- Installation type: ESP-IDF
- Version or commit hash: galactic

#### Steps to reproduce the issue
1. Sourced IDF
```
root@G14:/# cd /opt/esp/idf
root@G14:/opt/esp/idf# . $IDF_PATH/export.sh
Detecting the Python interpreter
Checking "python" ...
Python 3.8.10
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /opt/esp/python_env/idf5.0_py3.8_env/bin/python
Checking if Python packages are up to date...
Skipping the download of /root/.espressif/espidf.constraints.v5.0.txt because it was downloaded recently. If you believe that this is causing you trouble then remove it manually and re-run your install script.
Constraint file: /root/.espressif/espidf.constraints.v5.0.txt
Requirement files:
 - /opt/esp/idf/requirements.core.txt
Python being checked: /opt/esp/python_env/idf5.0_py3.8_env/bin/python
Python requirements are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```
2. Installed dependencies 
```
Successfully installed argcomplete-2.0.0 attrs-21.4.0 catkin_pkg-0.4.24 colcon-argcomplete-0.3.3 colcon-bash-0.4.2 colcon-cd-0.1.1 colcon-cmake-0.2.26 colcon-common-extensions-0.2.1 colcon-core-0.7.1 colcon-defaults-0.2.6 colcon-devtools-0.2.3 colcon-library-path-0.2.1 colcon-metadata-0.2.5 colcon-notification-0.2.13 colcon-output-0.2.12 colcon-package-information-0.3.3 colcon-package-selection-0.2.10 colcon-parallel-executor-0.2.4 colcon-pkg-config-0.1.0 colcon-powershell-0.3.7 colcon-python-setup-py-0.2.7 colcon-recursive-crawl-0.2.1 colcon-ros-0.3.22 colcon-test-result-0.3.8 colcon-zsh-0.4.0 coverage-6.3.1 distlib-0.3.4 docutils-0.18.1 empy-3.3.4 importlib-resources-5.4.0 iniconfig-1.1.1 lark-parser-0.12.0 notify2-0.3.1 packaging-21.3 pluggy-1.0.0 py-1.11.0 pytest-7.0.0 pytest-cov-3.0.0 pytest-repeat-0.9.1 pytest-rerunfailures-10.2 python-dateutil-2.8.2 tomli-2.0.1 zipp-3.7.0
```
3. Cloned repo to /opt/esp/idf/components
```
git clone https://github.com/micro-ROS/micro_ros_espidf_component
Cloning into 'micro_ros_espidf_component'...
remote: Enumerating objects: 703, done.
remote: Counting objects: 100% (420/420), done.
remote: Compressing objects: 100% (266/266), done.
remote: Total 703 (delta 269), reused 248 (delta 149), pack-reused 283
Receiving objects: 100% (703/703), 187.30 KiB | 792.00 KiB/s, done.
Resolving deltas: 100% (400/400), done.
```
4. Set target as esp32c3 and ran menuconfig
```
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/esp/idf/components/micro_ros_espidf_component/examples/int32_publisher/build
```
```
WARNING: Menuconfig may fail because of the TERM environment variable is set to "xterm". Please consult the documentation of your terminal to set it up. Some good, proved to been working setups include xterm-256color, screen-256color, rxvt-unicode-256color.
Loaded configuration '/opt/esp/idf/components/micro_ros_espidf_component/examples/int32_publisher/sdkconfig'
No changes to save (for '/opt/esp/idf/components/micro_ros_espidf_component/examples/int32_publisher/sdkconfig')
Loading defaults file /opt/esp/idf/components/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults...
```
5. Ran idf.py build

#### Expected behavior
idf.py build completes without any errors

#### Actual behavior
The following error was seen.
```
In file included from /opt/esp/idf/components/lwip/port/esp32/include/lwipopts.h:17,
                 from /opt/esp/idf/components/lwip/lwip/src/include/lwip/opt.h:51,
                 from /opt/esp/idf/components/lwip/lwip/src/include/lwip/sockets.h:42,
                 from /opt/esp/idf/components/lwip/port/esp32/include/sys/socket.h:33,
                 from /opt/esp/idf/components/micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/udp/udp_transport_posix_nopoll.c:5:
/opt/esp/idf/components/esp_system/include/esp_task.h:32:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:412: CMakeFiles/microxrcedds_client.dir/src/c/profile/transport/ip/udp/udp_transport_posix_nopoll.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
In file included from /opt/esp/idf/components/lwip/port/esp32/include/lwipopts.h:17,
                 from /opt/esp/idf/components/lwip/lwip/src/include/lwip/opt.h:51,
                 from /opt/esp/idf/components/lwip/lwip/src/include/lwip/sockets.h:42,
                 from /opt/esp/idf/components/lwip/port/esp32/include/sys/socket.h:33,
                 from /opt/esp/idf/components/micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/profile/discovery/transport/udp_transport_datagram_posix_nopoll.c:4:
/opt/esp/idf/components/esp_system/include/esp_task.h:32:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:454: CMakeFiles/microxrcedds_client.dir/src/c/profile/discovery/transport/udp_transport_datagram_posix_nopoll.c.obj] Error 1
In file included from /opt/esp/idf/components/lwip/port/esp32/include/lwipopts.h:17,
                 from /opt/esp/idf/components/lwip/lwip/src/include/lwip/opt.h:51,
                 from /opt/esp/idf/components/lwip/lwip/src/include/lwip/inet.h:43,
                 from /opt/esp/idf/components/lwip/port/esp32/include/arpa/inet.h:18,
                 from /opt/esp/idf/components/micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/ip_posix.c:18:
/opt/esp/idf/components/esp_system/include/esp_task.h:32:10: fatal error: freertos/FreeRTOS.h: No such file or directory
 #include "freertos/FreeRTOS.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:426: CMakeFiles/microxrcedds_client.dir/src/c/profile/transport/ip/ip_posix.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:83: CMakeFiles/microxrcedds_client.dir/all] Error 2
make[1]: *** [Makefile:136: all] Error 2
---
Failed   <<< microxrcedds_client [1.72s, exited with code 2]
Aborted  <<< rosidl_adapter [1.74s]
Aborted  <<< rmw [0.09s]
Aborted  <<< rcl_logging_noop [0.03s]

Summary: 13 packages finished [4.68s]
  1 package failed: microxrcedds_client
  3 packages aborted: rcl_logging_noop rmw rosidl_adapter
  9 packages had stderr output: microxrcedds_client rcl_logging_interface rcutils ros2trace rosidl_cli rosidl_runtime_c tracetools_launch tracetools_read tracetools_trace
  45 packages not processed
make: *** [libmicroros.mk:87: /opt/esp/idf/components/micro_ros_espidf_component/micro_ros_src/install] Error 2
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
#### Additional information
idf.py build commands works with without micro_ros_espidf_component added.

pablogs9, 2022-02-14:
Which version of ESP-IDF are you using?

SourabhSPrasad, 2022-02-14:
IDF version
```
root@G14:/# idf.py --version
ESP-IDF v5.0-dev-1538-g51bfd280f8
```

pablogs9, 2022-02-14:
ESP-IDF v5.0-dev is not supported as stated in this repo [README.md](https://github.com/micro-ROS/micro_ros_espidf_component#micro-ros-component-for-esp-idf)

The last stable release v4.4 is the supported one: https://github.com/espressif/esp-idf/releases

Please reopen if you find any problem using the supported versions

closed 2022-02-14 by pablogs9