micro-ROS/micro_ros_espidf_component #133: Example docker image unable to start building due to locale

AntumArk, 2022-02-28:
## Issue template

- Hardware description: Ubuntu 20.04.4 VM. 
- RTOS: None
- Installation type: Docker example
- Version or commit hash: galactic

#### Steps to reproduce the issue

1. Set up Ubuntu with locale lt_LT.UTF8 . ( Mine was set up while installing and selecting Vilnius as location)
2. Run `docker run -it --rm --user espidf --volume="/etc/timezone:/etc/timezone:ro" -v  $(pwd):/micro_ros_espidf_component -v  /dev:/dev --privileged --workdir /micro_ros_espidf_component microros/esp-idf-microros:latest /bin/bash  -c "cd examples/int32_publisher; idf.py menuconfig build flash monitor"` from example. [Example readme](https://github.com/micro-ROS/micro_ros_espidf_component/blob/eb6b9b7545ceb8da05eb99555a27e1f355f77387/README.md)

#### Expected behavior

- Docker image successfully pulled
- example project starts building

#### Actual behavior

- Docker image successfully pulled

```
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

/bin/bash: line 0: cd: examples/int32_publisher: No such file or directory
Your environment is not configured to handle unicode filenames outside of ASCII range. Environment variable LC_ALL is temporary set to C.UTF-8 for unicode support.
Executing action: menuconfig
CMakeLists.txt not found in project directory /micro_ros_espidf_component
```

#### Additional information

pablogs9, 2022-02-28:
Are you executing the docker in this repo folder? I have done:

```bash
git clone -b galactic https://github.com/micro-ROS/micro_ros_espidf_component
cd micro_ros_espidf_component
docker run -it --rm --user espidf --volume="/etc/timezone:/etc/timezone:ro" -v $(pwd):/micro_ros_espidf_component -v /dev:/dev --privileged --workdir /micro_ros_espidf_component microros/esp-idf-microros:latest /bin/bash -c "cd examples/int32_publisher; idf.py menuconfig build flash monitor"
```

Check here: https://asciinema.org/a/OtfGLGfELIMzhpaVUmuGANYMu (the final error is bc I had no ESP32 connected)

AntumArk, 2022-02-28:
Yes, that seem to be the case that i was not running it from repo folder. I also tried using the same snipped in Windows 
`micro_ros_espidf_component>docker run -it --rm --user espidf --volume="/etc/timezone:/etc/timezone:ro" -v $(pwd):/micro_ros_espidf_component -v /dev:/dev --privileged --workdir /micro_ros_espidf_component microros/esp-idf-microros:latest /bin/bash -c "cd examples/int32_publisher; idf.py menuconfig build flash monitor"
docker: Error response from daemon: create $(pwd): "$(pwd)" includes invalid characters for a local volume name, only "[a-zA-Z0-9][a-zA-Z0-9_.-]" are allowed. If you intended to pass a host directory, use absolute path.
See 'docker run --help'.`

Is this not compatible with windows?

pablogs9, 2022-02-28:
Not tested in windows. Is this solved for the original question?

AntumArk, 2022-02-28:
Yes, it did. Thank you!

closed 2022-02-28 by AntumArk