micro-ROS/micro_ros_espidf_component #134: How can I add a existing ROS2 package to my esp project and use it?

zzMiao123, 2022-03-08:
- Hardware description: esp32
- RTOS: FreeRTOS
- Installation type: micro_ros_setup, robot_state_publisher
- Version or commit hash: foxy

#### Steps to reproduce the issue

#### Expected behavior
ESP32 runs a launch.py file of "robot_state_publisher" package in the app_main function.

#### Actual behavior
I dont know how to apply a package for my project.
clone the package into component folder, and then what to do?

#### Additional information
please let me know if there is any useful tutorial. Thank you very much!

pablogs9, 2022-03-08:
Hi, you can add extra packages to the [extra packages folder](https://github.com/micro-ROS/micro_ros_espidf_component/tree/galactic/extra_packages), packages in this folder will be cross-compiled along with the micro-ROS stack and you will be able to run the in the MCU.

But those packages should be compatible with the MCU you are working on. I guess that by now a launch.py Python script won't work with micro-ROS. In any case, you can check the functionality of `robot_state_publisher` and replicate it using RCLC in micro-ROS.

Please reopen if you have more questions.

closed 2022-03-08 by pablogs9

zzMiao123, 2022-03-08:
Thanks, but without launch.py how can I "ros2 run <package>" in my MCU?

pablogs9, 2022-03-08:
Please take a look at our [tutorials for RTOS](https://micro.ros.org/docs/tutorials/core/first_application_rtos/). micro-ROS do not have a CLI interface, it just provides a set of libraries to write ROS 2 compatible embedded code.

zzMiao123, 2022-03-08:
OMG. Does that mean I cannot run executable files of packages in ROS2 community? I cannot use proven resources in ROS2 in my MCU for now, but only use rcl and rclc to create my program?
Sorry, I can't think of any other way to use the package's functions than to run the package directly.

hellas2000, 2022-06-13:
I'm interested in your situation too, have you sucessfully done your job?
If not, I guess maybe you need to copy the code from that package and add it to your project, and start a new node in your code to perform the function of that package.