micro-ROS/micro_ros_espidf_component #135: int32_publisher build error

huarunlin, 2022-03-18:
## Issue template

Hardware description: ESP32S2
RTOS: freertos
Installation type:
Version or commit hash: ESP-IDF 4.2

- Hardware description: esp32
- RTOS: freertos
- Installation type: 
- Version or commit hash: micro-ROS used: galactic, ESP-IDF 4.4

#### Steps to reproduce the issue

1. install ESP-IDF 4.4
2. git clone micro_ros_espidf_component, branch: galactic
3. . $IDF_PATH/export.sh
4. cd examples/int32_publisher
5. idf.py set-target esp32
6. edit sdkconfig.defaults, configuration CONFIG_ESP_WIFI_SSID, CONFIG_ESP_WIFI_PASSWORK.
7. idf.py build

```
<command-line>: note: this is the location of the previous definition
<command-line>: note: this is the location of the previous definition
In file included from /opt/esp-idf/v4.4/esp-idf-v4.4/components/lwip/lwip/src/include/lwip/opt.h:51,
                 from /opt/esp-idf/v4.4/esp-idf-v4.4/components/lwip/lwip/src/include/lwip/sys.h:40,
                 from /opt/micro-ROS/micro_ros_espidf_component/micro_ros_src/src/embeddedRTPS/include/rtps/ThreadPool.h:28,
                 from /opt/micro-ROS/micro_ros_espidf_component/micro_ros_src/src/embeddedRTPS/include/rtps/entities/Domain.h:28,
                 from /opt/micro-ROS/micro_ros_espidf_component/micro_ros_src/src/embeddedRTPS/include/rtps/rtps.h:28,
                 from /opt/micro-ROS/micro_ros_espidf_component/micro_ros_src/src/embeddedRTPS/src/rtps.cpp:25:
/opt/esp-idf/v4.4/esp-idf-v4.4/components/lwip/port/esp32/include/lwipopts.h:768: warning: "LWIP_IPV6" redefined
 #define LWIP_IPV6                       CONFIG_LWIP_IPV6
 
<command-line>: note: this is the location of the previous definition
---
Finished <<< embeddedrtps [6.10s]
Starting >>> rcl_logging_noop
--- stderr: rcl_yaml_param_parser
CMake Warning at /opt/micro-ROS/micro_ros_espidf_component/micro_ros_src/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /opt/micro-ROS/micro_ros_espidf_component/micro_ros_src/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  CMakeLists.txt:6 (find_package)

CMake Error at CMakeLists.txt:8 (find_package):
  By not providing "Findlibyaml_vendor.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "libyaml_vendor", but CMake did not find one.

  Could not find a package configuration file provided by "libyaml_vendor"
  with any of the following names:

    libyaml_vendorConfig.cmake
    libyaml_vendor-config.cmake

  Add the installation prefix of "libyaml_vendor" to CMAKE_PREFIX_PATH or set
  "libyaml_vendor_DIR" to a directory containing one of the above files.  If
  "libyaml_vendor" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< rcl_yaml_param_parser [0.98s, exited with code 1]
Aborted  <<< rcl_logging_log4cxx [0.86s]
Aborted  <<< rcl_logging_noop [0.68s]
Aborted  <<< rosidl_cmake [1.46s]

Summary: 13 packages finished [11.8s]
  1 package failed: rcl_yaml_param_parser
  3 packages aborted: rcl_logging_log4cxx rcl_logging_noop rosidl_cmake
  10 packages had stderr output: embeddedrtps microxrcedds_client rcl_logging_interface rcl_logging_log4cxx rcl_logging_noop rcl_yaml_param_parser rcutils rmw rosidl_cli rosidl_runtime_c
  51 packages not processed
make: *** [libmicroros.mk:51: /opt/micro-ROS/micro_ros_espidf_component/micro_ros_src/install] Error 1
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

pablogs9, 2022-03-18:
At some point in the configuration, you are selecting EmbeddedRTPS as default middleware. Is this correct?

huarunlin, 2022-03-18:
> ## Issue template
> Hardware description: ESP32S2 RTOS: freertos Installation type: Version or commit hash: ESP-IDF 4.2
>
> * Hardware description: esp32
> * RTOS: freertos
> * Installation type:
> * Version [...]

huarunlin, 2022-03-18:
@pablogs9 , I checked the configuration and the eProsima Micro XRCE-DDS was selected. But I clone embeddedRTPS, rmw_embeddedrtps in micro_ros_src/src directory. This may be a problem.

huarunlin, 2022-03-18:
@pablogs9 , I removed embeddedRTPS, rmw_embeddedrtps.build got the following error

```
Executing action: all (aliases: build)
Running ninja in directory /opt/micro-ROS/micro_ros_espidf_component/examples/int32_publisher/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /opt/micro-ROS/micro_ros_espidf_component/examples/int32_publisher/sdkconfig
Loading defaults file /opt/micro-ROS/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults...
-- App "int32_publisher" version: eb6b9b7-dirty
-- Adding linker script /opt/micro-ROS/micro_ros_espidf_component/examples/int32_publisher/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /opt/esp-idf/v4.4/esp-idf-v4.4/components/soc/esp32/ld/esp32.peripherals.ld
-- Using Micro XRCE-DDS middleware
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns micro_ros_espidf_component mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /opt/esp-idf/v4.4/esp-idf-v4.4/components/app_trace /opt/esp-idf/v4.4/esp-idf-v4.4/components/app_update /opt/esp-idf/v4.4/esp-idf-v4.4/components/asio /opt/esp-idf/v4.4/esp-idf-v4.4/components/bootloader /opt/esp-idf/v4.4/esp-idf-v4.4/components/bootloader_support /opt/esp-idf/v4.4/esp-idf-v4.4/components/bt /opt/esp-idf/v4.4/esp-idf-v4.4/components/cbor /opt/esp-idf/v4.4/esp-idf-v4.4/components/cmock /opt/esp-idf/v4.4/esp-idf-v4.4/components/coap /opt/esp-idf/v4.4/esp-idf-v4.4/components/console /opt/esp-idf/v4.4/esp-idf-v4.4/components/cxx /opt/esp-idf/v4.4/esp-idf-v4.4/components/driver /opt/esp-idf/v4.4/esp-idf-v4.4/components/efuse /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp-tls /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp32 /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_adc_cal /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_common /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_eth /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_event /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_gdbstub /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_hid /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_http_client /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_http_server /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_https_ota /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_https_server /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_hw_support /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_ipc /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_lcd /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_local_ctrl /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_netif /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_phy /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_pm /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_ringbuf /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_rom /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_serial_slave_link /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_system /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_timer /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_websocket_client /opt/esp-idf/v4.4/esp-idf-v4.4/components/esp_wifi /opt/esp-idf/v4.4/esp-idf-v4.4/components/espcoredump /opt/esp-idf/v4.4/esp-idf-v4.4/components/esptool_py /opt/esp-idf/v4.4/esp-idf-v4.4/components/expat /opt/esp-idf/v4.4/esp-idf-v4.4/components/fatfs /opt/esp-idf/v4.4/esp-idf-v4.4/components/freemodbus /opt/esp-idf/v4.4/esp-idf-v4.4/components/freertos /opt/esp-idf/v4.4/esp-idf-v4.4/components/hal /opt/esp-idf/v4.4/esp-idf-v4.4/components/heap /opt/esp-idf/v4.4/esp-idf-v4.4/components/idf_test /opt/esp-idf/v4.4/esp-idf-v4.4/components/ieee802154 /opt/esp-idf/v4.4/esp-idf-v4.4/components/jsmn /opt/esp-idf/v4.4/esp-idf-v4.4/components/json /opt/esp-idf/v4.4/esp-idf-v4.4/components/libsodium /opt/esp-idf/v4.4/esp-idf-v4.4/components/log /opt/esp-idf/v4.4/esp-idf-v4.4/components/lwip /opt/micro-ROS/micro_ros_espidf_component/examples/int32_publisher/main /opt/esp-idf/v4.4/esp-idf-v4.4/components/mbedtls /opt/esp-idf/v4.4/esp-idf-v4.4/components/mdns /opt/micro-ROS/micro_ros_espidf_co [... cut ...]
Aborted  <<< rcl_logging_noop [1.14s]

Summary: 10 packages finished [8.14s]
  1 package failed: rcl_logging_log4cxx
  3 packages aborted: rcl_logging_noop rmw rosidl_parser
  8 packages had stderr output: microxrcedds_client rcl_logging_interface rcl_logging_log4cxx rcl_logging_noop rcutils rmw rosidl_cli rosidl_runtime_c
  52 packages not processed
make: *** [libmicroros.mk:51: /opt/micro-ROS/micro_ros_espidf_component/micro_ros_src/install] Error 1
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

pablogs9, 2022-03-18:
Please try to do it in a clean environment. I have replicated your scenario with no problem, check here: https://asciinema.org/a/EV8cCZTwjbzE3XoiPlwVNkW1C

pablogs9, 2022-03-28:
Any update on this?

huarunlin, 2022-03-29:
> Any update on this?

I rebuilt the environment and it compiled successfully.

closed 2022-03-29 by huarunlin

VineetTambe, 2023-03-12:
Hey I am facing this exact issue - what do you mean by "rebuilt the environment?" 
I am new to the entire ESP-idf workflow so I find myself pretty lost with this. 
Any help would be appreciated :)

I have esp-idf v4.4.4 and a ESP32. My setup is a Ubuntu 22.04 I have not sourced my ROS 2 
This is the error I am facing:
 ```
[100%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
/home/vrex/idf/esp-idf/components/micro_ros_espidf_component/examples/int32_publisher/main/main.c:11:10: fatal error: rcl/rcl.h: No such file or directory
 #include <rcl/rcl.h>
          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:76: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:4891: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make failed with exit code 2
```

Is it some issue with the ```libmicroros.mk``` ???