micro-ROS/micro_ros_espidf_component #146: Service callback

SourabhSPrasad, 2022-05-05:
## Issue template

- Hardware description: ESP32-C3 DevKitM-1
- RTOS: FreeRTOS
- Version or commit hash: Galactic

Hi, I am trying to create a service server on an ESP32-C3. The ESP is connected to the agent via serial. I combined the int32_publisher_custom_transport and addtwoints_server to create the program. 
```
#include <string.h>
#include <stdio.h>
#include <unistd.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_system.h"
#include "driver/uart.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <std_srvs/srv/trigger.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <rmw_microxrcedds_c/config.h>
#include <rmw_microros/rmw_microros.h>
#include "esp32_serial_transport.h"

#define DOMAIN_ID 3
static char res_message = '\0';

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

void service_callback(const void * req, void * res)
{
  std_srvs__srv__Trigger_Request * req_in = (std_srvs__srv__Trigger_Request *) req;
  std_srvs__srv__Trigger_Response * res_in = (std_srvs__srv__Trigger_Response *) res;
  res_in->success = true;
}

void micro_ros_task(void * arg)
{
	rcl_allocator_t allocator = rcl_get_default_allocator();
	rclc_support_t support;

	// Create init_options.
	rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
	RCCHECK(rcl_init_options_init(&init_options, allocator));
	RCCHECK(rcl_init_options_set_domain_id(&init_options, DOMAIN_ID));
	RCCHECK(rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator));

	// create node
	rcl_node_t node;
	RCCHECK(rclc_node_init_default(&node, "test_node", "", &support));

	rcl_service_t service;
	RCCHECK(rclc_service_init_default(&service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(std_srvs, srv, Trigger), "/test_service"));

	// create executor
	rclc_executor_t executor;
	RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator));

	std_srvs__srv__Trigger_Response res;
	res.message.data = &res_message;
	std_srvs__srv__Trigger_Request req;
	RCCHECK(rclc_executor_add_service(&executor, &service, &req, &res, &service_callback));

	while(1){
		rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
		usleep(100000);
	}

	RCCHECK(rcl_service_fini(&service, &node));
	RCCHECK(rcl_node_fini(&node));
}

static size_t uart_port = UART_NUM_0;

void app_main(void)
{
	#if defined(RMW_UXRCE_TRANSPORT_CUSTOM)
		rmw_uros_set_custom_transport(
			true,
			(void *) &uart_port,
			esp32_serial_open,
			esp32_serial_close,
			esp32_serial_write,
			esp32_serial_read
		);
	#else
	#error micro-ROS transports misconfigured
	#endif  // RMW_UXRCE_TRANSPORT_CUSTOM

	xTaskCreate(micro_ros_task,
			"uros_task",
			CONFIG_MICRO_ROS_APP_STACK,
			NULL,
			CONFIG_MICRO_ROS_APP_TASK_PRIO,
			NULL);
}

```
The program builds fine and the service is visible on the command-line.
When I sent a request via the command-line using
```
ros2 service call /test_service std_srvs/srv/Trigger {}
```
The callback function is not being called and thus no response is given. The request is being received as I can see it within the micro_ros_agents output. 

#### Expected behavior
Callback function is called and response is given.

#### Actual behavior
No response given.

Acuadros95, 2022-05-09:
You need to configure the response string capacity as 1:
```
res.message.data = &res_message;
res.message.capacity = 1;
res.message.size = 0;
```

This way the terminate character will be processed correctly.
Please give this a quick test, if the problem persist we will try to replicate on our side.

SourabhSPrasad, 2022-05-09:
Yeah did the trick. Thank you!

closed 2022-05-09 by SourabhSPrasad