micro-ROS/micro_ros_espidf_component #148: Can't build addtwoints_server

SamyFrancelet, 2022-06-22:
## Can't build addtwoints_server
The component can't find the "example_interfaces/srv/add_two_ints.h" file, I also tried changing the "" with <> and nothing changes.

- Hardware description: ESP32 on an ESP32_DevKitC_V4 board
- RTOS: FreeRTOS
- Installation type: fully working esp-idf environement with int32_pub_sub and int32_publisher working, full ROS2 Desktop Humble env
- Version or commit hash: microROS with ROS2 Humble

#### Steps to reproduce the issue
Steps from the Example on the REAMDE.MD, except with addtwoints_server

#### Expected behavior
idf.py should build properly

#### Actual behavior
Error : /home/samy/work/micro_ros_espidf_component/examples/addtwoints_server/main/main.c:17:10: fatal error: example_interfaces/srv/add_two_ints.h: No such file or directory
 #include "example_interfaces/srv/add_two_ints.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

#### Additional information
Full console output : 
samy@ros-workstation:~/work/micro_ros_espidf_component/examples/addtwoints_server$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/samy/work/micro_ros_espidf_component/examples/addtwoints_server/build
Executing "ninja all"...
[0/2] Re-checking globbed directories...
[1/10] Performing build step for 'bootloader'
[1/1] cd /home/samy/work/micro_ros_espidf_component/examples/addtwoints_server/build/bootloader/esp-idf/esptool_py && /home/samy/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/samy/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/samy/work/micro_ros_espidf_component/examples/addtwoints_server/build/bootloader/bootloader.bin
Bootloader binary size 0x62c0 bytes. 0xd40 bytes (12%) free.
[2/8] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj 
/home/samy/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -I/home/samy/work/micro_ros_espidf_component/examples/addtwoints_server/build/config -I/home/samy/esp/esp-idf/components/newlib/platform_include -I/home/samy/esp/esp-idf/components/freertos/include -I/home/samy/esp/esp-idf/components/freertos/include/esp_additions/freertos -I/home/samy/esp/esp-idf/components/freertos/port/xtensa/include -I/home/samy/esp/esp-idf/components/freertos/include/esp_additions -I/home/samy/esp/esp-idf/components/esp_hw_support/include -I/home/samy/esp/esp-idf/components/esp_hw_support/include/soc -I/home/samy/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/samy/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/samy/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/home/samy/esp/esp-idf/components/heap/include -I/home/samy/esp/esp-idf/components/log/include -I/home/samy/esp/esp-idf/components/lwip/include/apps -I/home/samy/esp/esp-idf/components/lwip/include/apps/sntp -I/home/samy/esp/esp-idf/components/lwip/lwip/src/include -I/home/samy/esp/esp-idf/components/lwip/port/esp32/include -I/home/samy/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/samy/esp/esp-idf/components/soc/include -I/home/samy/esp/esp-idf/components/soc/esp32/. -I/home/samy/esp/esp-idf/components/soc/esp32/include -I/home/samy/esp/esp-idf/components/hal/esp32/include -I/home/samy/esp/esp-idf/components/hal/include -I/home/samy/esp/esp-idf/components/hal/platform_port/include -I/home/samy/esp/esp-idf/components/esp_rom/include -I/home/samy/esp/esp-idf/components/esp_rom/include/esp32 -I/home/samy/esp/esp-idf/components/esp_rom/esp32 -I/home/samy/esp/esp-idf/components/esp_common/include -I/home/samy/esp/esp-idf/components/esp_system/include -I/home/samy/esp/esp-idf/components/esp_system/port/soc -I/home/samy/esp/esp-idf/components/esp_system/port/public_compat -I/home/samy/esp/esp-idf/components/esp32/include -I/home/samy/esp/esp-idf/components/xtensa/include -I/home/samy/esp/esp-idf/components/xtensa/esp32/include -I/home/samy/esp/esp-idf/components/driver/include -I/home/samy/esp/esp-idf/components/driver/esp32/include -I/home/samy/esp/esp-idf/components/esp_pm/include -I/home/samy/esp/esp-idf/components/esp_ringbuf/include -I/home/samy/esp/esp-idf/components/efuse/include -I/home/samy/esp/esp-idf/components/efuse/esp32/include -I/home/samy/esp/esp-idf/components/vfs/include -I/home/samy/esp/esp-idf/components/esp_wifi/include -I/home/samy/esp/esp-idf/components/esp_event/include -I/home/samy/esp/esp-idf/components/esp_netif/include -I/home/samy/esp/esp-idf/components/esp_eth/include -I/home/samy/esp/esp-idf/components/tcpip_adapter/include -I/home/samy/esp/esp-idf/components/esp_phy/include -I/home/samy/esp/esp-idf/components/esp_phy/esp32/include -I/home/samy/esp/esp-idf/components/esp_ipc/include -I/home/samy/esp/esp-idf/components/app_trace/include -I/home/samy/esp/esp-idf/components/esp_timer/include -I/home/samy/esp/esp-idf/components/mbedtls/port/include -I/home/samy/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/samy/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/samy/esp/esp-idf/components/app_update/include -I/home/samy/esp/esp-idf/components/spi_flash/include -I/home/samy/esp/esp-idf/components/bootloader_support/include -I/home/samy/esp/esp-idf/components/nvs_flash/include -I/home/samy/esp/esp-idf/components/pthread/include -I/home/samy/esp/esp-idf/components/esp_gdbstub/include -I/home/samy/esp/esp-idf/components/esp_gdbstub/xtensa -I/home/samy/esp/esp-idf/components/esp_gdbstub/esp32 -I/home/samy/esp/esp-idf/components/espcoredump/include -I/home/samy/esp/esp-idf/components/espcoredump/include/port/xtensa -I/home/samy/esp/esp-idf/components/wpa_supplicant/include -I/home/samy/esp/esp-idf/components/wpa_supplicant/port/include -I/home/samy/esp/esp-idf/components/wpa_supplicant/esp_supplicant/include -I/home/samy/esp/esp-idf/components/ieee802154/include -I/home/samy/esp/esp-idf/components/console -I/home/samy/esp/esp-idf/components/asio/asio/asio/include -I/home/samy/esp/esp-idf/components/asio/port/include -I/home/samy/esp/esp-idf/components/cbor/port/include -I/home/samy/esp/esp-idf/components/unity/include -I/home/samy/esp/esp-idf/components/unity/unity/src -I/home/samy/esp/esp-idf/components/cmock/CMock/src -I/home/samy/esp/esp-idf/components/coap/port/include -I/home/samy/esp/esp-idf/components/coap/libcoap/include -I/home/samy/esp/esp-idf/components/nghttp/port/include -I/home/samy/esp/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/samy/esp/esp-idf/components/esp-tls -I/home/samy/esp/esp-idf/components/esp-tls/esp-tls-crypto -I/home/samy/esp/esp-idf/components/esp_adc_cal/include -I/home/samy/esp/esp-idf/components/esp_hid/include -I/home/samy/esp/esp-idf/components/tcp_transport/include -I/home/samy/esp/esp-idf/components/esp_http_client/include -I/home/samy/esp/esp-idf/components/esp_http_server/include -I/home/samy/esp/esp-idf/components/esp_https_ota/include -I/home/samy/esp/esp-idf/components/esp_lcd/include -I/home/samy/esp/esp-idf/components/esp_lcd/interface -I/home/samy/esp/esp-idf/components/protobuf-c/protobuf-c -I/home/samy/esp/esp-idf/components/protocomm/include/common -I/home/samy/esp/esp-idf/components/protocomm/include/security -I/home/samy/esp/esp-idf/components/protocomm/include/transports -I/home/samy/esp/esp-idf/components/mdns/include -I/home/samy/esp/esp-idf/components/esp_local_ctrl/include -I/home/samy/esp/esp-idf/components/sdmmc/include -I/home/samy/esp/esp-idf/components/esp_serial_slave_link/include -I/home/samy/esp/esp-idf/components/esp_websocket_client/include -I/home/samy/esp/esp-idf/components/expat/expat/expat/lib -I/home/samy/esp/esp-idf/components/expat/port/include -I/home/samy/esp/esp-idf/components/wear_levelling/include -I/home/samy/esp/esp-idf/components/fatfs/diskio -I/home/samy/esp/esp-idf/components/fatfs/vfs -I/home/samy/esp/esp-idf/components/fatfs/src -I/home/samy/esp/esp-idf/components/freemodbus/common/include -I/home/samy/esp/esp-idf/components/idf_test/include -I/home/samy/esp/esp-idf/components/idf_test/include/esp32 -I/home/samy/esp/esp-idf/components/jsmn/include -I/home/samy/esp/esp-idf/components/json/cJSON -I/home/samy/esp/esp-idf/components/libsodium/libsodium/src/libsodium/include -I/home/samy/esp/esp-idf/components/libsodium/port_include -I/home/samy/esp/e [... cut ...]  -o esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj -c /home/samy/work/micro_ros_espidf_component/examples/addtwoints_server/main/main.c
/home/samy/work/micro_ros_espidf_component/examples/addtwoints_server/main/main.c:17:10: fatal error: example_interfaces/srv/add_two_ints.h: No such file or directory
 #include "example_interfaces/srv/add_two_ints.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

pablogs9, 2022-06-22:
Let me know if this solves the issue so we can close it

pablogs9, 2022-06-22:
You need to add it to `extra_packages` and the build system will add this package in the library. Then when the library is built you will be able to use `#include <my_package/....>` in your app code.

closed 2022-06-22 by SamyFrancelet