micro-ROS/micro_ros_espidf_component #151: Service callback never called with a custom interfaces using strings

SamyFrancelet, 2022-07-12:
## Service callback never called with a custom interfaces using strings

- Hardware description: ESP32
- RTOS: FreeRTOS
- Installation type: idf component
- Version or commit hash: humble

I made two custom interfaces, one typical AddThreeInts, and one ping : 
[image]

The callback method in both cases looks like that : 
```
void ping_callback(const void* request, void* response) {
    ESP_LOGI(TAG, "Received ping request");
}
```
When I create the service using my AddThreeInts or the example AddTwoInts, the callback is called properly.
But when I use my ping interfaces, it never goes into the callback...

#### Steps to reproduce the issue
How I create the service :
```
    // Create ping server with default config (Reliable QoS)
    const rosidl_service_type_support_t* type_support =
        ROSIDL_GET_SRV_TYPE_SUPPORT(ble2ros_interfaces, srv, Ping);
    RCCHECK(rclc_service_init_default(&ping_srv, &node, type_support, ping_srv_name));

    // Create executor
    executor = rclc_executor_get_zero_initialized_executor();
    RCCHECK(rclc_executor_init(&executor, &support.context, 4, &allocator));

    // Add ping server to executor
    RCCHECK(rclc_executor_add_service(
            &executor,
            &ping_srv,
            &ping_req,
            &ping_rsp,
            ping_callback
    ));
```

pablogs9, 2022-07-12:
Could you share the `.srv` files of your custom interfaces?

SamyFrancelet, 2022-07-12:
```
# Sends a ping to the BT Mesh provisioner
# Receives the response from provisioner
string ping
---
string pong
```

SamyFrancelet, 2022-07-12:
And I must do it for both the request and response message ? The utility will know that those are service messages and do it automatically ?

pablogs9, 2022-07-12:
You need to do it for both messages, req and res. You can assign memory directly or use the utilities. In the case that is a simple string, maybe is easier to assign memory manually. You have instructions for both approaches in the link above.

Please provide feedback so we can close this issue.

SamyFrancelet, 2022-07-12:
Works like a charm, thanks a lot ! I assigned memory directly like that : (writing it in case someone find this issue too)
```
    // Initializing request and response msg
    ping_req.ping.capacity = 100;
    ping_req.ping.data = (char*) malloc(ping_req.ping.capacity * sizeof(char));

    ping_rsp.pong.capacity = 100;
    ping_rsp.pong.data = (char*) malloc(ping_rsp.pong.capacity * sizeof(char));
```

Quick last question, it doesn't appear possible to create two services with the ESP32, the second one always makes an error. I saw here :https://github.com/micro-ROS/micro_ros_arduino/issues/442 that the Teensy only supports one services. Is it also the case with the ESP32 ? (If not I will probably look further into that or make another issue)

pablogs9, 2022-07-12:
https://github.com/micro-ROS/micro_ros_espidf_component/blob/f426aa27e29984d8e8fa307c5442a6466cd2fa69/colcon.meta#L43