micro-ROS/micro_ros_espidf_component #152: How to do microROS secondary development with vscode+ESPIDF?

arilink, 2022-07-15:
I clone the micro_ros_espidF_component-humble on Linux and run a compile-burn test and it works. It's a good start, but I need to add microROS to my project, So I created a new project and set it in cmake (EXTRA_COMPONENT_DIRS "micro_ros_espidF_Component-humble ") but the compilation didn't pass How can I get all the source code for microROS and use microROS as a component?

pablogs9, 2022-07-15:
Basically, you have to clone this repo in your project `component` folder.

arilink, 2022-07-15:
I created the project, put micro_ros_espidF_component-humble in the Component folder, and overwrite the example code in the main.c file. Fatal error: uros_network_interfaces.h: No such file or directory is used in the RCL file in example, which is not found in micro_ros_espidF_component-humble.

pablogs9, 2022-07-15:
uros_network_interfaces.h is in this repo: https://github.com/micro-ROS/micro_ros_espidf_component/blob/humble/network_interfaces/uros_network_interfaces.h

arilink, 2022-07-15:
There is indeed uros_network_interfaces.h in micro_ros_espidf_component-humble. I also added this to the project. There is always a problem with compiling with vscode+espidf. Are you using vscode+espidf for development?

pablogs9, 2022-07-15:
> Are you using vscode+espidf for development?
Yes

arilink, 2022-07-15:
I feel that there is a problem with my project configuration, I will leave my email
Can you send me a test project? My development environment is vscode+esp idf 4.4.1

pablogs9, 2022-07-15:
We are not allowed to provide private support by this channel. If you need this kind of support just contact info@eProsima.com.

Another option is to generate a mock project (without your code if you don't want to share it) that has the same behavior and upload it directly here in the issue, so we can replicate it.

arilink, 2022-07-15:
[blink.zip](https://github.com/micro-ROS/micro_ros_espidf_component/files/9119792/blink.zip)

pablogs9, 2022-07-15:
I have modified the folder name from `micro_ros_espidf_component-humble` to `micro_ros_espidf_component`

And I have added to `blink/main/Kconfig.projbuild` the following:

```
menu "micro-ROS example-app settings"

    config MICRO_ROS_APP_STACK
        int "Stack the micro-ROS app (Bytes)"
        default 16000
        help
        Stack size in Bytes of the micro-ROS app

    config MICRO_ROS_APP_TASK_PRIO
        int "Priority of the micro-ROS app"
        default 5
        help
        Priority of micro-ros task higher value means higher priority

endmenu
```

And got it building doing `idf.py build`.

Please report if this solves your issue.

arilink, 2022-07-15:
no  still have error  --->ninja: error: '../components/micro_ros_espidf_component/libmicroros.a', needed by 'blink.elf', missing and no known rule to make it

pablogs9, 2022-07-15:
Are you running it in a clean environment?

arilink, 2022-07-15:
Yes, I created a simple led project with the micro ros component added

pablogs9, 2022-07-15:
And what is the difference between your simple led project and your compile-burn test?

arilink, 2022-07-15:
The simple project using led is to place the micro_ros_espidf_component in the component folder in the led project and compile it on windows (vscode+espidf) and the previous work was done under ubuntu

pablogs9, 2022-07-15:
Could you check your led project under ubuntu?

arilink, 2022-07-15:
The example project of the led project esp idf will not be a problem. It is also possible under ubuntu, but adding micro_ros_espidf_component as a component to the project and then compiling will fail

arilink, 2022-07-15:
Could you please use vscode+espidf to create a simple project and then compile micro_ros_espidf_component as a component into the project?

arilink, 2022-07-18:
@pablogs9  Hi, is there any progress here? I tried many times, but it still doesn't work

pablogs9, 2022-07-19:
I have created blink a project using vscode extension for ESP IDF and then I have been able to build it using `idf.py build` with no problems.

Please recheck your configuration. In any case, try to use de ESP-IDF CLI to verify that you are creating correctly the project.

closed 2022-07-19 by pablogs9

arilink, 2022-07-19:
Could you please package a project for me to compare and see what is wrong with the configuration? thank you very much

***@***.***
 
From: Pablo Garrido
Date: 2022-07-19 13:49
To: micro-ROS/micro_ros_espidf_component
CC: arilink; Author
Subject: Re: [micro-ROS/micro_ros_espidf_component] How to do microROS secondary development with vscode+ESPIDF? (Issue #152)
I have created blink a project using vscode extension for ESP IDF and then I have been able to build it using idf.py build with no problems.
Please recheck your configuration. In any case, try to use de ESP-IDF CLI to verify that you are creating correctly the project.

pablogs9, 2022-07-19:
[image]

[test_microros.zip](https://github.com/micro-ROS/micro_ros_espidf_component/files/9138030/test_microros.zip)

arilink, 2022-07-21:
hello,
    I'm sorry for not telling you the test results in time. I crashed the development environment while doing the test. In order to avoid the environment and configuration problems, I installed Ubuntu 22.04 LTS and ROS2 Humble ROS2. The publishing and subscription functions of ROS2 are normal, so the ROS2 environment is ok and the micro_ros_setup can be compiled normally. So the colcon build is normal, so I opened test_microROS and tried compiling according to your instructions and examples, but the problem still exists. Micro_ros_espidf_component still fails to compile. What should I do to solve this problem

    

***@***.***
 
From: Pablo Garrido
Date: 2022-07-19 14:19
To: micro-ROS/micro_ros_espidf_component
CC: arilink; Author
Subject: Re: [micro-ROS/micro_ros_espidf_component] How to do microROS secondary development with vscode+ESPIDF? (Issue #152)
test_microros.zip

arilink, 2022-10-11:
Ok,
    I will test it, thank you very much

***@***.***
 
From: Pablo Garrido
Date: 2022-07-19 14:19
To: micro-ROS/micro_ros_espidf_component
CC: arilink; Author
Subject: Re: [micro-ROS/micro_ros_espidf_component] How to do microROS secondary development with vscode+ESPIDF? (Issue #152)
test_microros.zip