micro-ROS/micro_ros_espidf_component #153: [humble] 1 package failed: ament_index_cpp due to arm64 architecture

anslex, 2022-07-25:
## Issue template

- Hardware description: arm64 Apple M1 Pro 
- RTOS: micro_ros_espidf_component/examples/int32_publisher
- Installation type: 
esp-idf, branch release/v4.4, commit 145c3cd000107733e24a56c8ed74e5620bebe1ad
micro_ros_espidf_component, branch humble, commit f426aa27e29984d8e8fa307c5442a6466cd2fa69
ament_cmake, branch humble, 1.3.2, commit d23ce2cf80d15a262f5395554ddddccad556f2e1
ament_index, branch humble, 1.4.0, commit f019d6c40991799a13b18c9c3dcc583e3fde0381
ament_cmake_ros, branch humble, 0.10.0, commit 60572fa1bec50b9e6fbe64e1b23640d21c15e9d0
ament_package, branch humble, 0.14.0, commit f8ea958fd02cff6f4192425e28566369c92b5e34

- Version or commit hash: humble f426aa27e29984d8e8fa307c5442a6466cd2fa69

Hello, 
Thank you for your work on micro-ROS. 
I am not sure if it is better to open an issue here or on ament_cmake git repo.

#### Steps to reproduce the issue
git clone -b humble --recursive https://github.com/micro-ROS/micro_ros_espidf_component.git
 . $IDF_PATH/export.sh   
cd examples/int32_publisher
 pip3 install catkin_pkg lark-parser empy colcon-common-extensions
idf.py set-target esp32 

#### Expected behavior
compile all packages

#### Actual behavior
stderr: ament_index_cpp Undefined symbols for architecture arm64
1 package failed: ament_index_cpp due to arm64 architecture
tested the same on amd64 chip and it worked but not on arm64

#### Additional information
Full log:

an@pro /tmp % git clone -b humble --recursive https://github.com/micro-ROS/micro_ros_espidf_component.git
Cloning into 'micro_ros_espidf_component'...
remote: Enumerating objects: 782, done.
remote: Counting objects: 100% (301/301), done.
remote: Compressing objects: 100% (91/91), done.
remote: Total 782 (delta 260), reused 226 (delta 210), pack-reused 481
Receiving objects: 100% (782/782), 428.85 KiB | 214.00 KiB/s, done.
Resolving deltas: 100% (466/466), done.
an@pro /tmp % cd micro_ros_espidf_component 
an@pro micro_ros_espidf_component % . $IDF_PATH/export.sh        

Detecting the Python interpreter
Checking "python" ...
/Users/an/esp/esp-idf/tools/detect_python.sh:16: command not found: python
Checking "python3" ...
Python 3.9.13
"python3" has been detected
Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.23.2.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
Using Python interpreter in /Users/an/.espressif/python_env/idf4.4_py3.9_env/bin/python
Checking if Python packages are up to date...
Python requirements from /Users/an/esp/esp-idf/requirements.txt are satisfied.
Updated PATH variable:
  /Users/an/esp/esp-idf/components/esptool_py/esptool:/Users/an/esp/esp-idf/components/espcoredump:/Users/an/esp/esp-idf/components/partition_table:/Users/an/esp/esp-idf/components/app_update:/Users/an/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/Users/an/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/Users/an/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/Users/an/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/Users/an/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/an/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/an/.espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Users/an/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin:/Users/an/.espressif/tools/ninja/1.10.2/:/Users/an/.espressif/python_env/idf4.4_py3.9_env/bin:/Users/an/esp/esp-idf/components/esptool_py/esptool:/Users/an/esp/esp-idf/components/espcoredump:/Users/an/esp/esp-idf/components/partition_table/:/Users/an/esp/esp-idf/tools/:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion Tech Preview.app/Contents/Public:/Library/Apple/usr/bin:/Users/an/Library/Python/3.8/bin:/Users/an/.local/bin:/Users/an/Library/Python/3.8/bin:/Users/an/.local/bin
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

an@pro micro_ros_espidf_component % cd examples/int32_publisher
an@pro int32_publisher % pip3 install catkin_pkg lark-parser empy colcon-common-extensions

Requirement already satisfied: catkin_pkg in /Users/an/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages (0.5.2)
[... 716 lines cut ...]
Summary: 59 packages finished [12.0s]
  1 package failed: ament_index_cpp
  59 packages had stderr output: ament_clang_format ament_clang_tidy ament_cmake ament_cmake_auto ament_cmake_clang_format ament_cmake_clang_tidy ament_cmake_copyright ament_cmake_core ament_cmake_cppcheck ament_cmake_cpplint ament_cmake_export_definitions ament_cmake_export_dependencies ament_cmake_export_include_directories ament_cmake_export_interfaces ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_export_targets ament_cmake_flake8 ament_cmake_gmock ament_cmake_google_benchmark ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_lint_cmake ament_cmake_mypy ament_cmake_nose ament_cmake_pclint ament_cmake_pep257 ament_cmake_pycodestyle ament_cmake_pyflakes ament_cmake_pytest ament_cmake_python ament_cmake_ros ament_cmake_target_dependencies ament_cmake_test ament_cmake_uncrustify ament_cmake_version ament_cmake_xmllint ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_cpp ament_index_python ament_lint ament_lint_auto ament_lint_cmake ament_lint_common ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint domain_coordinator gmock_vendor gtest_vendor
make: *** [/private/tmp/micro_ros_espidf_component/micro_ros_dev/install] Error 2
make: *** Waiting for unfinished jobs....
Cloning into 'src/rosidl_typesupport_microxrcedds'...
Cloning into 'src/rosidl'...
Cloning into 'src/rmw'...
Cloning into 'src/rcl_interfaces'...
Cloning into 'src/rosidl_defaults'...
Cloning into 'src/unique_identifier_msgs'...
Cloning into 'src/common_interfaces'...
Cloning into 'src/example_interfaces'...
Cloning into 'src/test_interface_files'...
Cloning into 'src/rmw_implementation'...
Cloning into 'src/rcl_logging'...
Cloning into 'src/ros2_tracing'...
warning: redirecting to https://gitlab.com/ros-tracing/ros2_tracing.git/
Cloning into 'src/micro_ros_utilities'...
touch: src/rcl_logging/rcl_logging_log4cxx/COLCON_IGNORE: No such file or directory
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns micro_ros_espidf_component mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/an/esp/esp-idf/components/app_trace /Users/an/esp/esp-idf/components/app_update /Users/an/esp/esp-idf/components/asio /Users/an/esp/esp-idf/components/bootloader /Users/an/esp/esp-idf/components/bootloader_support /Users/an/esp/esp-idf/components/bt /Users/an/esp/esp-idf/components/cbor /Users/an/esp/esp-idf/components/cmock /Users/an/esp/esp-idf/components/coap /Users/an/esp/esp-idf/components/console /Users/an/esp/esp-idf/components/cxx /Users/an/esp/esp-idf/components/driver /Users/an/esp/esp-idf/components/efuse /Users/an/esp/esp-idf/components/esp-tls /Users/an/esp/esp-idf/components/esp32 /Users/an/esp/esp-idf/components/esp_adc_cal /Users/an/esp/esp-idf/components/esp_common /Users/an/esp/esp-idf/components/esp_eth /Users/an/esp/esp-idf/components/esp_event /Users/an/esp/esp-idf/components/esp_gdbstub /Users/an/esp/esp-idf/components/esp_hid /Users/an/esp/esp-idf/components/esp_http_client /Users/an/esp/esp-idf/components/esp_http_server /Users/an/esp/esp-idf/components/esp_https_ota /Users/an/esp/esp-idf/components/esp_https_server /Users/an/esp/esp-idf/components/esp_hw_support /Users/an/esp/esp-idf/components/esp_ipc /Users/an/esp/esp-i [... cut ...] p-idf/components/tinyusb /Users/an/esp/esp-idf/components/ulp /Users/an/esp/esp-idf/components/unity /Users/an/esp/esp-idf/components/usb /Users/an/esp/esp-idf/components/vfs /Users/an/esp/esp-idf/components/wear_levelling /Users/an/esp/esp-idf/components/wifi_provisioning /Users/an/esp/esp-idf/components/wpa_supplicant /Users/an/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/micro_ros_espidf_component/examples/int32_publisher/build

anslex, 2022-07-25:
This chunk was removed from main log:

```
Finished <<< ament_cmake_ros [1.02s]
--- stderr: ament_index_cpp
Undefined symbols for architecture arm64:
  "std::logic_error::what() const", referenced from:
      vtable for ament_index_cpp::PackageNotFoundError in get_package_prefix.cpp.o
  "std::__1::__basic_string_common<true>::__throw_length_error() const", referenced from:
      std::__1::_MetaBase<__is_cpp17_forward_iterator<char*>::value>::_EnableIfImpl<void> std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::__init<char*>(char*, char*) in get_resource.cpp.o
  "std::__1::locale::has_facet(std::__1::locale::id&) const", referenced from:
      bool std::__1::has_facet<std::__1::codecvt<char, char, __mbstate_t> >(std::__1::locale const&) in get_resource.cpp.o
      bool std::__1::has_facet<std::__1::codecvt<char, char, __mbstate_t> >(std::__1::locale const&) in has_resource.cpp.o
  "std::__1::locale::use_facet(std::__1::locale::id&) const", referenced from:
      std::__1::codecvt<char, char, __mbstate_t> const& std::__1::use_facet<std::__1::codecvt<char, char, __mbstate_t> >(std::__1::locale const&) in get_resource.cpp.o
      std::__1::codecvt<char, char, __mbstate_t> const& std::__1::use_facet<std::__1::codecvt<char, char, __mbstate_t> >(std::__1::locale const&) in has_resource.cpp.o
  "std::logic_error::logic_error(char const*)", referenced from:
      std::length_error::length_error(char const*) in get_resource.cpp.o
      std::length_error::length_error(char const*) in get_resources.cpp.o
      std::length_error::length_error(char const*) in get_search_paths.cpp.o
  "std::logic_error::logic_error(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      std::out_of_range::out_of_range(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in get_package_prefix.cpp.o
  "std::length_error::~length_error()", referenced from:
      std::__1::__throw_length_error(char const*) in get_resource.cpp.o
      std::__1::__throw_length_error(char const*) in get_resources.cpp.o
      std::__1::__throw_length_error(char const*) in get_search_paths.cpp.o
  "std::out_of_range::~out_of_range()", referenced from:
      ament_index_cpp::PackageNotFoundError::PackageNotFoundError(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in get_package_prefix.cpp.o
      ament_index_cpp::PackageNotFoundError::~PackageNotFoundError() in get_package_prefix.cpp.o
  "std::runtime_error::runtime_error(char const*)", referenced from:
      ament_index_cpp::get_resource(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >*) in get_resource.cpp.o
      ament_index_cpp::get_resources(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in get_resources.cpp.o
      ament_index_cpp::get_search_paths() in get_search_paths.cpp.o
      ament_index_cpp::has_resource(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >*) in has_resource.cpp.o
  "std::runtime_error::~runtime_error()", referenced from:
      ament_index_cpp::get_resource(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >*) in get_resource.cpp.o
      ament_index_cpp::get_resources(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in get_resources.cpp.o
      ament_index_cpp::get_search_paths() in get_search_paths.cpp.o
      ament_index_cpp::has_resource(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >*) in has_resource.cpp.o
  "std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::__init(char const*, unsigned long)", referenced from:
      std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string<std::nullptr_t>(char const*) in get_package_prefix.cpp.o
      std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string<std::nullptr_t>(char const*) in get_packages_with_prefixes.cpp.o
      std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string<std::nullptr_t>(char const*) in get_resources.cpp.o
      std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string<std::nullptr_t>(char const*) in get_search_paths.cpp.o
[... cut ...]
      std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string(std::__1::allocator<char> const&) in get_package_prefix.cpp.o
      std::__1::allocator<char>::deallocate(char*, unsigned long) in get_package_prefix.cpp.o
      ...
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [libament_index_cpp.dylib] Error 1
make[2]: *** [CMakeFiles/ament_index_cpp.dir/all] Error 2
make[1]: *** [all] Error 2
---
Failed   <<< ament_index_cpp [1.68s, exited with code 2]

```

pablogs9, 2022-07-26:
Can you make a reproducible environment just by downloading this package in a folder and running `colcon build`. In any case, I guess that this is an `ament_index_cpp` issue, could you move the issue to this repo?

anslex, 2022-07-26:
Thank you, the issue persist so it is not relevant to micro_ros_espidf_component
Closing it

Have a nice day

closed 2022-07-26 by anslex

anslex, 2022-07-26:
I have found that it compiles with these flags (g++ CXX compiler and arm64):
colcon build --cmake-args -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=g++ -DCMAKE_SYSTEM_PROCESSOR=arm64