micro-ROS/micro_ros_espidf_component #154: vscode+espidf  Could not complete compilation Could not find rcl related files

arilink, 2022-08-01:
Hardware : ESP32  ESP32 S3
RTOS:FreeRTOS
ROS: ROS2 humble
Development environment : Ubuntu 22.04LTS +VSCode +ESP IDF 

1. I installed ROS2 humble on Ubuntu and tested the example ROS2 is normal
2. I pulled the code of the micro-ros setup through git and compiled and generated the agent, indicating that the colcon component is normal
3. I used vscode+espidf to create the function of led, which can be compiled normally and completely, and esp32 works normally, indicating that vscode+espidf is normal
4. I created a new project, then added the micro_ros_espidf component, and then set target can generate libmicroros.a normally, indicating that micro_ros_espidf compiles normally
5. When I changed the function in the main function to micro ros transceiver and then compiled the whole project, the problem appeared../main/blink_example_main.c:11:10: fatal error: rcl/rcl.h: No such file or directory I don't know what's causing this problem, but I think I'm one step away from success
6. I re-do the above operation on another computer. The problem is a bit different. Micro_ros_espidf does not generate libmicroros.a normally, but prompts clocon build failed (rcutils+7)
How can I solve the above problem?

arilink, 2022-08-01:
> Executing task: ninja  <

[0/2] Re-checking globbed directories...
[6/1042] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[169/1042] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/arilink/esp32_develop/microros-test04/sdkconfig
-- Adding linker script /home/arilink/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp32 esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /home/arilink/esp/esp-idf/components/bootloader /home/arilink/esp/esp-idf/components/bootloader_support /home/arilink/esp/esp-idf/components/efuse /home/arilink/esp/esp-idf/components/esp32 /home/arilink/esp/esp-idf/components/esp_common /home/arilink/esp/esp-idf/components/esp_hw_support /home/arilink/esp/esp-idf/components/esp_rom /home/arilink/esp/esp-idf/components/esp_system /home/arilink/esp/esp-idf/components/esptool_py /home/arilink/esp/esp-idf/components/freertos /home/arilink/esp/esp-idf/components/hal /home/arilink/esp/esp-idf/components/log /home/arilink/esp/esp-idf/components/bootloader/subproject/main /home/arilink/esp/esp-idf/components/bootloader/subproject/components/micro-ecc /home/arilink/esp/esp-idf/components/newlib /home/arilink/esp/esp-idf/components/partition_table /home/arilink/esp/esp-idf/components/soc /home/arilink/esp/esp-idf/components/spi_flash /home/arilink/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/arilink/esp32_develop/microros-test04/build/bootloader
[267/1042] Performing build step for 'bootloader'
[1/97] Generating project_elf_src_esp32.c
[2/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/lldesc.c.obj
[3/97] Building C object CMakeFiles/bootloader.elf.dir/project_elf_src_esp32.c.obj
[4/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/dac_periph.c.obj
[5/97] Building C object esp-idf/hal/CMakeFiles/__idf_hal.dir/mpu_hal.c.obj
[6/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/soc_include_legacy_warn.c.obj
[7/97] Building C object esp-idf/hal/CMakeFiles/__idf_hal.dir/cpu_hal.c.obj
[8/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/gpio_periph.c.obj
[9/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/adc_periph.c.obj
[10/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/lcd_periph.c.obj
[11/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/interrupts.c.obj
[12/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/i2s_periph.c.obj
[13/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/i2c_periph.c.obj
[14/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/mcpwm_periph.c.obj
[15/97] Building C object esp-idf/hal/CMakeFiles/__idf_hal.dir/wdt_hal_iram.c.obj
[16/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rmt_periph.c.obj
[17/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/rtc_io_periph.c.obj
[18/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/ledc_periph.c.obj
[19/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/sdio_slave_periph.c.obj
[20/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/pcnt_periph.c.obj
[21/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/timer_periph.c.obj
[22/97] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/esp32/sigmadelta_periph.c.obj
[... cut ...] main.dir/blink_example_main.c.obj -c ../main/blink_example_main.c
../main/blink_example_main.c:11:10: fatal error: rcl/rcl.h: No such file or directory
 #include <rcl/rcl.h>
          ^~~~~~~~~~~
compilation terminated.
[1031/1042] Building C object esp-idf/micro_ros_espidf_component/CMakeFiles/__idf_micro_ros_espidf_component.dir/network_interfaces/uros_ethernet_netif.c.obj
ninja: build stopped: subcommand failed.
The terminal process "/usr/bin/bash '-c', 'ninja '" terminated with exit code: 1.

pablogs9, 2022-08-01:
Hello @arilink:
- Could you provide the output of this: `but prompts colcon build failed (rcutils+7)`?
- Could you provide steps for replicating this issue using ESP-IDF?

arilink, 2022-08-01:
[image]
Here is the output from one of my computers (couldn't compile libmicroros.a)

pablogs9, 2022-08-01:
Please provide instructions using just ESP-IDF, not the VSCode extension.

pablogs9, 2022-08-01:
We need to eliminate the possibility of error in this extension. So I need a set of commands to replicate the issue based on the instructions we provide: https://github.com/micro-ROS/micro_ros_espidf_component#example

arilink, 2022-08-01:
Ok, I will try it according to your example
Under normal circumstances, the vscode esp-idf extension can also be supported.

arilink, 2022-08-01:
@pablogs9 
After testing git clone micro_ros_espidf_component, you can use idf.py to compile the firmware according to the example, which shows that the development environment is fine, but it is inconvenient to merge with my current project. I guess it is related to cmake, but there are many related problems and I don't know how to modify it

arilink, 2022-08-02:
I have new progress
1. I found that the header file path contained in the main.c file was not added to the idf_component_register function in the CMakeLists.txt file in the root directory of the micro_ros_espidf_component component when using vscode+extension. For example, I manually added micro_ros_src/rcl/rcl/include and re-installed Compile will not prompt fatal error: rcl/rcl.h: No such file or directory and the program will continue to compile down
2. Continue to add the path on the basis of 1. At this time, a new problem occurs, fatal error: std_msgs/msg/int32.h: No such file or directory After searching, the corresponding file is not found in the micro_ros_scr directory. After searching with idf After compiling micro_ros_espidf_component with .py, it is found that many folders are missing in the include directory. idf.py After compiling micro_ros_espidf_component, there is an int32.h file in the include folder. Does this mean that micro_ros_espidf_component is not fully compiled? Then why can libmicroros.a be generated? Can I manually compile the entire micro_ros_espidf_component component under vscode+extension?
3. Based on 2, because it is in China and the network environment is limited, git clone is very slow when setting config in the vscode+extension environment, so I manually cloned the relevant files and placed them in the micro_ros_src directory. Is this possible? The reason for not fully compiling?

arilink, 2022-08-02:
@pablogs9  Here is the screenshot
[image]

pablogs9, 2022-08-02:
The micro-ROS build step is failing, this include folder does not have all the required packages. Can you paste here the micro-ROS build process output?

arilink, 2022-08-03:
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/arilink/esp32_develop/microros-test04/sdkconfig
Loading defaults file /home/arilink/esp32_develop/microros-test04/sdkconfig.defaults...
-- Found Python3: /home/arilink/.espressif/python_env/idf4.4_py3.10_env/bin/python3.10 (found version "3.10.4") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "microros-test04" version: 1
-- Adding linker script /home/arilink/esp32_develop/microros-test04/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Using Micro XRCE-DDS middleware
mkdir -p /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component/micro_ros_src/libmicroros; cd /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component/micro_ros_src/libmicroros; \
for file in $(find /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component/micro_ros_src/install/lib/ -name '*.a'); do \
	folder=$(echo $file | sed -E "s/(.+)\/(.+).a/\2/"); \
	mkdir -p $folder; cd $folder; /home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar x $file; \
	for f in *; do \
		mv $f ../$folder-$f; \
	done; \
	cd ..; rm -rf $folder; \
done ; \
/home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar rc -s libmicroros.a *.obj; cp libmicroros.a /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component; \
cd ..; rm -rf libmicroros; \
cp -R /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component/micro_ros_src/install/include /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component/include;
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap ieee802154 jsmn json led_strip libsodium log lwip main mbedtls mdns micro_ros_espidf_component mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter test tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/arilink/esp/esp-idf/components/app_trace /home/arilink/esp/esp-idf/components/app_update /home/arilink/esp/esp-idf/components/asio /home/arilink/esp/esp-idf/components/bootloader /home/arilink/esp/esp-idf/components/bootloader_support /home/arilink/esp/esp-idf/components/bt /home/arilink/esp/esp-idf/components/cbor /home/arilink/esp/esp-idf/components/cmock /home/arilink/esp/esp-idf/components/coap /home/arilink/esp/esp-idf/components/console /home/arilink/esp/esp-idf/components/cxx /home/arilink/esp/esp-idf/components/driver /home/arilink/esp/esp-idf/components/efuse /home/arilink/esp/esp-idf/components/esp-tls /home/arilink/esp/esp-idf/components/esp32 /home/arilink/esp/esp-idf/components/esp_adc_cal /home/arilink/esp/esp-idf/components/esp_common /home/arilink/esp/esp-idf/components/esp_eth /home/arilink/esp/esp-idf/components/esp_event /home/arilink/esp/esp-idf/component [... cut ...]
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/arilink/esp32_develop/microros-test04/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/arilink/esp32_develop/microros-test04/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/arilink/esp32_develop/microros-test04"...

arilink, 2022-08-03:
I think we are very close to success. Currently libmicroros.a can be generated, but many header files are not generated as expected, so an error will be reported in main.c, as long as we can find out the reason for the failure to generate

arilink, 2022-08-03:
> Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 .. <

-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/arilink/esp32_develop/microros-test04/sdkconfig
Loading defaults file /home/arilink/esp32_develop/microros-test04/sdkconfig.defaults...
-- Found Python3: /home/arilink/.espressif/python_env/idf4.4_py3.10_env/bin/python3.10 (found version "3.10.4") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "microros-test04" version: 1
-- Adding linker script /home/arilink/esp32_develop/microros-test04/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/arilink/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Using Micro XRCE-DDS middleware
mkdir -p /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component/micro_ros_src/libmicroros; cd /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component/micro_ros_src/libmicroros; \
for file in $(find /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component/micro_ros_src/install/lib/ -name '*.a'); do \
        folder=$(echo $file | sed -E "s/(.+)\/(.+).a/\2/"); \
        mkdir -p $folder; cd $folder; /home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar x $file; \
        for f in *; do \
                mv $f ../$folder-$f; \
        done; \
        cd ..; rm -rf $folder; \
done ; \
/home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar rc -s libmicroros.a *.obj; cp libmicroros.a /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component; \
cd ..; rm -rf libmicroros; \
cp -R /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component/micro_ros_src/install/include /home/arilink/esp32_develop/microros-test04/components/micro_ros_espidf_component/include;
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap ieee802154 jsmn json led_strip libsodium log lwip main mbedtls mdns micro_ros_espidf_component mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter test tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/arilink/esp/esp-idf/components/app_trace /home/arilink/esp/esp-idf/components/app_update /home/arilink/esp/esp-idf/components/asio /home/arilink/esp/esp-idf/components/bootloader /home/arilink/esp/esp-idf/components/bootloader_support /home/arilink/esp/esp-idf/components/bt /home/arilink/esp/esp-idf/components/cbor /home/arilink/esp/esp-idf/components/cmock /home/arilink/esp/esp-idf/components/coap /home/arilink/esp/esp-idf/components/console /home/arilink/esp/esp-idf/components/cxx /home/arilink/esp/esp-idf/components/driver /home/arilink/esp/esp-idf/components/efuse /home/arilink/esp/esp-idf/components/esp-tls /home/arilink/esp/esp-idf/components/esp32 /home/arilink/esp/esp-idf/components/esp_adc_cal /home/arilink/esp/esp-idf/components/esp_commo [... cut ...] reference to `rclc_executor_add_timer'
/home/arilink/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/arilink/esp32_develop/microros-test04/build/../main/blink_example_main.c:74: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
The terminal process "/usr/bin/bash '-c', 'ninja '" terminated with exit code: 1.

arilink, 2022-08-03:
I manually put the include file in the micro_ros_espidf_component compiled by idf.py into the corresponding directory of the vscode esp idf project, and then compile it, there will still be problems I am very confused

arilink, 2022-08-04:
Do you have any suggestions for the current situation? Or is there a way I can manually compile this component and see the specific log output to find the problem?

pablogs9, 2022-08-04:
Have you installed the dependencies in the IDF environment? https://github.com/micro-ROS/micro_ros_espidf_component#dependencies

In any case, my recommendation is to use the recommended approach (ESP-IDF CLI) as it is explained in the README.md.

arilink, 2022-08-04:
Yes I have installed those dependencies, I know it is more professional to use esp cli, but I need to add micro-ros components to the existing project, it is inconvenient to use micro-ros-espidf as the whole project functions are merged. Or I use esp cli to create a project and then add the micro-ros-espidf component to compile it?

meetgandhi21, 2022-10-06:
I found the similar issue when I tried to build one of the examples given here.

```
[ 99%] Built target __idf_wifi_provisioning
[ 99%] Built target __idf_esp_local_ctrl
[ 99%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
/home/einfochips/esp-idf/components/micro_ros_espidf_component/examples/int32_publisher/main/main.c:11:10: fatal error: rcl/rcl.h: No such file or directory
 #include <rcl/rcl.h>
          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:76: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:4873: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make failed with exit code 2

```

Tried to build the same using the docker container provided with this repo and was able to build the example application successfully.

```
esptool.py v3.3.2-dev
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /home/espidf/micro_ros_espidf_component/examples/int32_publisher/build/bootloader/bootloader.bin
[997/997] Generating binary image from built executable
esptool.py v3.3.2-dev
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated /home/espidf/micro_ros_espidf_component/examples/int32_publisher/build/int32_publisher.bin

Project build complete. To flash, run this command:
/opt/esp/python_env/idf4.3_py3.6_env/bin/python ../../../../../opt/esp/idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/int32_publisher.bin
or run 'idf.py -p (PORT) flash'
```

When I tried debugging the same I found that while you're building this component for the first time, the ROS distro's setup file shouldn't be sourced into your current terminal. See [here](https://github.com/micro-ROS/micro_ros_espidf_component/issues/154)

_If you encounter issues during the build process, ensure that you are running in a clean shell environment without the ROS 2 setup script sourced._

I'm not sure what is the reason behind the same?

arilink, 2022-10-11:
Yes. This is a serious problem. When I try to add micro ros as a component to the project and compile it, I get an error

***@***.***
 
From: meetgandhi21
Date: 2022-10-06 14:44
To: micro-ROS/micro_ros_espidf_component
CC: arilink; Mention
Subject: Re: [micro-ROS/micro_ros_espidf_component] vscode+espidf Could not complete compilation Could not find rcl related files (Issue #154)
I found the similar issue when I tried to build one of the examples given here.
[ 99%] Built target __idf_wifi_provisioning
[ 99%] Built target __idf_esp_local_ctrl
[ 99%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
/home/einfochips/esp-idf/components/micro_ros_espidf_component/examples/int32_publisher/main/main.c:11:10: fatal error: rcl/rcl.h: No such file or directory
 #include <rcl/rcl.h>
          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:76: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:4873: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make failed with exit code 2

Tried to build the same using the docker container provided with this repo and was able to build the example application successfully.
esptool.py v3.3.2-dev
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /home/espidf/micro_ros_espidf_component/examples/int32_publisher/build/bootloader/bootloader.bin
[997/997] Generating binary image from built executable
esptool.py v3.3.2-dev
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated /home/espidf/micro_ros_espidf_component/examples/int32_publisher/build/int32_publisher.bin
Project build complete. To flash, run this command:
/opt/esp/python_env/idf4.3_py3.6_env/bin/python ../../../../../opt/esp/idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/int32_publisher.bin
or run 'idf.py -p (PORT) flash'

When I tried debugging the same I found that while you're building this component for the first time, the ROS distro's setup file shouldn't be sourced into your current terminal. See here
If you encounter issues during the build process, ensure that you are running in a clean shell environment without the ROS 2 setup script sourced.
I'm not sure what is the reason behind the same?

ManvithRaj, 2022-10-31:
Hello @pablogs9 and others  to add to this i am also facing the similar issue . 

I have cloned micro_ros_idf_component into esp-idf components folder and tried building example int32_publisher with RTPS Enabled.

After copying i given command esp-idf set-target esp32 which is trying to build micro_ros_idf_component  and generated two extra folders in it ie. 1. micro_ros_src 2. micro_ros_dev somehow the build process not generating all the required files. I could see below messages with error.

i could see only 3 folders in install directory of micro ros component 

[image]

[image]

After above step if i proceed with esp-idf set-target esp32 and esp-idf menuconfig still i could able to do these configurations.

But during idf.py build i am getting issues in following order

1. rcl/rcl.h include issue  ---->Found this file in esp_idf_microros_component install/include folder I have added this path manually in cmakelist.txt of esp_idf_microros_component
2. rmw/init.h include issue  ---> found this file also. Added this also Similarly 

like this many .h file issues came I have added those path manually in cmakelist.txt

Finally . msgs/int32.h  issue   -----> this file was not found as the build generation for component not happened as I said early.

Could you please help us on what might be missing since i am new to this i might be not aware much about micro-ROS and why we are unable to build like video you shared at link https://asciinema.org/a/G39IxQZkzipXSxrnbqepXh6iF.

wojciechmadry, 2023-10-04:
Hey, I'm fighting the same issue about missing "rcl.h" file. I don't know if it helps, but here is a simple Dockerfile which shows the problem:
```
FROM ros:iron-ros-core

ENV IDF_PATH /ws/esp-idf
WORKDIR /ws

RUN apt update \
    && apt install -y \
    git \
    python3-pip \
    libusb-1.0-0-dev \
    python3.10-venv \
    && pip3 install catkin_pkg lark-parser empy colcon-common-extensions

RUN mkdir -p /ws/src \
    && git clone -b v5.0 https://github.com/espressif/esp-idf ${IDF_PATH} \
    && cd ${IDF_PATH} \
    && git submodule init && git submodule update \
    && sh ${IDF_PATH}/install.sh

RUN mkdir -p /ws/int32_publisher/components \
    && git clone --recursive -b iron \
    https://github.com/micro-ROS/micro_ros_espidf_component.git \
    /ws/int32_publisher/components/micro_ros_espidf_component \
    && cp -r /ws/int32_publisher/components/micro_ros_espidf_component/examples/int32_publisher/* /ws/int32_publisher

RUN cd /ws/int32_publisher \
    && . ${IDF_PATH}/export.sh \
    && idf.py set-target esp32 \
    && idf.py build

```
[image]

pablogs9, 2023-10-04:
In ROS docker the ROS environment is sourced by default. Micro Ros needs it to be disabled. Could you try that?

wojciechmadry, 2023-10-04:
Hey, the error message is the same.

Dockerfile with ubuntu22:04 as a base:

```
FROM ubuntu:22.04

ENV IDF_PATH /ws/esp-idf
WORKDIR /ws

RUN apt update \
    && apt install -y \
    git \
    python3-pip \
    libusb-1.0-0-dev \
    python3.10-venv \
    software-properties-common \
    curl \
    && pip3 install catkin_pkg lark-parser empy colcon-common-extensions

RUN add-apt-repository universe -y

RUN apt update \
    && curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg \
    && echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

ENV DEBIAN_FRONTEND noninteractive
ENV TZ=Europe/Warsaw
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

RUN apt update \
    && apt install -y \
    ros-iron-ros-base

RUN mkdir -p /ws/src \
    && git clone -b v5.0 https://github.com/espressif/esp-idf ${IDF_PATH} \
    && cd ${IDF_PATH} \
    && git submodule init && git submodule update \
    && sh ${IDF_PATH}/install.sh

RUN mkdir -p /ws/int32_publisher/components \
    && git clone --recursive -b iron \
    https://github.com/micro-ROS/micro_ros_espidf_component.git \
    /ws/int32_publisher/components/micro_ros_espidf_component \
    && cp -r /ws/int32_publisher/components/micro_ros_espidf_component/examples/int32_publisher/* /ws/int32_publisher

RUN cd /ws/int32_publisher \
    && . ${IDF_PATH}/export.sh \
    && idf.py set-target esp32 \
    && idf.py build
```

pablogs9, 2023-10-05:
This one worked for me:

```
FROM espressif/idf:release-v5.0

RUN git clone --recursive -b iron https://github.com/micro-ROS/micro_ros_espidf_component.git \
    && apt update && apt install -y python3-pip \
    && . ${IDF_PATH}/export.sh \
    && pip3 install catkin_pkg lark-parser empy colcon-common-extensions \
    && pip3 install importlib-resources \
    && /usr/bin/pip3 install catkin_pkg lark-parser empy colcon-common-extensions importlib-resources \
    && cd micro_ros_espidf_component/examples/int32_publisher \
    && idf.py set-target esp32 \
    && idf.py build
```

wojciechmadry, 2023-10-05:
Hi, It worked - Thanks. espressif based on ubuntu20 so I switched the system in my Dockerfile to ubuntu20, but the error was the same. So I started digging into [idf](https://github.com/espressif/esp-idf/blob/master/tools/docker/Dockerfile) image.
And steps below fixed my problem:

1. Install dependencies (I'm not sure which package did that, but after installing it ROS package starts building):
```
apt-get install -y \
    apt-utils \
    bison \
    ca-certificates \
    ccache \
    check \
    curl \
    flex \
    git \
    git-lfs \
    gperf \
    lcov \
    libbsd-dev \
    libffi-dev \
    libncurses-dev \
    libusb-1.0-0-dev \
    make \
    ninja-build \
    python3 \
    python3-venv \
    ruby \
    unzip \
    wget \
    xz-utils \
    zip \
    bzip2 \
    libglib2.0-0 \
    libpixman-1-0 \
    libslirp0 \
    ament-cmake-core
```
2. Remove old build
3. First source and after install packages (This [comment](https://github.com/micro-ROS/micro_ros_espidf_component/issues/183#issuecomment-1577459440) helps me):
```
source ./export.sh
pip3 install catkin_pkg lark-parser empy colcon-common-extensions
```
4. Rebuild project
```
idf.py build
```
Or if you still have problem try to rebuild ROS packages:
```
idf.py clean-microros
```