micro-ROS/micro_ros_espidf_component #155: ESP32 crashes when listening_message_reliably

SamyFrancelet, 2022-08-06:
## ESP32 crashes when listening_message_reliably

- Hardware description: ESP32 serially linked to a nRF5340DK (using UART1 on pins 19/18 as TX/RX)
- RTOS: FreeRTOS
- Installation type: micro_ros_espidf_component, ESP IDF v4.4
- Version or commit hash: humble

#### Steps to reproduce the issue
My project consist on controlling a BT Mesh provisioner via ROS, and transmitting sensors data from the Mesh network to ROS.
The ESP32 serves as a message bridge between the provisioner and ROS.
For that I've got 4 threads running : 
- the uROS thread (priority 5, stack 16000 bytes) with 2 subs and 2 pubs
- an "uart_phy" thread (priority 6, stack 4096 bytes) which job is to read data from UART1 and sends it to uart_mac_rx with a message queue
- 2 "uart_mac_rx" and "uart_mac_tx" threads (priority 6, stack 4096 bytes). uart_mac_rx gets data from uart_phy and publishes message to ROS. uart_mac_tx receives messages from the subscribers callbacks and transmits data to UART1

The nRF5340 scans for unprovisioned ("unconnected") mesh nodes, and sends the UUIDs to the ESP32 via serial.
So the ESP32 receives a 16 bytes UUID from UART, the uart_phy sends this UUID to uart_mac_rx through a message queue.
uart_mac_rx wakes up because something is in the queue, sees that it is an UUID, and publish it with my "unprov_pub".
The ROS message is the following : 
ble2ros_interfaces/msg/Beacon.msg
```
# Represent an unprovisioned beacon from BT Mesh
uint8[16] uuid
```

#### Expected behavior
UUIDs are sent to ROS without error

#### Actual behavior
Sometimes, mostly when the ESP32 receives a lot of UUID's to transmit to ROS, i got the following error : 
`assert failed: pbuf_free IDF/components/lwip/lwip/src/core/pbuf.c:757 (pbuf_free: p->ref > 0)`
OR
`assert failed: tlsf_free heap_tlsf.c:872 (!block_is_free(block) && "block already marked as free")`

but mostly the pbuf_free assert failed.
The most frequent stack backtrace is the following : 
```
assert failed: pbuf_free IDF/components/lwip/lwip/src/core/pbuf.c:757 (pbuf_free: p->ref > 0)

Backtrace:0x40081ade:0x3ffd0a700x400891bd:0x3ffd0a90 0x4008fd5d:0x3ffd0ab0 0x400f3de7:0x3ffd0bd0 0x401057c1:0x3ffd0bf0 0x400f03ef:0x3ffd0c10 0x400f0ff2:0x3ffd0c50 0x400f1072:0x3ffd0cb0 0x400de19d:0x3ffd0cd0 0x400ddfad:0x3ffd0d00 0x400dd2fc:0x3ffd0d30 0x400dd488:0x3ffd0dd0 0x400e2836:0x3ffd0df0 0x400e0e4b:0x3ffd0e40 0x400da51d:0x3ffd0eb0 0x400da563:0x3ffd0ef0 0x400d6e6c:0x3ffd0f10 0x400d6e7e:0x3ffd0f30 0x4008c37d:0x3ffd0f50 
0x40081ade: panic_abort at /home/samy/esp/esp-idf/components/esp_system/panic.c:402
0x400891bd: esp_system_abort at /home/samy/esp/esp-idf/components/esp_system/esp_system.c:128
0x4008fd5d: __assert_func at /home/samy/esp/esp-idf/components/newlib/assert.c:85
0x400f3de7: pbuf_free at /home/samy/esp/esp-idf/components/lwip/lwip/src/core/pbuf.c:757 (discriminator 1)
0x401057c1: netbuf_delete at /home/samy/esp/esp-idf/components/lwip/lwip/src/api/netbuf.c:85
0x400f03ef: lwip_recvfrom_udp_raw at /home/samy/esp/esp-idf/components/lwip/lwip/src/api/sockets.c:1250
0x400f0ff2: lwip_recvfrom at /home/samy/esp/esp-idf/components/lwip/lwip/src/api/sockets.c:1292 (discriminator 4)
0x400f1072: lwip_recv at /home/samy/esp/esp-idf/components/lwip/lwip/src/api/sockets.c:1337
0x400de19d: uxr_read_udp_data_platform at ??:?
0x400ddfad: recv_udp_msg at udp_transport.c:?
0x400dd2fc: listen_message_reliably at session.c:?
0x400dd488: uxr_run_session_until_data at ??:?
0x400e2836: rmw_wait at ??:?
0x400e0e4b: rcl_wait at ??:?
0x400da51d: rclc_executor_spin_some$part$5 at executor.c:?
0x400da563: rclc_executor_spin_some at ??:?
0x400d6e6c: spin_uros at /home/samy/work/ble2ros/firmware/uros_node/main/uros/uros.c:119 (discriminator 1)
0x400d6e7e: uros_task at /home/samy/work/ble2ros/firmware/uros_node/main/uros/uros.c:126
0x4008c37d: vPortTaskWrapper at /home/samy/esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

#### Additional information
I also had other configuration with at most 4 pubs and 4 subs, and this problem can occur with any publisher.
I went back to 2 pubs and 2 subs to get back to the basic uROS config (also resetted the max pub/sub in colcon.meta)

I looked if the heap became corrupted at some point, but it was always clean.

When disabling asserts, it stills crash with Guru meditation error: Core 0 panic'ed

I tried lowering priorities, (12 for UART, 11 for uROS), switching priorities (4 for UART, 5 for uROS), still got the same problem.

Looked over the internet for solutions, looked in this repo's issues, found nothing...

Looked if I had memory leaks, but found none

pablogs9, 2022-08-08:
What are the differences between `the uROS thread` and ` "uart_mac_rx" and "uart_mac_tx" threads`? Are you calling the micro-ROS API from two different threads? Have you enabled [the multithread operation of the middleware](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/6f4edd25c0984fd30bf3bf84e2d1d33af03f24a1/CMakeLists.txt#L67)?

In any case, a corruption like this sounds to me like a stack overflow. Have you tried to increase stack sizes of each task?

SamyFrancelet, 2022-08-08:
Here is a quick diagram showing how to program is composed : 

[image]

Every subs run with the uros_thread executor, their callback functions sends a message to the txQueue.
uart_mac_rx receives messages from the uart_phy thread and calls the rcl_publish, so yeah it uses the micro-ROS API.

I already tried increasing stacks, but maybe not enough.
I'll try both propositions

pablogs9, 2022-08-08:
Ok, please provide feedback. And enable the multithread flag because if not, all the stack layers are not thread safe!

SamyFrancelet, 2022-08-08:
When I turn ON the UCLIENT_PROFILE_MULTITHREAD in the CMakeLists.txt of Micro-XRCE-DDS-Client, it doesn't change the build, and when I do a clean build, it reverts back to OFF. Am I missing something ?

pablogs9, 2022-08-08:
You need to use the [`colcon.meta`](https://github.com/micro-ROS/micro_ros_espidf_component/blob/humble/colcon.meta) file in order to confgure the micro-ROS stack.

SamyFrancelet, 2022-08-08:
I have added the line ` "-DUCLIENT_PROFILE_MULTITHREAD=ON"` in colcon.meta, but it stays off in the CMakeLists.txt

pablogs9, 2022-08-08:
Just check that is enabled in the generated `config.h` inside Micro XRCE-DDS Client include folder

SamyFrancelet, 2022-08-08:
Just a moment, my build keeps crashing (doesn't find <rcl/rcl.h> and micro-ros doesn't want to clean)

SamyFrancelet, 2022-08-08:
When I set the line in colcon.meta, the build keeps crashing

SamyFrancelet, 2022-08-08:
Already done, clean, fullclean, clean-microros, removed the microROS component and resetted it

pablogs9, 2022-08-08:
Provide more info about this crash

SamyFrancelet, 2022-08-08:
build failed with doesn't find <rcl/rcl.h>

SamyFrancelet, 2022-08-08:
```
Finished <<< ament_index_cpp [5.25s]

Summary: 60 packages finished [1min 15s]
  58 packages had stderr output: ament_clang_format ament_clang_tidy ament_cmake ament_cmake_auto ament_cmake_clang_format ament_cmake_clang_tidy ament_cmake_copyright ament_cmake_core ament_cmake_cppcheck ament_cmake_cpplint ament_cmake_export_definitions ament_cmake_export_dependencies ament_cmake_export_include_directories ament_cmake_export_interfaces ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_export_targets ament_cmake_flake8 ament_cmake_gmock ament_cmake_google_benchmark ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_lint_cmake ament_cmake_mypy ament_cmake_nose ament_cmake_pclint ament_cmake_pep257 ament_cmake_pycodestyle ament_cmake_pyflakes ament_cmake_pytest ament_cmake_python ament_cmake_ros ament_cmake_target_dependencies ament_cmake_test ament_cmake_uncrustify ament_cmake_version ament_cmake_xmllint ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_auto ament_lint_cmake ament_lint_common ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint domain_coordinator gmock_vendor gtest_vendor
cd /home/samy/work/ble2ros/firmware/uros_node/components/uros_component/micro_ros_src; \
unset AMENT_PREFIX_PATH; \
PATH="/home/samy/esp/esp-idf/components/esptool_py/esptool:/home/samy/esp/esp-idf/components/espcoredump:/home/samy/esp/esp-idf/components/partition_table:/home/samy/esp/esp-idf/components/app_update:/home/samy/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/samy/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/samy/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/home/samy/.espressif/python_env/idf4.4_py3.10_env/bin:/home/samy/esp/esp-idf/tools::/home/samy/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"; \
. ../micro_ros_dev/install/local_setup.sh; \
colcon build \
        --merge-install \
        --packages-ignore-regex=.*_cpp \
        --metas /home/samy/work/ble2ros/firmware/uros_node/components/uros_component/colcon.meta  \
        --cmake-args \
        "--no-warn-unused-cli" \
        -DCMAKE_POSITION_INDEPENDENT_CODE:BOOL=OFF \
        -DTHIRDPARTY=ON \
        -DBUILD_SHARED_LIBS=OFF \
        -DBUILD_TESTING=OFF \
        -DCMAKE_BUILD_TYPE=Release \
        -DCMAKE_TOOLCHAIN_FILE=/home/samy/work/ble2ros/firmware/uros_node/components/uros_component/esp32_toolchain.cmake \
        -DCMAKE_VERBOSE_MAKEFILE=OFF; \

Starting >>> rcutils
Starting >>> rosidl_cli
Starting >>> rosidl_typesupport_interface
Starting >>> microcdr
Finished <<< rosidl_typesupport_interface [6.60s]
Starting >>> rmw_implementation_cmake
Finished <<< microcdr [10.3s]
Starting >>> microxrcedds_client
--- stderr: rosidl_cli
/home/samy/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---Finished <<< rosidl_cli [11.5s]

Starting >>> rosidl_adapter
Finished <<< rmw_implementation_cmake [7.06s]
Starting >>> tracetools
--- stderr: microxrcedds_client
In file included from /home/samy/work/ble2ros/firmware/uros_node/components/uros_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/core/session/stream/output_best_effort_stream.h:27,
                 from /home/samy/work/ble2ros/firmware/uros_node/components/uros_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/output_best_effort_stream_internal.h:23,
                 from /home/samy/work/ble2ros/firmware/uros_node/components/uros_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/output_best_effort_stream.c:1:
/home/samy/work/ble2ros/firmware/uros_node/components/uros_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/profile/multithread/multithread.h:33:10: fatal error: FreeRTOS.h: No such file or directory
 #include "FreeRTOS.h"
          ^~~~~~~~~~~~
compilation terminated.
In file included from /home/samy/work/ble2ros/firmware/uros_node/components/uros_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/core/session/stream/input_best_effort_stream.h:27,
                 from /home/samy/work/ble2ros/firmware/uros_node/components/uros_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/input_best_effort_stream_internal.h:23,
                 from /home/samy/work/ble2ros/firmware/uros_node/components/uros_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/input_best_effort_stream.c:1:
/home/samy/work/ble2ros/firmware/uros_node/components/uros_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/profile/multithread/multithread.h:33:10: fatal error: FreeRTOS.h: No such file or directory
 #include "FreeRTOS.h"
          ^~~~~~~~~~~~
compilation terminated.
gmake[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:104 : CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/output_best_effort_stream.c.obj] Erreur 1
gmake[3]: *** Attente des tâches non terminées....
[... cut ...] y/work/ble2ros/firmware/uros_node/main/main.c
In file included from /home/samy/work/ble2ros/firmware/uros_node/main/main.c:13:
/home/samy/work/ble2ros/firmware/uros_node/main/uros/uros.h:23:10: fatal error: rcl/rcl.h: No such file or directory
 #include <rcl/rcl.h>
          ^~~~~~~~~~~
compilation terminated.
[1023/1045] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/manager.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

SamyFrancelet, 2022-08-08:
the stderr in the uROS build is typically : 
```
-- stderr: ament_cmake_clang_format
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_CXX_COMPILER
    CMAKE_C_COMPILER

---
```

pablogs9, 2022-08-08:
It is not finding your FreeRTOS headers, let me check

SamyFrancelet, 2022-08-11:
Found anything ?

SamyFrancelet, 2022-08-25:
Can someone at least guide me? Maybe my config in the colcon.meta wasn't done properly? It would help me a lot as I've already tried on multiple computers, with multiple codebases (including ones without any thread apart from the microROS thread), and still, impossible to turn on this multithreading. Am I missing something?

shic15, 2022-09-22:
Hey, I'm facing the exact same error. I'll try the things mentioned here and will update the results. 
If anyone has already got the resolution please provide it.

shic15, 2022-09-25:
Project builds successfully after adding "-DUUCLIENT_PROFILE_MULTITHREAD=ON" in colcon.meta.
Still, I'm getting a similar error. 

I'm using a single executor and thread for uros so I think changing the multi thread flag won't do anything. This error occurs only with a specific sequence of requests made (service clients on ESP32) which are working as usual when executed otherwise. 

```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000c291  PS      : 0x00060c30  A0      : 0x800deb56  A1      : 0x3ffc9d60  
A2      : 0x00000001  A3      : 0x3ffb5c8d  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x3ffb5c88  A12     : 0x05aed580  A13     : 0x00000001  
A14     : 0x3ffc9fa0  A15     : 0x00000003  SAR     : 0x0000000c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000001  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

Backtrace: 0x4000c28e:0x3ffc9d60 0x400deb53:0x3ffc9d70 0x400da0af:0x3ffc9da0 0x400d837d:0x3ffc9dc0 0x400ddef2:0x3ffc9df0 0x400da445:0x3ffc9e40 0x400dbac0:0x3ffc9e90 0x400dc0f3:0x3ffc9ef0 0x400dc1db:0x3ffc9f20 0x400d7841:0x3ffc9f40 0x4008ca2d:0x3ffca040
0x400deb53: ucdr_deserialize_endian_array_char at ??:?

0x400da0af: ucdr_deserialize_sequence_char at ??:?

0x400d837d: _DcapsTask_Response__cdr_deserialize at dcaps_task__type_support_c.c:?

0x400ddef2: rmw_take_response at ??:?

0x400da445: rcl_take_response at ??:?

0x400dbac0: _rclc_take_new_data at executor.c:?

0x400dc0f3: rclc_executor_spin_some$part$4 at executor.c:?

0x400dc1db: rclc_executor_spin_some at ??:?

0x400d7841: micro_ros_task at /home/shic/Foodlabs_Codes/AiKi/Saute/pasta-house/pasta-house-esp32/module_/build/../components/UROS/src/uros.c:306 (discriminator 1)

0x4008ca2d: vPortTaskWrapper at /home/shic/esp/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 45168d89d1a4c770

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6676
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
```
It'll be really helpful to understand what the error is about.

SamyFrancelet, 2022-11-10:
@pablogs9 still nothing ?

pablogs9, 2022-11-10:
Provide code for replicating this issue

SamyFrancelet, 2022-11-14:
I will try to provide a minimal code replicating this soon, as it currently requires connection with multiple other devices

trilokeshtarala, 2023-03-07:
```
#include <string.h>
#include <stdio.h>
#include <unistd.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_system.h"

#include <uros_network_interfaces.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <std_msgs/msg/int32.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <rmw_microros/rmw_microros.h>
#include "uxr/client/config.h"

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}
#define CONFIG_MICRO_ROS_APP_STACK 10000
#define CONFIG_MICRO_ROS_APP_TASK_PRIO 2

rcl_publisher_t publisher_1;
rcl_publisher_t publisher_2;

void thread_1(void * arg)
{
	std_msgs__msg__Int32 msg;
	msg.data = 0;
	while(1){
		RCSOFTCHECK(rcl_publish(&publisher_1, &msg, NULL));
		msg.data++;
		// usleep(1000000);
    vTaskDelay(1000 / portTICK_PERIOD_MS);
	}
}

void thread_2(void * arg)
{
	std_msgs__msg__Int32 msg;
	msg.data = 0;
	while(1){
		RCSOFTCHECK(rcl_publish(&publisher_2, &msg, NULL));
		msg.data--;
		// usleep(500000);
    vTaskDelay(500 / portTICK_PERIOD_MS);
	}
}

void micro_ros_task(void * arg)
{
	rcl_allocator_t allocator = rcl_get_default_allocator();
	rclc_support_t support;

	rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
	RCCHECK(rcl_init_options_init(&init_options, allocator));
	rmw_init_options_t* rmw_options = rcl_init_options_get_rmw_init_options(&init_options);

	// Static Agent IP and port can be used instead of autodisvery.
	RCCHECK(rmw_uros_options_set_udp_address(CONFIG_MICRO_ROS_AGENT_IP, CONFIG_MICRO_ROS_AGENT_PORT, rmw_options));

	// create init_options
	RCCHECK(rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator));

	// create node
	rcl_node_t node;
	RCCHECK(rclc_node_init_default(&node, "multithread_node", "", &support));

	// create two publishers
	RCCHECK(rclc_publisher_init_default(
		&publisher_1,
		&node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
		"multithread_publisher_1"));

	RCCHECK(rclc_publisher_init_default(
		&publisher_2,
		&node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
		"multithread_publisher_2"));

	xTaskCreate(thread_1,
		"thread_1",
		CONFIG_MICRO_ROS_APP_STACK,
		NULL,
		CONFIG_MICRO_ROS_APP_TASK_PRIO,
		NULL);

	xTaskCreate(thread_2,
		"thread_2",
		CONFIG_MICRO_ROS_APP_STACK,
		NULL,
		CONFIG_MICRO_ROS_APP_TASK_PRIO + 1,
		NULL);

	while(1){
		// sleep(100);
    vTaskDelay(100 / portTICK_PERIOD_MS);
	}

	// free resources
	RCCHECK(rcl_publisher_fini(&publisher_1, &node));
	RCCHECK(rcl_publisher_fini(&publisher_2, &node));
	RCCHECK(rcl_node_fini(&node));

  	vTaskDelete(NULL);
}

void app_main(void)
{
#ifdef UCLIENT_PROFILE_UDP
    // Start the networking if required
    ESP_ERROR_CHECK(uros_network_interface_initialize());
#endif  // UCLIENT_PROFILE_UDP

    //pin micro-ros task in APP_CPU to make PRO_CPU to deal with wifi:
    xTaskCreate(micro_ros_task,
            "uros_task",
            CONFIG_MICRO_ROS_APP_STACK,
            NULL,
            CONFIG_MICRO_ROS_APP_TASK_PRIO,
            NULL);
} ```

asasine, 2023-04-03:
I'm getting the same build failure when setting `-DUCLIENT_PROFILE_MULTITHREAD=ON` in the meta file

```
micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/profile/multithread/multithread.h:33:10: fatal error: FreeRTOS.h: No such file or directory
```