micro-ROS/micro_ros_espidf_component #156: ESP32 can't find an included file during building of int32_publisher example

starcore2022, 2022-08-08:
**ESP32 can't find the included file of example project.**

- Hardware description: ESP32 serially linked to laptop through usb
- RTOS: FreeRTOS
- Installation type: micro_ros_espidf_component, ESP IDF master(latest)
- Version or commit hash: humble

**Steps to reproduce the issue**
My goal was to make a pcap about the communication, when running the int32_publisher example using embeddedRTPS and micro-ROS component.

 * Installed the ESP IDF master.
 * Copied in the micro_ros_espidf_component into the esp-idf/component directory.
 * Copied the example out to the esp folder and with cd command navigated into it.
 * Used the set-target command to set it up for using esp32.
 * Using the menuconfig command set micro-ROS to embeddedRTPS and set WiFi settings.
 * Then used the build command.

**Actual behavior**
[image]

Complete output:

>  Executing action: all (aliases: build)
> Running ninja in directory /home/starcore/esp/int32_publisher/build
> Executing "ninja all"...
> [1/2] Re-running CMake...-- Component directory /home/starcore/Desktop does not contain a CMakeLists.txt file. No component will be added
> -- Component directory /home/starcore/Documents does not contain a CMakeLists.txt file. No component will be added
> -- Component directory /home/starcore/Downloads does not contain a CMakeLists.txt file. No component will be added
> -- Component directory /home/starcore/Music does not contain a CMakeLists.txt file. No component will be added
> -- Component directory /home/starcore/Pictures does not contain a CMakeLists.txt file. No component will be added
> -- Component directory /home/starcore/Public does not contain a CMakeLists.txt file. No component will be added
> -- Component directory /home/starcore/Templates does not contain a CMakeLists.txt file. No component will be added
> -- Component directory /home/starcore/Videos does not contain a CMakeLists.txt file. No component will be added
> -- Component directory /home/starcore/esp does not contain a CMakeLists.txt file. No component will be added
> -- Component directory /home/starcore/microros_ws does not contain a CMakeLists.txt file. No component will be added
> -- Component directory /home/starcore/snap does not contain a CMakeLists.txt file. No component will be added
> -- Component directory /home/starcore/teszt does not contain a CMakeLists.txt file. No component will be added
> -- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
> -- Building ESP-IDF components for target esp32
> -- Project sdkconfig file /home/starcore/esp/int32_publisher/sdkconfig
> Loading defaults file /home/starcore/esp/int32_publisher/sdkconfig.defaults...
> -- App "int32_publisher" version: 1
> -- Adding linker script /home/starcore/esp/int32_publisher/build/esp-idf/esp_system/ld/memory.ld
> -- Adding linker script /home/starcore/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
> -- Adding linker script /home/starcore/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
> -- Adding linker script /home/starcore/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
> -- Adding linker script /home/starcore/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
> -- Adding linker script /home/starcore/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
> -- Adding linker script /home/starcore/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
> -- Adding linker script /home/starcore/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
> -- Adding linker script /home/starcore/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
> -- Using embeddedRTPS middleware
> mkdir -p /home/starcore/esp/esp-idf/components/micro_ros_espidf_component-humble/micro_ros_src/libmicroros; cd /home/starcore/esp/esp-idf/components/micro_ros_espidf_component-humble/micro_ros_src/libmicroros; \
> for file in $(find /home/starcore/esp/esp-idf/components/micro_ros_espidf_component-humble/micro_ros_src/install/lib/ -name '*.a'); do \
> 	folder=$(echo $file | sed -E "s/(.+)\/(.+).a/\2/"); \
> 	mkdir -p $folder; cd $folder; /home/starcore/.espressif/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar x $file; \
> 	for f in *; do \
> 		mv $f ../$folder-$f; \
> 	done; \
> 	cd ..; rm -rf $folder; \
> done ; \
> /home/starcore/.espressif/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar rc -s libmicroros.a *.obj; cp libmicroros.a /home/starcore/esp/esp-idf/components/micro_ros_espidf_component-humble; \
> cd ..; rm -rf libmicroros; \
> cp -R /home/starcore/esp/esp-idf/components/micro_ros_espidf_component-humble/micro_ros_src/install/include /home/starcore/esp/esp-idf/components/micro_ros_espidf_component-humble/include;
> -- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls micro_ros_espidf_component-humble mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
> -- Component paths: /home/starcore/esp/esp-idf/components/app_trace /home/starcore/esp/esp-idf/components/app_update /home/starcore/esp/esp-idf/components/bootloader /home/starcore/esp/esp-idf/components/bootloader_support /home/starcore/esp/esp-idf/components/bt /home/starcore/esp/esp-idf/components/cmock /home/starcore/esp/esp-idf/components/console /home/starcore/esp/esp-idf/components/cxx /home/starcore/esp/esp-idf/components/driver /home/starcore/esp/esp-idf/components/efuse /home/starcore/esp/esp-idf/components/esp-tls /home/starcore/esp/esp-idf/components/esp_adc /home/starcore/esp/esp-idf/components/esp_common /home/starcore/esp/esp-idf/components/esp_eth /home/starcore/esp/esp-idf/components/esp_event /home/starcore/esp/esp-idf/components/esp_gdbstub /home/starcore/esp/esp-idf/components/esp_hid /home/starcore/esp/esp-idf/components/esp_http_client /home/starcore/esp/esp-idf/components/esp_http_server /home/starcore/esp/esp-idf/components/esp_https_ota /home/starcore/esp/esp-idf/components/esp_https_server /home/starcore/esp/esp-idf/components/esp_hw_support /home/starcore/esp/esp-idf/components/esp_lcd /home/starcore/esp/esp-idf/components/esp_local_ctrl /home/starcore/esp/esp-idf/components/esp_netif /home/starcore/esp/esp-idf/components/esp_phy /home/starcore/esp/esp-idf/components/esp_pm /home/starcore/esp/esp-idf/components/esp_psram /home/starcore/esp/esp-idf/components/esp_ringbuf /home/starcore/esp/esp-idf/components/esp_rom /home/starcore/esp/esp-idf/components/esp_serial_slave_link /home/starcore/esp/esp-idf/components/esp_system /home/starcore/esp/esp-idf/components/esp_timer /home/starcore/esp/esp-idf/components/esp_wifi /home/starcore/esp/esp-idf/components/espcoredump /home/starcore/esp/esp-idf/components/esptool_py /home/starcore/esp/esp-idf/components/fatfs /home/starcore/esp/esp-idf/components/freertos /home/starcore/esp/esp-idf/components/hal /home/starcore/esp/esp-idf/components/heap /home/starcore/esp/esp-idf/components/http_parser /home/starcore/esp/esp-idf/components/idf_test /home/starcore/esp/esp-idf/components/ieee802154 /home/starcore/esp/esp-idf/components/json /home/starcore/esp/esp-idf/components/log /home/starcore/esp/esp-idf/components/lwip /home/starcore/esp/int32_publisher/main /home/starcore/esp/esp-idf/components/mbedtls /home/starcore/esp/esp-idf/components/micro_ros_espidf_component-humble /home/starcore/esp/esp-idf/components/mqtt /home/starcore/esp/esp-idf/components/newlib /home/starcore/esp/esp-idf/components/nvs_flash /home/starcore/esp/esp-idf/components/openthread /home/starcore/esp/esp-idf/components/partition_table /home/starcore/esp/esp-idf/components/perfmon /home/starcore/esp/esp-idf/components/protobuf-c /home/starcore/esp/esp-idf/components/protocomm /home/starcore/esp/esp-idf/components/pthread /home/starcore/esp/esp-idf/components/sdmmc /home/starcore/esp/esp-idf/components/soc /home/starcore/esp/esp-idf/components/spi_flash /home/starcore/ [... cut ...]
> HINT: Please check if you've specified all component dependencies with 'idf_component_register(REQUIRES ...)'. If the component is not present then it should be added by the IDF Component Manager. For more information run 'idf.py docs -sp api-guides/build-system.html'.
> ninja failed with exit code 1, output of the command is in the /home/starcore/esp/int32_publisher/build/log/idf_py_stderr_output_101849 and /home/starcore/esp/int32_publisher/build/log/idf_py_stdout_output_101849
> 

**Additional information**
Modified nothing else and installed the dependencies.

pablogs9, 2022-08-09:
Does this work if you create a component directory in your project and use `EXTRA_COMPONENT_DIRS` CMake variable?

starcore2022, 2022-08-09:
If I use that the problem remains the same, but if i move the conents of the "micro_ros_espidf_component-humble" folder out and delete the folder, then throws error, for not finding this forlder.
[image]

starcore2022, 2022-08-09:
Changed things around, now I removed the **"micro_ros_espidf_component-humble"** folder, from the component folder of esp-idf, because it started clashing with the **EXTRA_COMPONENT_DIRS** setting, now gets trough the the **"#include esp_wifi.h"**, but stops at the "#include rcl/rcl.h" line.
[image]

pablogs9, 2022-08-09:
Remove the "humble" from the folder name: `micro_ros_espidf_component`

starcore2022, 2022-08-09:
Removed and redid the set-target and build:
[image]

starcore2022, 2022-08-09:
If I source the ROS2 and micro-ROS setup, then it stops at the **"esp_wifi.h"**:
[image]

starcore2022, 2022-08-09:
If I do this in a new terminal without sourcing any ROS2 or micro-ROS setup:
[image]

starcore2022, 2022-08-09:
I deleted the ESP IDF and reinstalled the 4.4 version, after redoing the same steps: 
[image]

starcore2022, 2022-08-09:
I redid it with the 4.3 version and gives the same error as with the 4.4 version.

pablogs9, 2022-08-09:
Does it works if you follow [the instructions](https://github.com/micro-ROS/micro_ros_espidf_component#example)?

starcore2022, 2022-08-09:
Tried, but the same error remains:
[image]

pablogs9, 2022-08-09:
Please retry in a clean environment and follow the instructions because CI is passing https://github.com/micro-ROS/micro_ros_espidf_component/actions/runs/2822730553

starcore2022, 2022-08-09:
Deleted the esp folder, where I installed the ESP IDF and now I tried with the esp-idf folder included in the micro-ROS I installed at the microros_ws workspace and redoing as the instructions:
[image]

pablogs9, 2022-08-09:
Provide replicable steps in a clean environment

starcore2022, 2022-08-09:
Thank you for the help, after a lot of clean up and retry finally managed to complete building process with the latest stable (4.4.2) version and placing the int32_publisher project into the examples folder as well as manually creating and cloning in the micro-ROS component into the components folder.

closed 2022-08-09 by starcore2022