micro-ROS/micro_ros_espidf_component #157: multithread_publisher with custom_transport

shecker, 2022-08-17:
## Issue template

- Hardware description: 
- RTOS: 
- Installation type: 
- Version or commit hash: 

#### Steps to reproduce the issue

#### Expected behavior

#### Actual behavior

#### Additional information

Acuadros95, 2022-08-17:
Hi @shecker, a few questions:

- Did you modify the example in any way? (I see three publishers are created instead of two as in the example).
- Does your code works with the default UDP or the provided serial transport?
- Could you elaborate on how your custom transport works?

Acuadros95, 2022-08-18:
> I believe the issue occurs when I change this line from ON to OFF:
`"-DUCLIENT_PROFILE_MULTITHREAD=ON"`

That is expected, this option enables micro-ROS multithread support.
**It needs to be enabled for any multithreaded app.**

closed 2022-09-06 by Acuadros95