micro-ROS/micro_ros_espidf_component #163: Error trying to build with idf.py build ninja: error: '/opt/esp/idf/components/micro_ros_espidf_component/libmicroros.a', needed by 'testEsp32.elf', missing and no known rule to make it ninja failed with exit code 1

Ben01233210, 2022-10-24:
## Issue template
Hi,
i am trying to build a dev setup with devcontainer this i hope works. But every time i try to build any project, i get this error.
`ninja: error: '/opt/esp/idf/components/micro_ros_espidf_component/libmicroros.a', needed by 'testEsp32.elf', missing and no known rule to make it
ninja failed with exit code 1`
I don't think that this error is related with the devcontainer. But i had no more ideas how to fix it. So the last option crossed my mind was writing an issue. I hope my form and info's are enough (If you have tips for writing Dockerfile or devcontainer.json please tell me).
Thanks!! 

- Hardware description: just using en esp32
- Version or commit hash: on my pc i have a nother devcontainer with humbel, but i think its not relevant

#### Steps to reproduce the issue
I think just run the devcontainer

#### Expected behavior
So of course that is builds : )

#### Actual behavior
```
-- Configuring done
-- Generating done
-- Build files have been written to: /workspaces/testEsp32/build
Running ninja in directory /workspaces/testEsp32/build
Executing "ninja all"...
[0/2] Re-checking globbed directories...
ninja: error: '/opt/esp/idf/components/micro_ros_espidf_component/libmicroros.a', needed by 'testEsp32.elf', missing and no known rule to make it
ninja failed with exit code 1`
```

#### Additional information

This is my devcontainer.json 
```
/ For format details, see https://aka.ms/devcontainer.json. For config options, see the README at:
// https://github.com/microsoft/vscode-dev-containers/tree/v0.245.2/containers/ubuntu
{
	"name": "Esp 32 test",
	"build": {
		"dockerfile": "Dockerfile"
	},
	"runArgs": [
        // This will allow you to use a ptrace-based debugger like C++, Go, and Rust.
        "--cap-add=SYS_PTRACE",
        "--security-opt", "seccomp=unconfined",
		"--device=/dev/ttyUSB0"
    ],
	"settings": {
		"idf.showOnboardingOnInit": false,
		"idf.toolsPath": "/opt/esp",
		"idf.espIdfPath": "/opt/esp/idf",
		"idf.pythonBinPath": "/opt/esp/python_env/idf4.2_py3.6_env/bin/python",
		"idf.port": "/dev/ttyUSB0",
		"terminal.integrated.shell.linux": "/bin/bash"
	},
	"extensions": [
		"ms-azuretools.vscode-docker",
        "ms-vscode.cpptools",
        "twxs.cmake",
        "ms-vscode.cmake-tools",
        "smilerobotics.urdf",
        "yzhang.markdown-all-in-one",
		"espressif.esp-idf-extension"
	],
	"userEnvProbe": "loginInteractiveShell"
	// Forward serial device to enable flash & monitor
	// Use 'forwardPorts' to make a list of ports inside the container available locally.
	// "forwardPorts": [],
}
```

an this my Dockerfile for the devcontainer:
```
FROM espressif/idf:release-v4.4
ENV DEBIAN_FRONTEND noninteractive
RUN echo "Set disable_coredump false" >> /etc/sudo.conf
RUN apt update -q && \
    apt install -yq sudo lsb-release gosu nano 
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

RUN apt update -q
RUN apt install -yq python3-pip
RUN pip3 install catkin_pkg lark-parser empy colcon-common-extensions importlib-resources
RUN git clone https://github.com/micro-ROS/micro_ros_espidf_component.git --recurse-submodules
RUN cp -r /micro_ros_espidf_component $IDF_PATH/components/
RUN rm -Rf /micro_ros_espidf_component
RUN echo "source /opt/esp/idf/export.sh" >> /root/.bashrc
```

Ben01233210, 2022-10-25:
I found the error. I cloned the repo in to the global components of espidf an not into the components of the project.

But now got a new error:
```
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /workspaces/testEsp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/opt/esp/python_env/idf5.1_py3.8_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 /workspaces/testEsp32"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /workspaces/testEsp32/sdkconfig
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /opt/esp/python_env/idf5.1_py3.8_env/bin/python (found version "3.8.10") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "testEsp32" version: 1
-- Adding linker script /workspaces/testEsp32/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /opt/esp/idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /opt/esp/idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Using Micro XRCE-DDS middleware
mkdir -p /workspaces/testEsp32/components/micro_ros_espidf_component/micro_ros_src/libmicroros; cd /workspaces/testEsp32/components/micro_ros_espidf_component/micro_ros_src/libmicroros; \
for file in $(find /workspaces/testEsp32/components/micro_ros_espidf_component/micro_ros_src/install/lib/ -name '*.a'); do \
        folder=$(echo $file | sed -E "s/(.+)\/(.+).a/\2/"); \
        mkdir -p $folder; cd $folder; /opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar x $file; \
        for f in *; do \
                mv $f ../$folder-$f; \
        done; \
        cd ..; rm -rf $folder; \
done ; \
/opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar rc -s libmicroros.a *.obj; cp libmicroros.a /workspaces/testEsp32/components/micro_ros_espidf_component; \
cd ..; rm -rf libmicroros; \
cp -R /workspaces/testEsp32/components/micro_ros_espidf_component/micro_ros_src/install/include /workspaces/testEsp32/components/micro_ros_espidf_component/include;
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls micro_ros_espidf_component mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /opt/esp/idf/components/app_trace /opt/esp/idf/components/app_update /opt/esp/idf/components/bootloader /opt/esp/idf/components/bootloader_support /opt/esp/idf/components/bt /opt/esp/idf/components/cmock /opt/esp/idf/components/console /opt/esp/idf/components/cxx /opt/esp/idf/components/driver /opt/esp/idf/components/efuse /opt/esp/idf/components/esp-tls /opt/esp/idf/components/esp_adc /opt/esp/idf/components/esp_app_format /opt/esp/idf/components/esp_common /opt/esp/idf/components/esp_eth /opt/esp/idf/components/esp_event /opt/esp/idf/components/esp_gdbstub /opt/esp/idf/components/esp_hid /opt/esp/idf/components/esp_http_client /opt/esp/idf/com [... cut ...] /idf/components/newlib /opt/esp/idf/components/partition_table /opt/esp/idf/components/soc /opt/esp/idf/components/spi_flash /opt/esp/idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /workspaces/testEsp32/build/bootloader
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /workspaces/testEsp32/build/log/idf_py_stderr_output_35625 and /workspaces/testEsp32/build/log/idf_py_stdout_output_35625
```

Ben01233210, 2022-10-26:
ok, so if fixed also the last issue :-)

closed 2022-10-26 by Ben01233210

pablogs9, 2022-10-27:
How did you solve it?

JuliusSeif, 2022-11-29:
@Ben01233210  I have the same problem, with the "esp_wifi.h" and "esp_netif.h", how did you solved it?

tianrking, 2022-12-20:
how to solve it

Usama-Arshad16, 2023-04-04:
how did you solve it?