micro-ROS/micro_ros_espidf_component #166: Missing Headers esp_netif.h and esp_wifi.h

JuliusSeif, 2022-11-29:
- Hardware description: esp32s3
- RTOS: freeRTOS
- Version or commit hash: humble

#### issue
I get always an error, when I try to build my projects.
[log.txt](https://github.com/micro-ROS/micro_ros_espidf_component/files/10112601/log.txt)

- I have installed ESP-IDF with the VS-Code extension 
- I haven`t sourced humble
- I cloned this repository  in the esp-idf/components/ folder
- when I try to build an example the error occurs

pablogs9, 2022-11-29:
Please provide instructions for replicating the issue, also provide which ESP-IDF version are you using

JuliusSeif, 2022-11-29:
I use ES-IDF 5.0

   - I have installed ESP-IDF with the VS-Code extension
   - I haven`t sourced humble
   - I cloned this repository in the esp-idf/components/ folder
   - copied the int32-publisher example into my main esp folder
   - opened it with vs-Code
   - select esp32s3 as target and /dev/ttyUSB0 as port
   - edited the menuconfig with my wifi`s ssid and password
   - Hit build and the error from the log.txt  file occurs

pablogs9, 2022-11-30:
ESP-IDF v5 is not supported: 
[image]

Check this: https://github.com/micro-ROS/micro_ros_espidf_component/pull/162

closed 2022-12-05 by pablogs9