micro-ROS/micro_ros_espidf_component #170: libmicroros.a  missing and no known rule to make it   with idf py4.3

tianrking, 2022-12-22:
```
ninja: error: '/home/w0x7ce/Desktop/AAA/micro_ros_espidf_component/examples/int32_publisher/components/micro_ros_espidf_component/libmicroros.a', needed by 'int32_publisher.elf', missing and no known rule to make it
```

Hi , how to solve this problem

pablogs9, 2022-12-22:
Hi, could you provide instructions for replicating your error?

tianrking, 2022-12-23:
I follow this step 

```bash
. $IDF_PATH/export.sh
cd examples/int32_publisher
# Set target board [esp32|esp32s2|esp32s3|esp32c3]
idf.py set-target esp32
idf.py menuconfig
# Set your micro-ROS configuration and WiFi credentials under micro-ROS Settings
idf.py build
idf.py flash
idf.py monitor
```

but dependence in libmicroros.mk seems not be installed , how to auto install them  in  libmicroros.mk

tianrking, 2022-12-23:
after I run 

``` bash 
make -f libmicroros.mk
···

```bash
Starting >>> rosidl_typesupport_interface
--- stderr: rcutils                                                                                     
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:176,
                 from /usr/include/stdlib.h:395,
                 from /root/O/micro_ros_espidf_component/micro_ros_src/src/rcutils/src/allocator.c:15:
/root/O/esp-idf-4.4/components/newlib/platform_include/endian.h:51:10: fatal error: machine/endian.h: No such file or directory
   51 | #include <machine/endian.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[3]: *** [CMakeFiles/rcutils.dir/build.make:82: CMakeFiles/rcutils.dir/src/allocator.c.obj] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:192: CMakeFiles/rcutils.dir/all] Error 2
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [Makefile:146: all] Error 2
---
Failed   <<< rcutils [1.97s, exited with code 2]
```

tianrking, 2022-12-23:
I try  

````bash
$micro_ros_espidf_component/examples/int32_sub_pub#  rm -rf build
$micro_ros_espidf_component# make -f libmicroros.mk clean
```

And then

```bash
. $IDF_PATH/export.sh
cd examples/int32_publisher
idf.py set-target esp32
idf.py menuconfig
idf.py build
```

Then error output

```bash
[100%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
/root/O/micro_ros_espidf_component/examples/int32_publisher/main/main.c:11:10: fatal error: rcl/rcl.h: No such file or directory
 #include <rcl/rcl.h>
          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:76: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:4839: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make failed with exit code 2
```

pablogs9, 2022-12-23:
Your micro-ROS library is not building correctly, which version of ESP-IDF are you using?

tianrking, 2022-12-23:
before idf-py 4.3

now  idf-py4.4 docker

I follow this [https://asciinema.org/a/EV8cCZTwjbzE3XoiPlwVNkW1C](https://asciinema.org/a/EV8cCZTwjbzE3XoiPlwVNkW1C)  video
I have successfully  build for esp32,esp32s3 but still fail for esp32c3 

```bash
idf.py set-target esp32c3 
idf.py menuconfig
idf.py build
```

here is the output

```bash
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a(librcl-init_options.c.obj): relocations in generic ELF (EM: 94)
/opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /micro_ros_espidf_component/libmicroros.a: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

pablogs9, 2022-12-23:
Seems that something is not being built correctly. Please retry in a clean environment.

VineetTambe, 2023-03-12:
Did anyone find a solution to this issue??? I am stuck on this for almost 5 hours now. Any help would be great :)

pablogs9, 2023-03-13:
@Acuadros95 ptal

Acuadros95, 2023-03-13:
Hi @VineetTambe,

Could you give more details on your exact issue? 
- Which board and idf version are you using?
- Did you try with the docker approach?
- What is your error output on the micro-ROS library build step?

Acuadros95, 2023-07-24:
Closing due to inactivity, reopen if needed

closed 2023-07-24 by Acuadros95